mu-editor/mu #553: REPL text size - always start with default size

martinohanlon, 2018-07-13:
When you start the REPL it always starts up in the default text size, not the size of the text in the code window.

To replicate.

+ Open Mu
+ Close REPL (if its open)
+ Zoom in all the way on the code editor
+ Open the REPL
+ It will be at the default text size.

[image]

If you open the REPL before zooming in / out, the text in the REPL is sized correctly.

ZanderBrown, 2018-07-13:
We really need #530 to abstract panels

carlosperate, 2018-07-13:
I think in this case we also need to abstract zoom level as well.
Clicking on the zoom buttons only affects the stuff that's open (so for instance if you open a new text tab it will start with the default zoom level), we need something to hold state and every feature that can change its zoom should be able to check what's the current level.

@ntoll is this likely a candidate for "post-1.0"?

ntoll, 2018-07-13:
Given how zooming is implemented, it should be trivial to fix. Let me see how far I get today. ;-)

ntoll, 2018-07-15:
Hmm... OK, I've just looked into this. It's not so simple so I'll say this is a todo for 1.1 rather than a quick fix for 1.0 (out this week!). ;-)

tjguk, 2018-07-25:
To confirm: is the general idea here to retain zoom level as state (ie in the session file) and to re-apply to all panels when restarted? Because I think that's probably what's wanted, ie if we have visually impaired users (or people with *really* good eyesight and long lines of code) they'll presumably want to always have very big (or very small) text.

carlosperate, 2018-07-25:
While I agree that zoom level saved in the settings file could be a good addition (I'm not totally convinced yet, as I like having a known state when opened, but the use case about impaired users is a very good one that's hard to argue against), I don't think it should be done with the panes as well. Having a default known state when they are opened can be useful for the classroom and to match any posible screenshots.

But I haven't given this much thought, so there could be a very good reason to do it.

ntoll, 2018-12-18:
Fixed by  5b82f22

closed 2018-12-18 by ntoll