mu-editor/mu #557: Disabling/enabling flashing button on micro:bit FS flashing

carlosperate, 2018-07-16:
Perhaps not one for v1.0, since RC is about to drop, but this could be something to look into after the flashing method is refactored.

When the micro:bit is flashed via the old method (embed code into hex file, or flashing the runtime on its own) then the flash button is deactivated/greyed-out which is a great indication to the user that the flashing is occurring.

When the flash is accomplied via seding a main.py file via serial this happens very quickly (thanks again Nicholas! 🎊 ) and users might be used to seeing a change in the button that doesn't happen anymore. 
There is a message shown in the status bar, but that's quite far away from where most people will be looking at the moment the click "Flash". Could it be a good idea to disable and enable the button as well on this type of operation?

ntoll, 2018-07-16:
I thought it did that (disable/enable) the button. If it doesn't, that's a regression and it'll be fixed in a moment. ;-)

carlosperate, 2018-07-16:
Ups, clicked the wrong button.

I didn't check if the code was doing it, but if it does already then the UI is not updated. I tried flashing a larger script so it takes a bit longer and cannot see the button change.

ntoll, 2018-07-16:
Yeah I'm just looking into it right now before deep diving into the PyGame issue.

ntoll, 2018-07-16:
OK... this needs a refactor of the `flash` function. Essentially it's currently blocking when using the new serial based method, so the call to turn off the buttons isn't happening in a way that updates the UI since it's all part of the blocking function. Once 1.0 is out the door, this'll be refactored and solved.

Leaving this issue open until such time as the `flash` function is rewritten to be non-blocking.