mu-editor/mu #568: 'Files' feature not working with micro:bit (Mu 1.0.0.rc.1)

AlasdairAtKitronik, 2018-07-19:
Trying to use the 'Files' feature in Mu to access the micro:bit file system and then add my own library files. 
Every time I click the 'Files' button, I get this error message:

'There was a problem getting the list of files on the micro:bit. Please check Mu's logs for technical information. Alternatively, try unplugging/plugging-in your micro:bit and/or restarting Mu.'

I have reset the micro:bit and retried. I have unplugged and plugged in the micro:bit and retried. I have restarted Mu and retried. It fails every time.

ntoll, 2018-07-19:
Hi @AlasdairAtKitronik does your micro:bit have the latest firmware (if not, please update it, see: https://support.microbit.org/support/solutions/articles/19000019131-how-to-upgrade-the-firmware-on-the-micro-bit). 

You don't mention which operating system you're using. If you're on Linux, as the download page states (https://codewith.mu/en/download), you'll need to add yourself to the correct group in order to have the correct permissions to use USB serial (which is how the "Files" feature works).

Hope this helps!

ntoll, 2018-07-19:
I'm closing this and we can track the serial issue on #569 since they're the same fundamental cause.

closed 2018-07-19 by ntoll