mu-editor/mu #571: Debugger - highlighted line goes over by 1 character

martinohanlon, 2018-07-20:
When you start the debugger, the highlighted line bleeds over in the next line by 1 character.

[image]

I have seen this a couple of times now. It doesnt always happen, but I havent been able to work out what makes it happen!

ntoll, 2018-07-20:
That was fixed in rc.1. See: https://mu.readthedocs.io/en/latest/changes.html#rc-1

(It was an ObOE caused by Windows newlines being `\r\n` rather than `\n` which meant the reported `len` of the line was one (invisible) character too many). :-)

closed 2018-07-20 by ntoll

martinohanlon, 2018-07-20:
Thanks, I was just installing rc1 to see if it was fixed.

ntoll, 2018-07-20:
Give me a couple of hours and (hopefully) 1.0.0 will be out. ;-)

martinohanlon, 2018-07-20:
Damn it, Ill have to install 2 versions in 1 day!

ntoll, 2018-07-20:
...and inevitably you'll find a bug. ;-)