mu-editor/mu #572: Locale settings are not reflected in macOS

MinoruInachi, 2018-07-21:
I checked Japanese translation at mu 1.0.0, but labels and messages remained in English.

The cause is that locale setting does not pass to python 3 when launching `mu-editor.app` from macOS 's Finder.  The values of the locale related environment variables at the time of starting python 3 were as follows:

```
LANG=
LC_COLLATE="C"
LC_CTYPE="C"
LC_MESSAGES="C"
LC_MONETARY="C"
LC_NUMERIC="C"
LC_TIME="C"
LC_ALL=
```
I made the following modifications to the startup script `/Applications/mu-editor.app/Contents/MacOS/mu_editor` and confirmed that the display will be in Japanese. I have confirmed this fix only in Japanese, and I have not confirmed whether it is okay with other languages.

```
--- mu-editor.org	2018-07-20 21:18:29.000000000 +0900
+++ mu-editor	2018-07-21 09:32:55.000000000 +0900
@@ -1,4 +1,5 @@
 #!/bin/sh
+"eval" "export LC_ALL=`defaults read -g AppleLocale | sed 's/@.*$//g'`.UTF-8"
 "exec" "`dirname $0`/../Resources/python/bin/python3" "$0" "$@"
 # EASY-INSTALL-ENTRY-SCRIPT: 'mu-editor==1.0.0','console_scripts','mu-editor'
 __requires__ = 'mu-editor==1.0.0'
```

carlosperate, 2018-07-29:
Yes, I can confirm that when launching the installed app from Finder the locale data is `(None, None)`:
[image]

tmontes, 2018-08-01:
Quick feedback:
* I confirm the behaviour noted by @MinoruInachi.
* I confirm the patch to the `mu-editor` shell script fixes the issue on my system.

A few extra observations, running macOS Sierra 10.12.6, where the "Language & Region" settings are:
* Preferred language: English
* Region: Portugal

With that, the `defaults read -g AppleLocale` prints out a non-obvious `en_PT`. This is a language string I had never seen before but that, given some thought, ends up making sense, I guess.

AFAICT, mu is currently using only the first two characters in the language code to determine its UI and help URL languages so this fact does not raise any limitation I can think of if a macOS fix for this based on `defaults read -g AppleLocale` is put in place.

If/when mu takes the full language code into consideration, then whichever fallback rules are put in place, may want to account for this fact.

tmontes, 2018-08-01:
For completeness, the "proper macOS way" of getting the `AppleLocale` user preference, in Python, is something along these lines:

```
import Foundation

PREF_NAME = 'AppleLocale'
FALL_BACK = 'en_GB'

sud = Foundation.NSUserDefaults.standardUserDefaults()
pref_value = sud.stringForKey_(PREF_NAME) or FALL_BACK

print(f'macOS global {PREF_NAME!r}: {pref_value!r}')
```

This depends on `pyobjc-framework-Cocoa`, though (see https://pypi.org/project/pyobjc-framework-Cocoa/).

JohnGuan, 2018-10-10:
So... will we fix this bug? @ntoll @tmontes

ntoll, 2018-10-10:
Yes! :-)

tmontes, 2018-12-19:
@MinoruInachi, @JohnGuan,

PR #672 containing a fix for this issue has been merged, so you can count on a fix being included in the next release which should be out soon, AFAICT.

tmontes, 2019-01-27:
I guess this issue can be closed. Do you confirm things are working nicely with regards to language detection on macOS, @MinoruInachi , @JohnGuan?

closed 2019-03-31 by MinoruInachi