mu-editor/mu #575: PyQt5/QtCore.so: undefined symbol: PySlice_AdjustIndices

slippers, 2018-07-23:
### starting mu-editor results in error
```
>mu-editor

Traceback (most recent call last):
  File "/home/projects/mu/.env/bin/mu-editor", line 7, in <module>
    from mu.app import run
  File "/home/projects/mu/.env/lib/python3.5/site-packages/mu/app.py", line 29, in <module>
    from PyQt5.QtCore import QTimer, Qt
ImportError: /home/projects/mu/.env/lib/python3.5/site-packages/PyQt5/QtCore.so: undefined symbol: PySlice_AdjustIndices
```

###  setup mu by
mkdir mu
cd mu
virtualenv -p python3 .env
activate
pip install mu-editor
mu-editor

### machine
using linux ubuntu and mint with same results

>uname -a

4.4.0-130-generic #156-Ubuntu SMP Thu Jun 14 08:53:28 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

>python --version

Python 3.5.2

>pip list

Package          Version   
---------------- ----------
appdirs          1.4.3     
backcall         0.1.0     
certifi          2018.4.16 
chardet          3.0.4     
cycler           0.10.0    
decorator        4.3.0     
gpiozero         1.4.1     
guizero          0.5.3     
idna             2.7       
ipykernel        4.8.2     
ipython          6.4.0     
ipython-genutils 0.2.0     
jedi             0.12.1    
jupyter-client   5.2.3     
jupyter-core     4.4.0     
kiwisolver       1.0.1     
matplotlib       2.2.2     
mu-editor        1.0.0     
numpy            1.15.0    
parso            0.3.1     
pexpect          4.6.0     
pgzero           1.2       
pickleshare      0.7.4     
pigpio           1.40.post1
Pillow           5.2.0     
pip              18.0      
pkg-resources    0.0.0     
prompt-toolkit   1.0.15    
ptyprocess       0.6.0     
pycodestyle      2.4.0     
pyflakes         2.0.0     
pygame           1.9.4     
Pygments         2.2.0     
pyparsing        2.2.0     
PyQt5            5.11.2    
PyQt5-sip        4.19.12   
PyQtChart        5.11.2    
pyserial         3.4       
python-dateutil  2.7.3     
pytz             2018.5    
pyzmq            17.1.0    
QScintilla       2.10.7    
qtconsole        4.3.1     
requests         2.19.1    
semver           2.8.1     
setuptools       40.0.0    
simplegeneric    0.8.1     
six              1.11.0    
tornado          5.1       
traitlets        4.3.2     
urllib3          1.23      
wcwidth          0.1.7     
wheel            0.31.1

carlosperate, 2018-07-23:
Thank you for the report @slippers 

This is likely to be the same issue as https://github.com/mu-editor/mu/issues/538.
@tjguk as you mentioned in https://github.com/mu-editor/mu/issues/538#issuecomment-404096351, this looks like a problem from the Python 3.5 wheel generated by RiverBank not being compatible and using a Python 3.6 API. Did you find any bug reports or issue in the mainling list?
Their last wheel was released on the 2nd of July, so this will likely be fixed in the next release, if it has been reported already.

carlosperate, 2018-07-23:
Okay, this specific issue is first reported here: https://www.riverbankcomputing.com/pipermail/pyqt/2018-July/040511.html
So it has been reported, and the maintainer is aware of the problem, so it's safe to assume it will be fixed in the next release (whenever that might be).

For more info, looks like the incompatibility is with versions =< 3.5.3 (deprecation): https://www.riverbankcomputing.com/pipermail/pyqt/2018-July/040514.html

florianletsch, 2018-07-24:
I found this issue because I have the same problem.

Just FYI: I can confirm that it does not work with Python 3.5.2. I switched my environment to Python 3.6.5 and there it works.

ntoll, 2018-07-27:
Hi Folks,

Thanks for your patience with this. It looks like we just have to wait on Riverbank to fix their wheels (sorry).

N.

closed 2018-10-18 by ntoll