mu-editor/mu #579: Please provide an AppImage for Linux

probonopd, 2018-07-27:
Currently it is extremely easy to download and run Mu on the Mac, but on Linux the command line is involved, making it more complex than on the Mac.

Providing an [AppImage](http://appimage.org/) would have, among others, these advantages:
- Applications packaged as an AppImage can run on many distributions (including Ubuntu, Fedora, openSUSE, CentOS, elementaryOS, Linux Mint, and others)
- One app = one file = super simple for users: just download one AppImage file, [make it executable](http://discourse.appimage.org/t/how-to-make-an-appimage-executable/80), and run
- No unpacking or installation necessary
- No root needed
- No system libraries changed
- Works out of the box, no installation of runtimes needed
- Optional desktop integration with `appimaged`
- Optional binary delta updates, e.g., for continuous builds (only download the binary diff) using AppImageUpdate
- Can optionally GPG2-sign your AppImages (inside the file)
- Works on Live ISOs
- Can use the same AppImages when dual-booting multiple distributions
- Can be listed in the [AppImageHub](https://appimage.github.io/apps) central directory of available AppImages
- Can double as a self-extracting compressed archive with the `--appimage-extract` parameter

An AppImage is essentially a self-mounting filesystem (similar to a Mac disk image) that contains Python, PyQt, Python libraries, icons, other resources, and the application.

Complex Python/PyQt applications such as Ultimaker Cura are already being shipped in AppImage format.

[Here is an overview](https://appimage.github.io/apps) of projects that are already distributing upstream-provided, official AppImages.

Here is a simple recipe that can produce an AppImage from pip:
https://github.com/AppImage/AppImages/blob/master/recipes/Mu.yml

On a Debian/Ubuntu based distribution, you can run it with

```
wget https://raw.githubusercontent.com/AppImage/AppImages/master/pkg2appimage
bash -ex pkg2appimage Mu
```

If you have questions, AppImage developers are on #AppImage on irc.freenode.net.

ntoll, 2018-07-27:
Oooh... I didn't know about this. I'll take a look.

A quick question: what testing is available to check the appimage on all supported platforms?

probonopd, 2018-07-27:
I am using [`testappimage`](https://github.com/AppImage/AppImageKit/blob/appimagetool/master/testappimage) along with Live ISOs of various distros.

abixadamj, 2018-08-29:
I agree that AppImage would be great solution! I will help with creating this.

probonopd, 2018-08-29:
On a Debian/Ubuntu based distribution, please try

```
wget https://raw.githubusercontent.com/AppImage/AppImages/master/pkg2appimage
bash -ex pkg2appimage Mu
```

It should produce a working AppImage.

tim1mw, 2019-07-02:
+1 for this, I would have much preferred an appimage.

 I'm currently wrestling with the pip install which is crashing due to other changes in the latest distros (#875 ) .

closed 2021-03-24 by carlosperate