mu-editor/mu #580: Investigate minification options / consolidation.

ntoll, 2018-07-27:
It has been pointed out to me that there's an existing minification utility for Python:

https://github.com/liftoff/pyminifier

We currently use nudatus by @ZanderBrown -- I'd like to explore how we may, perhaps, work out the best way forward. For example, could nudatus use pyminifier? Is there a reason why pyminifier won't work with MicroPython on the micro:bit? Does nudatus give us something pyminifier cannot? Could nudatus and pyminifier consolidate efforts (e.g. avoid re-inventing the wheel).

Just throwing out thoughts. Comments, critique and ideas most welcome.

ZanderBrown, 2018-07-27:
As far as I'm aware pyminifier doesn't preserve line numbers which nudatus does to aid debugging

Otherwise pyminifier is technically superior in just about every other way largely as it is a 'true' minifier

eduvik, 2020-06-14:
I have just been trying out pyminifier; the obfuscation mode reduces tokens to single characters, which is great for reducing code size, but it's worth considering that when using "from microbit import *" (or some other "from X import Y" statements) at the start (as is often used in microbit programs) things from the microbit library get replaced with single chars incorrectly (e.g. sleep() might get replaced with k(), without a k=sleep assignment). This might be as pyminifier doesn't know anything about the microbit library or micropython, and might be possible to fix with the correct parameters.

dybber, 2020-06-14:
I guess you want to keep all names as is anyway, otherwise, error messages will also be obfuscated?

eduvik, 2020-06-14:
Depends on what the purpose of the minification is; the project I'm currently working on I need every last bit of disk space I can get for data logging. Yes, that makes debugging harder; but I end up debugging on pre-minified code. Possibly I'm not quite the target audience, but might be worth considering anyway.

dybber, 2020-06-15:
I don't think that use case is in scope for what Mu should handle, as first and foremost a beginner editor, it's all about ease of use.

In anycase, I guess you would want to move that data to more permanent storage once in a while, for small devices it will just be more frequent. If the device is internet-connected, I can recommend using something like https://io.adafruit.com/ for datalogging - I've used it in class with succes (there's several of these services, I tested around 10 of them an I think adafruit's the simplest)