mu-editor/mu #581: Debug not working with requests module

g70104sm, 2018-07-27:
When debugging with Mu I've encountered problem while debugging program using 'requests' module.
To reproduce:
1. Write a program in Python3 mode:
import requests
print("hello")
2. Click debug, then Step Over

Expected: 
debug moves on to next line

Actually Happened (debug window output):
exception: Traceback (most recent call last):
  File "C:\Users\smiotk\AppData\Local\Mu\pkgs\mu\debugger\runner.py", line 494, in run
    debugger._runscript(filename)
  File "C:\Users\smiotk\AppData\Local\Mu\pkgs\mu\debugger\runner.py", line 469, in _runscript
    self.run(e)
  File "bdb.py", line 431, in run
  File "<string>", line 1, in <module>
  File "c:\users\smiotk\mu_code\hello.py", line 1, in <module>
    import requests
  File "C:\Users\smiotk\AppData\Local\Mu\pkgs\requests\__init__.py", line 43, in <module>
    import urllib3
  File "C:\Users\smiotk\AppData\Local\Mu\pkgs\urllib3\__init__.py", line 8, in <module>
    from .connectionpool import (
  File "C:\Users\smiotk\AppData\Local\Mu\pkgs\urllib3\connectionpool.py", line 11, in <module>
    from .exceptions import (
  File "C:\Users\smiotk\AppData\Local\Mu\pkgs\urllib3\exceptions.py", line 2, in <module>
    from .packages.six.moves.http_client import (
  File "C:\Users\smiotk\AppData\Local\Mu\pkgs\urllib3\packages\six.py", line 203, in load_module
    mod = mod._resolve()
  File "C:\Users\smiotk\AppData\Local\Mu\pkgs\urllib3\packages\six.py", line 115, in _resolve
    return _import_module(self.mod)
  File "C:\Users\smiotk\AppData\Local\Mu\pkgs\urllib3\packages\six.py", line 82, in _import_module
    __import__(name)
ModuleNotFoundError: No module named 'http'

---------- FINISHED ----------
exit code: 0 status: 0

ENVIRONMENT:
Win 10 
Mu 1.0.0

Thanks for creating Mu, I love its simplicity!

tjguk, 2018-07-27:
Hand-wavingly, I suspect that this may be relevant: http://mitogen.readthedocs.io/en/stable/shame.html#six

ntoll, 2018-07-27:
@g70104sm thank you for reporting this. Ouch.

I concur with @tjguk -- it looks like the `six` module is trying to do something that breaks things on Windows (it all works perfectly on Linux).

Confusing. :-/

tim-mccurrach, 2018-08-17:
This seems very similar to issue #612. Running the above code works fine, on my development version but doesn’t work on my installed version, and on the installed version only fails when run in debug mode (windows 10). 

Even just `import http` fails on debug. Could it not actually be the issue with `six.py`? 

It seems like python is searching in different places to import when running in debug mode for some reason (see comment in issue #612).  Some experimentation suggests quite a lot of modules from the standard library are affected.

carlosperate, 2018-08-20:
I agree with @tim-mccurrach, in this case the `http` module is part of python36.zip, so it looks like the same problem as #612.

tmontes, 2019-06-01:
Dear all, and @g70104sm in particular,

This issue is quite old by now and I suppose may have been fixed with PR #726, included in Mu 1.0.2 released back in January. Can you please confirm that is the case and, if so, that this issue may be closed?

Thanks.

JLFra, 2020-03-30:
Hello,
I use Mu v1.1.0 alpha2
I try to import request, but
ModuleNotFoundError: No module named 'requests'
So how to import requests module in Mu ide?
thanks to help me please.
regards.
Julien

carlosperate, 2020-09-29:
It has been a long time, so we'll close this issue as resolved (probably by PR https://github.com/mu-editor/mu/pull/726).
@JLFra if you are still experiencing issues please open a new ticket and we'll be able to look into it, thanks!

closed 2020-09-29 by carlosperate