mu-editor/mu #582: Mu 1.0.0 for Windows x64 report an error after flash

jxgx072037, 2018-07-28:
Hi,

I met a problem after I flash the main.py to micro:bit.

[image]

The 'magibit' is a module I developed. And I combined the runtime.hex and the module into a new .hex file (attached). I have set the file path in Mu.

[image]

Mu Version：1.0.0, Windows, X64
I can get the right response when I write the code above in REPL, and I could also get the right response when I use Mu of Mac. The LED on micro:bit should display the distance num.

[log.txt](https://github.com/mu-editor/mu/files/2237794/log.txt)

[hex_file.zip](https://github.com/mu-editor/mu/files/2237796/hex_file.zip)

Thanks!

jxgx072037, 2018-07-28:
I found a way to bypass the problem 0.0

I found the uflash.py, and copy the new .hex string to replace the default _RUMTIME.

And, it works!!!

carlosperate, 2018-07-28:
Hi,

How did you create the hex file you linked in your original message?
Mu/uflash would expect that hex file to contain only the MicroPython runtime, without any additional code attached to it. Does that hex file have any code added to the runtime?

jxgx072037, 2018-07-29:
Hi Carlos,

The .hex file does have a module which I developed to control sensors like ultrasonic. I make it into the runtime part in .hex file.

carlosperate, 2018-07-29:
Did you add that module into the MicroPython hex using uFlash? What version of MicroPython and what verison of uFlash have you used to do that?
The flashing method has changed recently in Mu and MicroPython has also been updated to prioritise files saved in the file system vs. files added by attaching the code into the hex.
uFlash, older versions of Mu and the online editor (https://python.microbit.org/v/1) used to add code to the hex file by attaching it to a known flash location.

The newest version Mu will use the internal micro:bit MicroPython filesystem instead, so there could be conflicts if you are using older versions of MicroPython + inject code + the new flashing method in Mu.

The best way to solve this is to simply use the tools available in Mu.
* First make sure there is no additional hex file specified in the settings, so make sure it looks like this:
    [image]
* Create a new file without any text in it and flash that to the micro:bit.
    * This will flash to the micro:bit the latest MicroPython interpreter
* Open the REPL to ensure you have the latest version flashed. It should show the following text, note the listed versions it's important they are the same:
    ```
    MicroPython v1.9.2-34-gd64154c73 on 2017-09-01; micro:bit v1.0.0-rc.2 with nRF51822
    Type "help()" for more information.
    >>> 
    ```
* Close the REPL and use the Files button to transfer your module, make sure this module is not named `main.py`, any other name should be fine (as long as it uses UTF-8 characters).
    * More info can be found in the Mu website: https://codewith.mu/en/howto/copy_files_microbit
* Use the flash functionality in Mu, it should now transfer your code as the main.py file and should keep both files in the micro:bit internal filesystem.

Could you give this a go? This way it should work a lot more reliably.

jxgx072037, 2018-07-30:
Hi Carlos,

Thanks for your reply! I have tried the way you said, but the micro:bit RAM(16kb) is not enough for the python module.

I made the .hex  file by the newest micropython tool, it seems work well on mac.

carlosperate, 2020-10-26:
This issue should be resolved by using "overwriting MicroPython hex" as described in https://github.com/mu-editor/mu/issues/582#issuecomment-408680494.

[image]

I'll close this issue as it is quite old, but feel free to reopen if this is a problem with the new method.

closed 2020-10-26 by carlosperate