mu-editor/mu #584: REPL crashes on macOS

PeterWAWood, 2018-07-29:
I've just installed Mu on a Mac (macOS 10.12.6). When clicking on the `REPL` button, the Python interpreter crashes, reloads, crashes ...

MU is showing the name of the REPL as Python3 (Jupiter) REPL. The message in the REPL windows are "Kernel died, Restarting".

I have Python 3.7.0 installed on my machine, both `ipython` and `python3` commands open a Python 3.7 REPL. The 'python' command runs the Apple installed  Python 2.7.5 REPL.

The mu log:
```text
2018-07-29 12:17:47,261 - root:112(run) INFO: 

-----------------

Starting Mu 1.0.0
2018-07-29 12:17:47,261 - root:113(run) INFO: uname_result(system='Darwin', node='mbp.local', release='16.7.0', version='Darwin Kernel Version 16.7.0: Thu Jun 21 20:07:39 PDT 2018; root:xnu-3789.73.14~1/RELEASE_X86_64', machine='x86_64', processor='i386')
2018-07-29 12:17:47,261 - root:114(run) INFO: Python path: ['/Applications/mu-editor.app/Contents/MacOS', '/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/mu-editor.app/Contents/Resources/app', '/Applications/mu-editor.app/Contents/Resources/app_packages', '/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']
2018-07-29 12:17:47,531 - mu.logic:540(__init__) INFO: Setting up editor.
2018-07-29 12:17:47,531 - mu.logic:556(__init__) DEBUG: Creating directory: /Users/peter/Library/Application Support/mu
2018-07-29 12:17:47,531 - mu.logic:313(get_admin_file_path) DEBUG: Creating admin file: /Users/peter/Library/Application Support/mu/settings.json
2018-07-29 12:17:47,531 - mu.logic:558(__init__) INFO: Settings path: /Users/peter/Library/Application Support/mu/settings.json
2018-07-29 12:17:47,532 - mu.logic:313(get_admin_file_path) DEBUG: Creating admin file: /Users/peter/Library/Application Support/mu/session.json
2018-07-29 12:17:47,532 - mu.logic:559(__init__) INFO: Session path: /Users/peter/Library/Application Support/mu/session.json
2018-07-29 12:17:47,532 - mu.logic:560(__init__) INFO: Log directory: /Users/peter/Library/Logs/mu
2018-07-29 12:17:47,532 - mu.logic:561(__init__) INFO: Data directory: /Users/peter/Library/Application Support/mu
2018-07-29 12:17:47,539 - mu.logic:574(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2018-07-29 12:17:47,539 - mu.logic:578(setup) DEBUG: Creating directory: /Users/peter/mu_code
2018-07-29 12:17:47,539 - mu.logic:586(setup) DEBUG: Creating directory: /Users/peter/mu_code/images
2018-07-29 12:17:47,540 - mu.logic:593(setup) DEBUG: Creating directory: /Users/peter/mu_code/fonts
2018-07-29 12:17:47,540 - mu.logic:596(setup) DEBUG: Creating directory: /Users/peter/mu_code/sounds
2018-07-29 12:17:47,541 - mu.logic:601(setup) DEBUG: Creating directory: /Users/peter/mu_code/music
2018-07-29 12:17:47,663 - mu.logic:1110(change_mode) INFO: Workspace directory: /Users/peter/mu_code
2018-07-29 12:17:47,672 - mu.logic:623(restore_session) INFO: Restoring session from: /Users/peter/Library/Application Support/mu/session.json
2018-07-29 12:17:47,672 - mu.logic:624(restore_session) DEBUG: {}
2018-07-29 12:17:47,672 - mu.logic:1067(select_mode) INFO: Showing available modes: ['python', 'adafruit', 'microbit', 'debugger', 'pygamezero']
2018-07-29 12:18:07,197 - mu.logic:671(restore_session) INFO: Starting with blank file.
2018-07-29 12:18:07,207 - mu.logic:1110(change_mode) INFO: Workspace directory: /Users/peter/mu_code
2018-07-29 12:19:10,871 - mu.interface.main:244(get_save_path) DEBUG: Getting save path: /Users/peter/Desktop/spam.py
2018-07-29 12:19:10,871 - mu.logic:874(check_for_shadow_module) INFO: Checking path "/Users/peter/Desktop/spam.py" for shadow module.
2018-07-29 12:19:10,871 - mu.logic:840(save_tab_to_file) INFO: Saving script to: /Users/peter/Desktop/spam.py
2018-07-29 12:19:10,871 - mu.logic:841(save_tab_to_file) DEBUG: print(f'This is a surprise {1}')
2018-07-29 12:19:10,880 - mu.modes.python3:191(run_script) DEBUG: print(f'This is a surprise {1}')
2018-07-29 12:19:10,886 - mu.interface.panes:642(start_process) INFO: Running script: /Users/peter/Desktop/spam.py
2018-07-29 12:19:10,886 - mu.interface.panes:644(start_process) INFO: Running with interactive mode.
2018-07-29 12:19:10,886 - mu.interface.panes:647(start_process) INFO: Command args: []
2018-07-29 12:19:10,886 - mu.interface.panes:665(start_process) INFO: Working directory: /Users/peter/mu_code
2018-07-29 12:19:10,887 - mu.interface.panes:670(start_process) INFO: Python path: ['/Applications/mu-editor.app/Contents/MacOS', '/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/mu-editor.app/Contents/Resources/app', '/Applications/mu-editor.app/Contents/Resources/app_packages', '/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']
2018-07-29 12:19:21,416 - mu.modes.python3:207(stop_script) DEBUG: Stopping script.
2018-07-29 12:19:24,324 - mu.modes.python3:269(toggle_plotter) INFO: Toggle plotter on.
2018-07-29 12:19:24,339 - mu.modes.python3:280(add_plotter) INFO: Started plotter
2018-07-29 12:19:27,113 - mu.modes.python3:229(toggle_repl) INFO: Toggle REPL on.
2018-07-29 12:19:27,115 - mu.modes.python3:58(start_kernel) INFO: ['/Applications/mu-editor.app/Contents/MacOS', '/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/mu-editor.app/Contents/Resources/app', '/Applications/mu-editor.app/Contents/Resources/app_packages', '/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']
2018-07-29 12:19:27,116 - mu.modes.python3:63(start_kernel) INFO: Starting iPython kernel with user defined envars: {}
2018-07-29 12:19:27,224 - traitlets:257(start_kernel) DEBUG: Starting kernel: ['/usr/local/opt/python/bin/python3.7', '-m', 'ipykernel_launcher', '-f', '/var/folders/qs/06qdgbms3x195gjt304531840000gp/T/tmptpfa4mcd.json']
2018-07-29 12:19:27,241 - traitlets:542(_create_connected_socket) DEBUG: Connecting to: tcp://127.0.0.1:64171
2018-07-29 12:19:27,254 - asyncio:65(__init__) DEBUG: Using selector: KqueueSelector
2018-07-29 12:19:27,255 - asyncio:65(__init__) DEBUG: Using selector: KqueueSelector
2018-07-29 12:19:27,255 - traitlets:140(shell_channel) DEBUG: connecting shell channel to tcp://127.0.0.1:64168
2018-07-29 12:19:27,255 - traitlets:542(_create_connected_socket) DEBUG: Connecting to: tcp://127.0.0.1:64168
2018-07-29 12:19:27,256 - traitlets:152(iopub_channel) DEBUG: connecting iopub channel to tcp://127.0.0.1:64169
2018-07-29 12:19:27,256 - traitlets:542(_create_connected_socket) DEBUG: Connecting to: tcp://127.0.0.1:64169
2018-07-29 12:19:27,257 - traitlets:164(stdin_channel) DEBUG: connecting stdin channel to tcp://127.0.0.1:64170
2018-07-29 12:19:27,257 - traitlets:542(_create_connected_socket) DEBUG: Connecting to: tcp://127.0.0.1:64170
2018-07-29 12:19:27,257 - traitlets:176(hb_channel) DEBUG: connecting heartbeat channel to tcp://127.0.0.1:64172
2018-07-29 12:19:30,091 - traitlets:110(poll) INFO: KernelRestarter: restarting kernel (1/5), new random ports
2018-07-29 12:19:30,092 - traitlets:539(_handle_kernel_restarted) WARNING: kernel restarted
2018-07-29 12:19:30,098 - traitlets:257(start_kernel) DEBUG: Starting kernel: ['/usr/local/opt/python/bin/python3.7', '-m', 'ipykernel_launcher', '-f', '/var/folders/qs/06qdgbms3x195gjt304531840000gp/T/tmptpfa4mcd.json']
2018-07-29 12:19:30,112 - traitlets:542(_create_connected_socket) DEBUG: Connecting to: tcp://127.0.0.1:64273
2018-07-29 12:19:30,262 - traitlets:527(_handle_kernel_died) WARNING: kernel died: 3.0041260719299316
2018-07-29 12:19:33,054 - traitlets:110(poll) INFO: KernelRestarter: restarting kernel (2/5), new random ports
2018-07-29 12:19:33,055 - traitlets:539(_handle_kernel_restarted) WARNING: kernel restarted
[... cut ...]
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x134), Broadcom BCM43xx 1.0 (7.21.171.133.1a2)
Bluetooth: Version 5.0.5f3, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0512F, 500.28 GB
USB Device: USB 3.0 Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.1
```

ntoll, 2018-07-29:
Wow, that's odd and I've never seen that before. Interesting!

It looks like iPython is doing something funky to work out which version of Python3 to use for starting the kernel. Since Mu is bundled with Python 3.6 and is itself running in "isolated" mode my guess is that when Python3.7 attempts to start, something (no idea what) doesn't line up between the versions and it breaks.

What should be happening is the REPL in Mu should be using the version of Python3 bundled with Mu. I think it's related to #548 and Thomas (from project Jupyter) has suggested a fix that I'll make sure gets into the next version.

By the looks of the crash report, I see: `_Py_FatalInitError` which IIRC was happening in some pre-release versions of Python 3.7 (are you using the final released version?). I'm clutching at straws here, but it could be something to do with: https://bugs.python.org/issue33932 (and the version of iPython in Mu is causing this in some unknown-to-me way).

Not sure -- and that looks like quite a deep rabbit hole. ;-)

I think the simplest solution would be to fix #548 so Mu forces iPython to use the version of Python being used to run Mu itself.

PeterWAWood, 2018-07-30:
Thanks for looking in to this so quickly. I install Python3 via Homebrew. When Homebrew updated the installed Python 3.6 to 3.7, I did experience some issues with Jupiter which I fixed by manually updating some configuration files. This maybe contributing to the REPLs confusion.

When I get time, I'll perform a clean install of Jupiter and see if that fixes the problem.

I think that implementing #548 would be a wise move.

carlosperate, 2021-03-24:
@PeterWAWood the way the iPython REPL is configured has been updated in the latest beta release. Could you try this with the latest version and see if it is still a problem? https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.2

PeterWAWood, 2021-03-25:
@carlosperate I installed the Beta and the REPL opened with no problem

carlosperate, 2021-03-25:
That's great, thanks for the report and testing it again!

closed 2021-03-25 by carlosperate