mu-editor/mu #589: Issue launching installed Mu in macOS

carlosperate, 2018-07-29:
From a user at the sprints using macOS. 
Installing the app went well, but then when it was double clicked it closed unexpectedly. No window appeared, but you could see the icon quickly appearing and disappearing in the apps doc.

By executing the launch script directly this was the traceback:

```
Traceback (most recent call last):
  File "/Applications/mu-editor.app/Contents/MacOS/mu-editor", line 15, in <module>
    from pkg_resources import load_entry_point
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/site-packages/pkg_resources/__init__.py", line 3088, in <module>
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/site-packages/pkg_resources/__init__.py", line 3072, in _call_aside
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/site-packages/pkg_resources/__init__.py", line 3101, in _initialize_master_working_set
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/site-packages/pkg_resources/__init__.py", line 574, in _build_master
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/site-packages/pkg_resources/__init__.py", line 892, in require
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/site-packages/pkg_resources/__init__.py", line 778, in resolve
pkg_resources.DistributionNotFound: The 'pytest; python_version >= "3.4" or python_version == "2.7" and extra == "test"' distribution was not found and is required by jupyter-client
```

tmontes, 2019-06-01:
Given:
* This issue's age,
* The level of detail we have on the failure (not much),
* The fact that since then macOS packaging has been revised at least once, but maybe more

...I'm left wondering if this issue is actionable. If so, what would the "next action" be? If not, hopefully the user that faced it has overcome it, and this can be closed.

PS: Yes, I'm trying to contribute to some kind of "issue triaging" thing here. :)

carlosperate, 2019-06-01:
We can close this.

closed 2019-06-01 by carlosperate