mu-editor/mu #591: Feature request/idea: show how often each line of code was executed

askvictor, 2018-07-29:
An idea I had some time ago while teaching Python was that it would be great for students to be able to see how often a particular line of code ran. This would allow students to better see what happened after their code ran (or even during a debug session). For example, it would let them see which branch of an if statement executed, or how often a particular loop looped. It is conceptually similar to coverage.py (I don't have much experience with that so don't know if it measures the count of each line executed, or just a yes/no). It might be represented by a heat-map type scale on the side of the window, perhaps with hover-text to show the actual number for that line.

No idea how feasible this is, nor if it would complicate the editor too much, but it would make a useful visual for learning.

btw - great work on the latest version!

tjguk, 2018-07-30:
I have no particular view of the usefulness of this but, technically, it could be supported by using the built-in trace module

dybber, 2021-02-25:
I will convert this to a discussion in our new discussion board here at Github, as that's where we discuss feature requests from now on.

closed 2021-02-25 by dybber