mu-editor/mu #592: Mu not appearing when you type mu into Windows 10 menu

askvictor, 2018-07-29:
I'm not certain this has to do with mu or with Windows (or my particular instance of Windows). After installing mu, it appears fine in the Windows menu if I scroll down to 'M'. However, a common way of launching Windows applications is to hit the Windows button (on screen or on keyboard), then start typing the name of the application you want to run (similar to spotlight on OSX). When I do this (press Windows button then immediately type 'mu'), the application 'mu' does not appear in the results (another application I have installed, MuseScore, is the one that shows first, followed by Groove Music). This might cause confusion to some users. Could some others running Windows please try to confirm this? I don't know much about Windows development - perhaps each application needs to register what keywords it should appear for in Windows application searches? Perhaps there is a lower limit on the number of characters an app name automatically registers with in the index?

ZanderBrown, 2018-07-30:
If you've been using MuseScore a lot it will take precedence over Mu (although Mu should still show in the other results) not sure there is much we can do about this

askvictor, 2018-07-30:
Haven't been using musescore at all, so it's not likely a precedence thing.

lawsie, 2018-08-21:
I'm using Windows 10 and it works fine for me (I don't have Musescore)
[image]

carlosperate, 2018-08-21:
The windows 10 search is a HUGE hit or miss for me, which is a shame, because previous versions of Windows were really good in this area (or at least they always behaved how I expected). I don't know how the results are personalised, but the % error I get is high enough that it's made this functionality pretty much useless for me.
</rant over>

Maybe there is some metadata we can pack with Mu to help these search results, but this is one of those things were simple google searches don't really give out any good results, and you probably need to know what to look for to find it.

@tjguk is a Windows expert, perhaps you have some thoughts or tips in this area?

askvictor, 2018-08-23:
I just tried it on a relatively fresh install of Windows, and it now _does_ appear in the Windows menu (both for all-users and just-me installs). So maybe it was just a quirk of the system I first tried it on.

ntoll, 2018-09-07:
Closing since this appears to be a local quirk. Thanks for all the input / checking folks! :-)

closed 2018-09-07 by ntoll