mu-editor/mu #595: Renames the filename by 'itself'

marcovanurk, 2018-07-30:
If I have two files open in mu-editor. During editing both files, with different file names, after a while both files have the same name in the tab.

I can not say exactly when. It is during editing both files and loading to micro:bit.

marcovanurk, 2018-07-30:
By the way, to be complete: it is when using mu-editor in 'BBC micro:bit' mode.

dybber, 2020-06-11:
My best guess is that this is due to a race condition, where a `QsciScintilla.modificationChanged` event from the EditorPane is fired from one tab while the focus has already changed to another tab, and the handler then updates the name of the wrong tab:
https://github.com/mu-editor/mu/blob/1f14fb753f9a46bd191efb1f18966864100ae355/mu/interface/main.py#L445-L451

Instead of calling `self.tabs.currentIndex()`, we should look through the tabs to find the index of `new_tab`, before calling `setTabText`

dybber, 2020-06-11:
Yeah, that's it. Here's how to reproduce the issue:

1. Open two tabs with two files (can't just be untitled)
2. Edit a bit in one of them, and switch to the other (without saving the first!)
3. Wait until Mu autosaves the first file
4. The autosave will provoke a modificationChanged event, which will change the name of the second tab

tjguk, 2020-06-12:
Fixed by #1063 -- thanks @dybber

closed 2020-06-12 by tjguk