mu-editor/mu #598: Turkish Translation

osaatcioglu, 2018-07-30:
I like to congratulate all the contributors for such a beginner friendly IDE. I am occasionally helping beginners to get in to the development using Python and Mu Editor looks like the best available option. Kudos! :)

I started to do the Turkish translation. Please put this issue on me. And, if there are any other Turkish fellows, I will be happy to collaborate together.

ntoll, 2018-09-07:
Hi @osaatcioglu. Thank you for your offer of help with the Turkish translation. :-)

osaatcioglu, 2018-09-11:
Hi @ntoll, I am very happy to help. Unfortunately, it is taking a bit more time than I anticipated though. I am 60% complete and I hope that I can finish the first version by end of next week. :)

libreajans, 2019-08-25:
Hi @osaatcioglu 

Hazırladığınız çevirileri son haliyle paylaşabilir misiniz? Eksik kalan kısımları topluluk olarak tamamlamaya çalışırız.

carlosperate, 2021-12-27:
Hi @osaatcioglu and @libreajans, it's been a while, but if you are still happy to provide translations the Mu project is finally ready  🎉 

More info in:
- https://github.com/mu-editor/mu/discussions/1931

carlosperate, 2022-03-02:
Hi @osaatcioglu and @libreajans, we'll synchronise translations in this discussion:
- https://github.com/mu-editor/mu/discussions/1931

To keep things tidy I'll close this issue, but if you can still provide a Turkish translation it would be really appreciated!

closed 2022-03-02 by carlosperate