mu-editor/mu #604: Problems installing mu-editor on various models of rpi

IanHun, 2018-08-01:
I have been installing mu-editor on rpi0w, rpi2, rpi3, rpi3b+.  

No problem on the 3 and 3b+, where I used dist-upgrade - but this failed on 0w and 2.

Did a fresh install on 0w and 2, and then tried to install mu-editor - got the result "Temporary failure resolving 'raspian.raspberrypi.org'".

Does anyone know what's going on, please?

tjguk, 2018-08-01:
That looks more like a (possibly transient) issue with the Pi itself or its configuration more than with the Mu installation. You don't say how you're installing Mu but I assume that you're using apt install? If so, can you apt install anything else? Or apt update? Those should both use the same raspbian.raspberrypi.org connection.

IanHun, 2018-08-01:
You’re right - I get the same response trying to install other things (via both apt and apt-get), and I can’t apt update either.  Clearly nothing to do with mu-editor.

Any thoughts what might be causing it?

(I'll  close this thread)

closed 2018-08-01 by IanHun

IanHun, 2018-08-01:
Tim

Many thanks

Ian

tjguk, 2018-08-01:
@IanHun You're welcome. (I have no idea what the underlying cause might be I'm afraid; you'd better try the RPi forums for that).