mu-editor/mu #610: Accidental python2 pip install, doesn't give good message.

illume, 2018-08-08:
* What you were trying to do,
A user was trying to do this:
`pip install mu-editor --user`

Except pip was their python2 version.

* What steps you took to make this happen,
* What you expected to happen,
* What actually happened,

```
$ pip install mu-editor --user
Collecting mu-editor
  Using cached https://files.pythonhosted.org/packages/67/d4/3ea00caa61d214044db8fddb868e903019f9e866593933aa39a2c11f8514/mu-editor-1.0.0.tar.gz
Collecting pycodestyle==2.4.0 (from mu-editor)
  Using cached https://files.pythonhosted.org/packages/e5/c6/ce130213489969aa58610042dff1d908c25c731c9575af6935c2dfad03aa/pycodestyle-2.4.0-py2.py3-none-any.whl
Collecting pyflakes==2.0.0 (from mu-editor)
  Using cached https://files.pythonhosted.org/packages/44/98/af7a72c9a543b1487d92813c648cb9b9adfbc96faef5455d60f4439aa99b/pyflakes-2.0.0-py2.py3-none-any.whl
Collecting pyserial==3.4 (from mu-editor)
  Using cached https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl
Collecting pyqt5==5.11.2 (from mu-editor)
  Could not find a version that satisfies the requirement pyqt5==5.11.2 (from mu-editor) (from versions: )
No matching distribution found for pyqt5==5.11.2 (from mu-editor)
```

* Why this difference is problematic (it may not be a bug!),

It doesn't say to use python3, it says pyqt isn't there. So we almost gave up installing pip after some research into how to get pyqt5 5.11.2 installed.

I'm not actually sure how to fix this entirely yet. It might require a pip fix for binary wheels. pip should really see this and give a nice message:
```
'Programming Language :: Python :: 3 :: Only',
```

Also the binary wheel is this with the 'py3' in it. So again, pip should know.
```
 mu_editor-1.0.0-py3-none-any.whl (1.7 MB) 
```

However, in the setup.py file you can do something for source installs.
In pygame we have this in the setup.py:
```
if not hasattr(sys, 'version_info') or sys.version_info < (2,7):
    compilation_help()
    raise SystemExit("Pygame requires Python version 2.7 or above.")
```

* A work around could be to remove the wheel file until pip is fixed (but I guess there might be some downsides).
* I'm still researching if pip can already help with this somehow, so haven't made a pip issue yet. But I guess that would be the next step. Or perhaps ask on that packaging mailing list.
* Also, I can make a PR for the setup.py file version check if you(s) want that.

ntoll, 2018-08-08:
@illume quickly (I'm on holiday right now, so sneaking some computer time in this morning)... ;-)

Yes. What you say. Absolutely. Especially since Mu is supposed to be beginner friendly, we need to make sure Mu fails in a friendly / helpful way.

I agree with your assessment that this sort of thing should probably happen in `pip`, but since it doesn't, your proposed solution LGTM.

In the case of Mu, I guess it should say something like:

> Mu only works with Python version 3.5 or above. For more information see: https://codewith.mu/en/howto/install_with_python

Does this look right? Thoughts?

illume, 2018-08-08:
Holiday! Sorry for the interruption :) Have a cherry day.

closed 2018-09-29 by ntoll