mu-editor/mu #612: Mu Debugger Failes if numpy is importet

jrast, 2018-08-15:
The import of numpy failes if the file is debugged instead of run.

Content of Testfile:

````testfile.py
import numpy as np
print(np.arange(10))
````

If i run this file by clicking "Run", I get the expected output:
```
[0 1 2 3 4 5 6 7 8 9]
````

But if I run the file by clicking "Debug", followed by "Continue", I get the following output:
````
exception: Traceback (most recent call last):
  File "D:\MuEditor\pkgs\mu\debugger\runner.py", line 494, in run
    debugger._runscript(filename)
  File "D:\MuEditor\pkgs\mu\debugger\runner.py", line 469, in _runscript
    self.run(e)
  File "bdb.py", line 431, in run
  File "<string>", line 1, in <module>
  File "c:\users\jrast\mu_code\test1.py", line 1, in <module>
    import numpy as np
  File "D:\MuEditor\pkgs\numpy\__init__.py", line 142, in <module>
    from . import add_newdocs
  File "D:\MuEditor\pkgs\numpy\add_newdocs.py", line 13, in <module>
    from numpy.lib import add_newdoc
  File "D:\MuEditor\pkgs\numpy\lib\__init__.py", line 24, in <module>
    from .financial import *
  File "D:\MuEditor\pkgs\numpy\lib\financial.py", line 15, in <module>
    from decimal import Decimal
ModuleNotFoundError: No module named 'decimal'

---------- FINISHED ----------
exit code: 0 status: 0
````

Technical Information:
Mu: 1.0.0, fresh install (64bit)
Operating System: Windows 7, 64bit

Complete Log: [mu.log](https://github.com/mu-editor/mu/files/2290433/mu.log)

carlosperate, 2018-08-16:
Hi @jrast, thank you for the bug report. I wasn't able to replicate, this in macOS. I'll see if next time I'm using Windows I remember to try to replicate there.

In the mean-time, would it be possible you have a `decimal.py` file in the same directory than your test file? or in the mu_code directory?
If there is a file named like that in any of the directories that Python looks for modules to import it's possible it's trying to import that one instead of the one provided by numpy.

tim-mccurrach, 2018-08-16:
Not a solution I'm afraid, but perhaps this might point someone who knows more than me in the right direction. I ran the code (on windows 10) and it also works fine, when just run, but get the same error when run with debug mode. Interestingly though, it works absolutely fine when run from my development version on the same machine. 

I think `decimal.py` is probably referring to the module in the standard library (although I don't know numpy well so could be wrong there).

I ran the following code, both running as normal and in debug mode, on my installed version of mu and got the following results:
```
import sys
for i in sys.path:
    print(i)
```
**Normal run**
```
C:\Users\timmc\AppData\Local\Mu\Python\python36.zip
C:\Users\timmc\AppData\Local\Mu\Python
C:\Users\timmc\AppData\Local\Mu\pkgs

```
**Debug mode**
```
c:\users\timmc\documents\python
C:\Users\timmc\AppData\Local\Mu\Python
C:\Users\timmc\AppData\Local\Mu\pkgs
C:\Users\timmc\AppData\Local\Mu\pkgs\IPython\extensions
```
The module `decimal.py` can be found in the python36.zip directory, which would explain why it runs normally, but not in debug mode. Indeed just `import decimal` produces the same differing results. 

I don't know how to fix the error, but I hope this helps :)

carlosperate, 2018-08-20:
This is a bit odd, Mu should be spawning the new process the same way when "running" or "debugging". There is an exception already for adding `sys.path` on the macOS bundled package, perhaps this needs to be extended to windows as well.

In contrast this is what I get when running Mu in macOS from the installer:

```
Run:
/Users/microbit-carlos/mu_code
/Users/microbit-carlos/Downloads/micropython/mu-editor.app/Contents/MacOS
/Users/microbit-carlos/Downloads/micropython/mu-editor.app/Contents/Resources/python/lib/python36.zip
/Users/microbit-carlos/Downloads/micropython/mu-editor.app/Contents/Resources/python/lib/python3.6
/Users/microbit-carlos/Downloads/micropython/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload
/Users/microbit-carlos/Downloads/micropython/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages
/Users/microbit-carlos/Downloads/micropython/mu-editor.app/Contents/Resources/app
/Users/microbit-carlos/Downloads/micropython/mu-editor.app/Contents/Resources/app_packages
/Users/microbit-carlos/Downloads/micropython/mu-editor.app/Contents/Resources/app_packages/IPython/extensions

Debug:
/Users/microbit-carlos/mu_code
/Users/microbit-carlos/Downloads/micropython/mu-editor.app/Contents/MacOS
/Users/microbit-carlos/Downloads/micropython/mu-editor.app/Contents/Resources/python/lib/python36.zip
/Users/microbit-carlos/Downloads/micropython/mu-editor.app/Contents/Resources/python/lib/python3.6
/Users/microbit-carlos/Downloads/micropython/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload
/Users/microbit-carlos/Downloads/micropython/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages
/Users/microbit-carlos/Downloads/micropython/mu-editor.app/Contents/Resources/app
/Users/microbit-carlos/Downloads/micropython/mu-editor.app/Contents/Resources/app_packages
/Users/microbit-carlos/Downloads/micropython/mu-editor.app/Contents/Resources/app_packages/IPython/extensions
/Users/microbit-carlos/Downloads/micropython/mu-editor.app/Contents/Resources/app_packages/IPython/extensions
```

tmontes, 2019-06-01:
Dear all, and @jrast in particular,

This issue is quite old by now and I suppose may have been fixed with PR #726, included in Mu 1.0.2 released back in January. Can you please confirm that is the case and, if so, that this issue may be closed?

Thanks.

carlosperate, 2020-09-29:
As it has been over a year and  #726 has very likely fixed this problem we can close this issue.
Thank you @jrast and @tim-mccurrach for the reports! If you are still experiencing this problem please do reopen this issue.

closed 2020-09-29 by carlosperate

carlosperate, 2020-09-29:
I just gave this a test and can confirm it works on Windows.