mu-editor/mu #614: Run panel does not remember its position/size

malbarbo, 2018-08-15:
To reproduce run some code. Then resize (or attach to left of right) the run console panel. Then stop the running code. The next time the code is run the panel show in the original position. I would expect that the position to be remembered and the panel opens in the last position.

carlosperate, 2018-08-16:
We had some discussion about this area in this thread: https://github.com/mu-editor/mu/issues/553#issuecomment-407794074

I think at the moment we are inclined to save zoom state, but not the panel size/position.

malbarbo, 2018-09-18:
I see. The issue here is bad when the screen monitor has more horizontal than vertical space. Think a 16:9 screen with top an bottom panel, splitting the mu editor vertically leaves the editor and run console too small, splitting horizontally temporally fix the problem, but it appears when the code in run again.

ntoll, 2018-12-18:
Fixed by Fixed by  5b82f22

closed 2018-12-18 by ntoll