mu-editor/mu #622: Pressing enter in the middle of an expression in the run panel causes trouble

malbarbo, 2018-08-21:
After pressing the Run button the executing panel opens and show the python prompt (>>>).

If one type 2 * 3 and press enter with the cursor after * the expression is break in two and 2 * is evaluated generating a syntax error. Pressing enter again evaluate 3 (the second part of the expression) which is confusing, because there is no indication that 3 is the next expression.

If the same thing is done the REPL (Jupyter), it evaluates the whole 2 * 3 expression.

I'm using mu 1.0.0 in a Debian machine.

tim-mccurrach, 2018-08-25:
I’m not 100% sure what the expected behaviour should be here. Experimenting with IDLE it seems that when you press enter in the middle of a line:
a) if there is already a new line below the current text, the text on the current line to the right of the cursor moves to the new line
b) if there is nothing below the current text, the current line is not changed at all, but the cursor effectively jumps to the end of the line and then continues as per normal (either evaluating the expression, or going to a new line). 

To achieve b), we could swap round the order of inserting the new msg, and sending the current line to the interpreter. This seems to work for me:
```
        # last part of parse_input (line 837 onwards)
        if key == Qt.Key_Enter or key == Qt.Key_Return:
            content = self.toPlainText()
            line = content[self.start_of_current_line:].encode('utf-8')+msg
            self.write_to_stdin(line)
            if line.strip():
                self.input_history.append(line.replace(b'\n', b''))
            self.history_position = 0
            cursor = self.textCursor()
            cursor.movePosition(QTextCursor.End)
            self.setTextCursor(cursor)
            self.insert(msg)
            self.start_of_current_line = self.textCursor().position()
        elif not self.isReadOnly() and msg:
            self.insert(msg)
```
To achieve a) seems a lot more difficult than b). Currently mu sends each line to the interpreter after it has been entered, which will then return either the result plus >>> or … as appropriate. 

As far as I understand it, if we wanted to achieve a) we would need to (amongst a few other things) only send the code to the interpreter once a complete expression or statement was ready to be sent. This would in mean when the user presses enter, mu would need to figure out some other way if a line continuation is needed. 

The jupyter REPL does achieve a) but I'm not sure how it handles it atm. Does anyone have any thoughts on how this could be best achieved?

malbarbo, 2018-09-18:
Can we use Jupyter as the only REPL?

carlosperate, 2020-10-26:
Assume you originally meant the ESP mode this should now be fixed by https://github.com/mu-editor/mu/pull/1026.

Please feel free to re-open if the problem you meant is still present in the master branch.

closed 2020-10-26 by carlosperate