mu-editor/mu #624: Consider dropping hard-coded dependency on Source Code Pro font

knowledgejunkie, 2018-08-22:
Whilst distributed under a Debian Free Software Guidelines-compatible license, the bundled Source Code Pro font files are not yet themselves packaged for Debian ([1],[2])

  [1] https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=762252
  [2] https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=736681

Per Debian policy 11.8.5 [3], fonts need to be provided in a separate binary package. Due to the absence of Source Code Pro in the Debian repositories, I've replaced references to it with Inconsolata (which is packaged, additionally see #83) in my initial Debian packaging of Mu. Additionally, the source has been repacked to remove the font files from the proposed Debian source tarball.

  [3] https://www.debian.org/doc/debian-policy/ch-customized-programs.html#packages-providing-fonts

Whilst this specific example applies to Debian, it may also apply to other free software distributions wishing to package Mu.

Is it worth considering removing the hard-coded dependency on Source Code Pro and allowing for the use of e.g. alternative system fonts automatically?

ZanderBrown, 2018-08-22:
This is hard because we want to maintain cross-platform consistency (We _deliberately_ avoid native integrations in some areas) so students aren't put off by things being different.

At the same time we want to be in Debian! Currently we successfully bundle the fonts for Windows, macOS & Fedora making Debian (thus far) the odd one out. Perhaps Debian can package Source Code Pro or find a similar font that _is_ available on Debian and start bundling that instead?

dybber, 2020-09-05:
Is this still a problem? It seems someone submitted a package for the font in the second of the two links, but I'm not sure if it has been integrated with Debian yet?

carlosperate, 2020-09-29:
@knowledgejunkie is this is still an issue? Is Source Code Pro now in the Debian repos?

knowledgejunkie, 2020-10-07:
The package (fonts-source-code-pro) mentioned by @dybber and linked to from the Debian RFP [1] is **not** part of Debian, and does not build the SCP fonts from source.

Packaging of this font from source is contingent on the availability of the Adobe Font Development Kit for OpenType. This software (afdko) was accepted into Debian in July 2020. The fonts-source-code-pro package linked in [1] was created almost a year beforehand, simply bundles the precompiled fonts, and therefore needs to be reworked to take advantage of the availability of afdko in order to build the font from source and make it acceptable for inclusion in Debian/Ubuntu.

  [1] https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=736681

carlosperate, 2020-10-26:
To be fair, if this simplifies packaging I wouldn't mind changing the font for something similar.

@knowledgejunkie is there an easy way to see what fonts are available in the Debian official repos? If there was an easy way to get a list of good programming fonts from the bunch that would be even better.

[image]

[image]

knowledgejunkie, 2020-10-27:
@carlosperate The underlying issue is not that the SCP font is not packaged for Debian, but that bundled fonts (and other compiled assets) are removed from upstream sources when packaging for Debian. Such assets are then linked to using the binary package(s) for those specific assets, if available in Debian.

Thus, swapping SCP for Inconsolata **will** reduce the amount of patching required, but won't remove the requirement to exclude it from the source tarball.

carlosperate, 2020-10-27:
Yeah, I understand that, but I assume you had to patch the Mu source code to set a different font family?
So by swapping to a font available via deb package then it should save you from patching the Mu source code, no?
And if that configuration is contained in the https://github.com/mu-editor/mu/tree/master/debian folder, then I assume that no patch would be required at all for the fonts?

That was my reasoning, as I'd rather reduce the number of patches required for debian packaging.

knowledgejunkie, 2020-10-28:
Switching to e.g. Inconsolata would reduce the amount of patching required for Mu a little, but honestly it's not onerous. There's also a chance that Source Code Pro will be packaged in Debian and Ubuntu now that the Adobe font toolkit is packaged. I/we will work with whatever Mu decides as a project.

Please note that an upstream's debian/ folder is ignored during packaging for Debian/Ubuntu.

ZanderBrown, 2020-10-29:
Generally we have tried to be as common as possible across platforms so we'd probably want to keep SCP (if viable)

carlosperate, 2020-11-04:
If Source Code Pro will end up being packaged in Debian/Ubuntu then it's probably not worth the change really. 

> Please note that an upstream's debian/ folder is ignored during packaging for Debian/Ubuntu.

Anything worth pushing to this repo to reduce the patching? If there are any tests we could add to the CI to reduce breaking changes to the Debian packaging that could be beneficial as well?