mu-editor/mu #625: Setting permenant alternative 'save directories' for multiple users

G1llmatthews, 2018-08-24:
During the installation of Mu it currently sets a default path to save work on the local hard drive of the computer. I understand that this is a feature rather than a bug, but the difficulty is that in many (Most probably) secondary schools, student work is saved on a networked drive not on the local hard drive.  This happens because it common for students to have to access their work on different computers across the school, as well as supporting simpler back up procedures on large networks.  The problem is compounded by the fact that faults in the computer systems are routinely dealt with by rebuilding the machines, losing any user data that is stored on them. Of course the students could change the directory every time they create a new file, but some secondary students are much better than primary when it comes to remembering to save their files in the correct locations.

I am currently attempting to resolve this issue by creating a .bat file to edit settings.json that could be run by the students but have the problem that settings.json does not seem to interpret environment variables as expected.  

In the longer term it would be helpful to have the option to select the default directory path at installation time.

I am currently using Mu 1.0.0 version in mixed Windows 7 Window 10 environment with windows 2016 servers

tim-mccurrach, 2018-08-28:
Definitely +1 for this request. The (secondary school) students I teach are all bright, enthusiastic, intelligent young people, but their ability to completely ignore a very simple instruction like "make sure you save your work in your 'my documents' folder" would amaze you. And it can be a real pain in the neck too, because different machines will not work each lesson, and so the same students may not be on the same machines each lesson, and so have to start work again. So this would make a massive difference.

For what its worth, I was in school today, so asked the computer technicians about the problem, and he said what should happen is that since students won't have access to the C drive, when they click save it should automatically re-route them to their my documents folder. So hopefully it won't be too much of a problem, but obviously every school is different and this might not be the case in other schools. 

@G1llmatthews If you can get it working, the technicians at your school might be able to get the .bat file to run automatically on each machine or user-account, so might be worth talking to them and asking. But if it is somehow possible for this to be set during installation, that would be brilliant, as in my experience, IT technicians in school work really hard and people hardly ever thank them, and so anything to make their jobs easier would be a plus!!

ZanderBrown, 2018-08-28:
Does the [workspace](https://codewith.mu/en/tutorials/1.0/configuration) setting not do what you need or is it too hidden?

tim-mccurrach, 2018-08-28:
In my particular case, students won't be able to change the settings.json file, just because of the permission settings on their accounts. I think (on windows) the AppData folder which it is stored in will be hidden, and again students won't have the ability to change their settings to 'show hidden files'.

I think it will vary quite a lot depending on how the different school networks are set up though.

G1llmatthews, 2018-08-29:
In my network AppData directory is hidden but has User write access because that is where things like tmp and temp are situated. I would be surprised if it was different on other school networks.  My solution to the problem has been to create a .bat file which uses xcopy to create an appropriately named folder in the students networked drive and update settings.json to point to it.  In theory the students could do this for themselves. In practice a) AppData is hidden for a reason and b) a lot of them do not spell well enough to type a file path for themselves.  I have placed the batch file and associated gubbins in a Directory on a public drive that allows execute. I have tested it with test users and it will simply require them to double click on the file once the first time they use a particular work station or if a workstation has been reimaged.  It is slightly clunkier than being able to set a default directory during the install process, but not unmanageably clunky compared with getting them to update the settings themselves or hoping that they will always remember to save their files in the correct place.

tim-mccurrach, 2018-08-30:
@ZanderBrown Yes, a very good point. Here's hoping it's possible to do without too much trouble :) 

@G1llmatthews Yes, I hadn't thought that through properly (that it would need to be for each user/computer combination). And can appreciate that.