mu-editor/mu #626: `pip install mu-editor` fails with missing dependency for PyQtChart

wuffi, 2018-08-24:
Trying to run `pip install mu-editor` fails on different computers with different Linux versions for me due to for PyQtChart.
I have a lot of trouble Installing the required version for this package turns out to be difficult on Ubuntu 18, although I installed all required (distribution) packages in advance.
I also tried installing the mu-editor in a virtualenv, but it also didn't work.
Is there a simple possibility to install the mu-editor on linux?

ntoll, 2018-08-24:
Hi @wuffi,

Sorry to hear you're having problems. I actually develop Mu on Ubuntu on 18.04 and can assure you that pip works! To be certain we're on the same page... you should create a virtualenv that uses Python 3, types `pip install mu-editor` and then, once installed `mu-editor` to start the application. I've just done these steps again on my Ubuntu 18.04 machine to ensure there's not something odd going on with PyPI, but I can confirm it works perfectly for me.

What architecture are you using? If 32bit/64bit Intel then I know it works for certain and this is something local to you.

It might be helpful if you could attach a copy of the output you see from pip on your machine.

Thanks,

N.

wuffi, 2018-08-24:
Thanks so much for the quick reply.
On a i686 Thinkpad, I run

    virtualenv -p /usr/bin/python3 ~/mu-env
    source ~/mu-env/bin/activate
    pip install mu-editor

Now the error message is
    
     Could not find a version that satisfies the requirement qscintilla

ntoll, 2018-08-24:
@wuffi 

I'm on a ThinkPad X260. ;-)

I just tried your instructions exactly. It works for me. ;-/

I wonder if your local (pip) cache is corrupted in some way. Why not try..?

```
pip install mu-editor --no-cache
```

...and let me know what happens?

wuffi, 2018-08-24:
No, did unfortunately not help.
The error message persists.
But the idea was good.

ntoll, 2018-08-24:
OK... this seems to be local to you since it works for me and I've asked others to check too. Perhaps you're hitting some sort of CDN cache that's stale and/or incomplete (i.e. this is a PyPI problem rather than anything to do with Mu).

wuffi, 2018-08-24:
Then I'll close the issue.
It's helpful to know, that it is in principal working on Ubuntu 18.
Thanks again for your quick help.

ntoll, 2018-08-24:
NP... sorry I can't be of more help.

carlosperate, 2018-08-24:
Out of curiosity, are you running 32 bits Ubuntu? I wonder if perhaps any of the PyQt wheels is misconfigured.
If you try to pip install with the verbose flag it should show you the list of wheels/packages it finds on PyPi and you can see if it finds one for your specific architecture. If you could copy and paste the full output here it would be interesting to see the result.

wuffi, 2018-08-24:
Please find the verbose output of `pip3` below:

`Created temporary directory: /tmp/pip-ephem-wheel-cache-v18nwcna
Created temporary directory: /tmp/pip-req-tracker-fjratcgh
Created requirements tracker '/tmp/pip-req-tracker-fjratcgh'
Created temporary directory: /tmp/pip-install-2fwcw7ji
Requirement already satisfied: mu-editor in /usr/lib/python3/dist-packages (1.0.0)
Requirement already satisfied: pycodestyle==2.4.0 in /usr/local/lib/python3.5/dist-packages (from mu-editor) (2.4.0)
Requirement already satisfied: qtconsole==4.3.1 in /usr/local/lib/python3.5/dist-packages (from mu-editor) (4.3.1)
Requirement already satisfied: pgzero==1.2 in /usr/local/lib/python3.5/dist-packages (from mu-editor) (1.2)
Collecting PyQtChart==5.11.2 (from mu-editor)
  1 location(s) to search for versions of PyQtChart:
  * https://pypi.org/simple/pyqtchart/
  Getting page https://pypi.org/simple/pyqtchart/
  Starting new HTTPS connection (1): pypi.org:443
  https://pypi.org:443 "GET /simple/pyqtchart/ HTTP/1.1" 200 3574
  Analyzing links from page https://pypi.org/simple/pyqtchart/
    Skipping link https://files.pythonhosted.org/packages/8b/73/357511cf1d231d7cbc0464ec283ad9c5a9cfc6e4045b0e8f10ddfec33974/PyQtChart-5.7-cp35-cp35m-macosx_10_6_intel.whl#sha256=a78b30e51257b1286733f66ff64beb717f6a6e4929de794fcc921b7627f6cff4 (from https://pypi.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://files.pythonhosted.org/packages/a0/25/64a49f1c023c3ebddc1e8f3305b062f463e5674b6a1bdaab67dde7b4552d/PyQtChart-5.7-cp35-cp35m-manylinux1_x86_64.whl#sha256=2cd9db5361abb2b032a960b05b2b7eb136e9adea5f104d879a029aaf59fd4f16 (from https://pypi.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://files.pythonhosted.org/packages/73/0b/7fc7aeffce6fb264374f8a345cbbe2967841eceac018376cafe1c1f77d7e/PyQtChart-5.7-cp35-none-win32.whl#sha256=df8db40212646ac9667b6bdcf1296b9b2073d6b2f3ee46df5331bd3a884e3e7d (from https://pypi.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://files.pythonhosted.org/packages/d9/40/9fcf332dae9f8c72b8355c54eceb0af6661179c53bb9f80604651b8b708b/PyQtChart-5.7-cp35-none-win_amd64.whl#sha256=1daa5ece0af12880b3e890b540794652acf4955e2c29828343cf71a97875ff31 (from https://pypi.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://files.pythonhosted.org/packages/e5/d2/56927d8760592617c49b28c54ad80ed98321041fa050a104398d7afd8bd6/PyQtChart-5.7.1-5.7.1-cp34.cp35.cp36-abi3-macosx_10_6_intel.whl#sha256=a4224ea30c60bf5ee7f0b9c04dbfdaa4d99d4a7ee50a60fd177baa44764a9c25 (from https://pypi.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://files.pythonhosted.org/packages/3c/bf/15a6bf4dd7bd91696ad67b7281a92fa3ac0efbc69a579f944757a933b45e/PyQtChart-5.7.1-5.7.1-cp34.cp35.cp36-abi3-manylinux1_x86_64.whl#sha256=f82a2caf533f81883cd0390ede8bf82f51565bc9701e5dfec340206d2be9392d (from https://pypi.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://files.pythonhosted.org/packages/2c/ad/44ce426ec312b6dec67abebe59d029b5cc196471aee16b3cd2b9ea6a972f/PyQtChart-5.7.1-5.7.1-cp34.cp35.cp36-none-win32.whl#sha256=c9cff5fcd68bc252135d6002208916d203b9bb198154db37103f7b322f6df8d1 (from https://pypi.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://files.pythonhosted.org/packages/8d/41/263846474f7ac01ac744fa5e176c8ae70fd725cb424c0f08bbc2826228d4/PyQtChart-5.7.1-5.7.1-cp34.cp35.cp36-none-win_amd64.whl#sha256=7a7419911800e45c3876c9d37404b97b68af64b3d9576f57d1daa300305a1dc3 (from https://pypi.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://files.pythonhosted.org/packages/5a/40/5a942f5e29e43c6844951aac92087c9fd65440e5f192464f923f6ceb4f0c/PyQtChart-5.8-5.8.0-cp34.cp35.cp36.cp37-abi3-macosx_10_6_intel.whl#sha256=f1af02e6f12c77d70f3512d8d2493ea8e6b817c999e62c04319e9f8321a8993a (from https://pypi.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://files.pythonhosted.org/packages/dc/8b/4e537924cc8c5c52222827b46135fdc9984066bf20d595a62c88bac1fbdd/PyQtChart-5.8-5.8.0-cp34.cp35.cp36.cp37-abi3-manylinux1_x86_64.whl#sha256=1d697b11bd5db0f54b863c230c2739271f3de77ce0e1ddde311e4a2188cae9f6 (from https://pypi.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://files.pythonhosted.org/packages/a6/7d/140b6f5bd30fbf77582455e9160f200e3c30e190c8f5acbfd5908274f3d6/PyQtChart-5.8-5.8.0-cp34.cp35.cp36.cp37-none-win32.whl#sha256=9bff7e11a26b37efa7706cf435925d6d340bd160ecc74b0c70ae8e7a768a5660 (from https://pypi.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://files.pythonhosted.org/packages/e7/04/62951e3a875c0e22d08e83bc73a2ea1f9b3995c63da09693f42b7dedbeb7/PyQtChart-5.8-5.8.0-cp34.cp35.cp36.cp37-none-win_amd64.whl#sha256=9207db20568b07b73b490baa1fe6ec0a9ade996b5a49b6c1d54175e2eaa674b3 (from https://pypi.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://files.pythonhosted.org/packages/8d/8a/cd24a9576fb3a837f53d85d92f4f5e28f1f1095bcd077f6df91baea4c3a7/PyQtChart-5.9-5.9.1-cp35.cp36.cp37-abi3-macosx_10_6_intel.whl#sha256=b67c7dc6175ffee14d3e2c2cd5621b9dde189368103e07685d8f74a411c928b8 (from https://pypi.org/simple/pyqtchart/); it is not compatible with this Python
[... cut ...]
    state = SelfCheckState(cache_dir=options.cache_dir)
  File "/usr/local/lib/python3.5/dist-packages/pip/_internal/utils/outdated.py", line 25, in __init__
    self.statefile_path = os.path.join(cache_dir, "selfcheck.json")
  File "/usr/lib/python3.5/posixpath.py", line 89, in join
    genericpath._check_arg_types('join', a, *p)
  File "/usr/lib/python3.5/genericpath.py", line 143, in _check_arg_types
    (funcname, s.__class__.__name__)) from None
TypeError: join() argument must be str or bytes, not 'bool'
`

plundertasche, 2020-12-26:
**Rant! Why are unresolved issues so often closed? Why not let it open until it is resolved and someone leave the solution in thread?
/Rant!**
MX-Linux pip3 install mu-editor:

**summary:** 
first run with error pathon.h not found (what?)
second run error for requirement qscintilla==2.11.4

**solution in my case: pip3 uninstall flake8**

**$ pip3 install mu-editor**
Collecting mu-editor
  Downloading https://files.pythonhosted.org/packages/fa/2f/6f7eb7e836a528d6959e5c79491d90faef2df7a78e9d953b84451551a98e/mu_editor-1.0.3-py3-none-any.whl (2.0MB)
    100% |████████████████████████████████| 2.0MB 1.1MB/s 
Requirement already satisfied: Pillow>=5.2.0 in /usr/lib/python3/dist-packages (from mu-editor) (5.4.1)
Collecting pgzero==1.2 (from mu-editor)
  Downloading https://files.pythonhosted.org/packages/48/e5/e5f14292373cb5fc7539aa01307b184c1e3c954d68945d8c44778669dd82/pgzero-1.2-py3-none-any.whl (69kB)
    100% |████████████████████████████████| 71kB 4.6MB/s 
Collecting pyqt5==5.14.1 (from mu-editor)
  Downloading https://files.pythonhosted.org/packages/3a/fb/eb51731f2dc7c22d8e1a63ba88fb702727b324c6352183a32f27f73b8116/PyQt5-5.14.1.tar.gz (3.2MB)
    100% |████████████████████████████████| 3.2MB 743kB/s 
  Installing build dependencies ... error
  Complete output from command /usr/bin/python3 -m pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-32ifh0k1 --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- "sip >=5.0.1, <6" "PyQt-builder >=1.1.0, <2":
  Collecting sip<6,>=5.0.1
    Downloading https://files.pythonhosted.org/packages/33/e9/27730ac17713c0a80d81d8f3bb56213f1549d96f9dc183fd16a7eec6287c/sip-5.5.0.tar.gz (1.1MB)
    Installing build dependencies: started
    Installing build dependencies: finished with status 'done'
  Collecting PyQt-builder<2,>=1.1.0
    Downloading https://files.pythonhosted.org/packages/6d/f6/727decf244b9f31a19ef4fa57de6daa307a558858a3c6716e4720f961b99/PyQt_builder-1.6.0-py3-none-any.whl (5.5MB)
  Collecting packaging (from sip<6,>=5.0.1)
    Using cached https://files.pythonhosted.org/packages/b1/a7/588bfa063e7763247ab6f7e1d994e331b85e0e7d09f853c59a6eb9696974/packaging-20.8-py2.py3-none-any.whl
  Collecting toml (from sip<6,>=5.0.1)
    Using cached https://files.pythonhosted.org/packages/44/6f/7120676b6d73228c96e17f1f794d8ab046fc910d781c8d151120c3f1569e/toml-0.10.2-py2.py3-none-any.whl
  Collecting pyparsing>=2.0.2 (from packaging->sip<6,>=5.0.1)
    Using cached https://files.pythonhosted.org/packages/8a/bb/488841f56197b13700afd5658fc279a2025a39e22449b7cf29864669b15d/pyparsing-2.4.7-py2.py3-none-any.whl
  Building wheels for collected packages: sip
    Running setup.py bdist_wheel for sip: started
    Running setup.py bdist_wheel for sip: finished with status 'error'
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-p4opycjx/sip/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-8m4ehnex --python-tag cp37:
    running bdist_wheel
    running build
    running build_py
    creating build
    creating build/lib.linux-x86_64-3.7
    creating build/lib.linux-x86_64-3.7/sipbuild
    copying sipbuild/py_versions.py -> build/lib.linux-x86_64-3.7/sipbuild
    copying sipbuild/helpers.py -> build/lib.linux-x86_64-3.7/sipbuild
    copying sipbuild/distutils_builder.py -> build/lib.linux-x86_64-3.7/sipbuild
    copying sipbuild/api.py -> build/lib.linux-x86_64-3.7/sipbuild
    copying sipbuild/project.py -> build/lib.linux-x86_64-3.7/sipbuild
    copying sipbuild/abstract_builder.py -> build/lib.linux-x86_64-3.7/sipbuild
    copying sipbuild/argument_parser.py -> build/lib.linux-x86_64-3.7/sipbuild
    copying sipbuild/buildable.py -> build/lib.linux-x86_64-3.7/sipbuild
    copying sipbuild/version.py -> build/lib.linux-x86_64-3.7/sipbuild
    copying sipbuild/installable.py -> build/lib.linux-x86_64-3.7/sipbuild
    copying sipbuild/pyproject.py -> build/lib.linux-x86_64-3.7/sipbuild
    copying sipbuild/bindings.py -> build/lib.linux-x86_64-3.7/sipbuild
    copying sipbuild/exceptions.py -> build/lib.linux-x86_64-3.7/sipbuild
    copying sipbuild/builder.py -> build/lib.linux-x86_64-3.7/sipbuild
    copying sipbuild/__init__.py -> build/lib.linux-x86_64-3.7/sipbuild
    copying sipbuild/abstract_project.py -> build/lib.linux-x86_64-3.7/sipbuild
    copying sipbuild/configurable.py -> build/lib.linux-x86_64-3.7/sipbuild
    creating build/lib.linux-x86_64-3.7/sipbuild/tools
    copying sipbuild/tools/sdist.py -> build/lib.linux-x86_64-3.7/sipbuild/tools
    copying sipbuild/tools/build.py -> build/lib.linux-x86_64-3.7/sipbuild/tools
    copying sipbuild/tools/wheel.py -> build/lib.linux-x86_64-3.7/sipbuild/tools
    copying sipbuild/tools/install.py -> build/lib.linux-x86_64-3.7/sipbuild/tools
    copying sipbuild/tools/__init__.py -> build/lib.linux-x86_64-3.7/sipbuild/tools
    creating build/lib.linux-x86_64-3.7/sipbuild/distinfo
    copying sipbuild/distinfo/distinfo.py -> build/lib.linux-x86_64-3.7/sipbuild/distinfo
[... 182 lines cut ...]
  ----------------------------------------
Command "/usr/bin/python3 -m pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-32ifh0k1 --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- "sip >=5.0.1, <6" "PyQt-builder >=1.1.0, <2"" failed with [... cut ...]
  Downloading https://files.pythonhosted.org/packages/da/1d/e6d9af0b5045597869537391f1036ab841c613c3f3e40f16bbc1d75450ee/matplotlib-2.2.2-cp37-cp37m-manylinux1_x86_64.whl (12.6MB)
    100% |████████████████████████████████| 12.6MB 154kB/s 
Collecting qscintilla==2.11.4 (from mu-editor)
  **Could not find a version that satisfies the requirement qscintilla==2.11.4 (from mu-editor) (from versions: 2.9.4, 2.10, 2.10.1, 2.10.2, 2.10.3, 2.10.4, 2.10.7, 2.10.8, 2.11, 2.11.1, 2.11.2, 2.11.3)**
No matching distribution found for qscintilla==2.11.4 (from mu-editor)

cjrw42, 2021-01-08:
Hi

I've had a look at this and its a python versions issue by the looks of it... The deps are python3.6, but in the i386 debian repos for python 3 the highest ver is 3.5

A manual python update should solve

cdesqaz, 2021-03-14:
..." (mu) masua@w2l-pc:~/0l2p0/Python/ide/mu$  pip3 install mu-editor
Collecting mu-editor
  Downloading https://files.pythonhosted.org/packages/fa/2f/6f7eb7e836a528d6959e5c79491d90faef2df7a78e9d953b84451551a98e/mu_editor-1.0.3-py3-none-any.whl (2.0MB)
    100% |████████████████████████████████| 2.0MB 144kB/s 
Collecting pycodestyle==2.4.0 (from mu-editor)
  Cache entry deserialization failed, entry ignored
  Downloading https://files.pythonhosted.org/packages/e5/c6/ce130213489969aa58610042dff1d908c25c731c9575af6935c2dfad03aa/pycodestyle-2.4.0-py2.py3-none-any.whl (62kB)
    100% |████████████████████████████████| 71kB 78kB/s 
Collecting matplotlib==2.2.2 (from mu-editor)
  Downloading https://files.pythonhosted.org/packages/49/b8/89dbd27f2fb171ce753bb56220d4d4f6dbc5fe32b95d8edc4415782ef07f/matplotlib-2.2.2-cp36-cp36m-manylinux1_x86_64.whl (12.6MB)
    100% |████████████████████████████████| 12.6MB 32kB/s 
Collecting PyQtChart==5.14.0 (from mu-editor)
  Downloading https://files.pythonhosted.org/packages/c8/6e/08e68bb5c3c64fbb56f1755c0b4748bb82775fa1b7c9fbaa61572eeaa76a/PyQtChart-5.14.0.tar.gz (68kB)
    100% |████████████████████████████████| 71kB 154kB/s 
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/usr/lib/python3.6/tokenize.py", line 452, in open
        buffer = _builtin_open(filename, 'rb')
    FileNotFoundError: [Errno 2] No such file or directory: '/tmp/pip-build-18td42kf/PyQtChart/setup.py'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-18td42kf/PyQtChart/
" 

hi everyone!
I just wondering about this issues, already read every answer, and at the end put attention on my terminal output en run the command with sudo. That's it. Installation run perfectly.

virtual environment on python3.6 python3  -m venv mu
source ~/myvenv/mu/bin/activate
sudo pip3 install mu-editor

masua@w2l-pc:~$  cd ~/myvenv/
masua@w2l-pc:~/myvenv$  python3 -m venv mu
masua@w2l-pc:~/0l2p0/Python/ide/mu$  source ~/myvenv/mu/bin/activate
(mu) masua@w2l-pc:~/0l2p0/Python/ide/mu$  sudo pip3 install mu-editor
[sudo] contraseña para masua: 
WARNING: The directory '/home/masua/.cache/pip' or its parent directory is not owned or is not writable by the current user. The cache has been disabled. Check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting mu-editor
  Downloading mu_editor-1.0.3-py3-none-any.whl (2.0 MB)
     |████████████████████████████████| 2.0 MB 350 kB/s 
Collecting pyflakes==2.0.0
  Downloading pyflakes-2.0.0-py2.py3-none-any.whl (53 kB)
     |████████████████████████████████| 53 kB 246 kB/s 
Collecting matplotlib==2.2.2
  Downloading matplotlib-2.2.2-cp36-cp36m-manylinux1_x86_64.whl (12.6 MB)
     |████████████████████████████████| 12.6 MB 375 kB/s 
Collecting Pillow>=5.2.0
  Downloading Pillow-8.1.2-cp36-cp36m-manylinux1_x86_64.whl (2.2 MB)
     |████████████████████████████████| 2.2 MB 523 kB/s 
Collecting pyserial==3.4
  Downloading pyserial-3.4-py2.py3-none-any.whl (193 kB)
     |████████████████████████████████| 193 kB 295 kB/s 
Collecting qtconsole==4.3.1
  Downloading qtconsole-4.3.1-py2.py3-none-any.whl (108 kB)
     |████████████████████████████████| 108 kB 278 kB/s 
Collecting gpiozero>=1.4.1
  Downloading gpiozero-1.5.1-py2.py3-none-any.whl (138 kB)
     |████████████████████████████████| 138 kB 286 kB/s 
Requirement already satisfied: requests>=2.19.1 in /usr/local/lib/python3.6/dist-packages (from mu-editor) (2.25.0)
Collecting qscintilla==2.11.4
  Downloading QScintilla-2.11.4-1-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl (1.9 MB)
     |████████████████████████████████| 1.9 MB 318 kB/s 
Collecting PyQtChart==5.14.0
  Downloading PyQtChart-5.14.0-5.14.2-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl (1.2 MB)
     |████████████████████████████████| 1.2 MB 346 kB/s 
Collecting semver>=2.8.0
  Downloading semver-2.13.0-py2.py3-none-any.whl (12 kB)
Collecting appdirs>=1.4.3
  Downloading appdirs-1.4.4-py2.py3-none-any.whl (9.6 kB)
[... 71 lines cut ...]
  Attempting uninstall: Pillow
    Found existing installation: Pillow 5.1.0
    Uninstalling Pillow-5.1.0:
      Successfully uninstalled Pillow-5.1.0
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
flake8 3.8.4 requires pycodestyle<2.7.0,>=2.6.0a1, but you have pycodestyle 2.4.0 which is incompatible.
flake8 3.8.4 requires pyflakes<2.3.0,>=2.2.0, but you have pyflakes 2.0.0 which is incompatible.
autopep8 1.5.5 requires pycodestyle>=2.6.0, but you have pycodestyle 2.4.0 which is incompatible.
Successfully installed Pillow-8.1.2 PyQt5-sip-12.8.1 PyQtChart-5.14.0 appdirs-1.4.4 colorzero-1.1 cycler-0.10.0 gpiozero-1.5.1 guizero-1.1.1 ipykernel-5.5.0 jupyter-client-6.1.12 jupyter-core-4.7.1 kiwisolver-1.3.1 matplotlib-2.2.2 mu-editor-1.0.3 nudatus-0.0.5 numpy-1.19.5 pgzero-1.2 pigpio-1.78 pycodestyle-2.4.0 pyflakes-2.0.0 pygame-1.9.6 pyparsing-2.4.7 pyqt5-5.14.1 pyserial-3.4 pytz-2021.1 qscintilla-2.11.4 qtconsole-4.3.1 semver-2.13.0 tornado-6.1
(mu) masua@w2l-pc:~/0l2p0/Python/ide/mu$  mu-editor
Registrando en /home/masua/.cache/mu/log/mu.log
QApplication: invalid style override 'gtk' passed, ignoring it.
	Available styles: Windows, Fusion
[... cut ...]
Gtk-Message: 09:47:19.852: GtkDialog mapped without a transient parent. This is discouraged.
QObject::killTimer: Timers cannot be stopped from another thread
QObject::~QObject: Timers cannot be stopped from another thread
QThread: Destroyed while thread is still running
Abortado (`core' generado)
(mu) masua@w2l-pc:~/0l2p0/Python/ide/mu$ [IPKernelApp] WARNING | Parent appears to have exited, shutting down.
^C
(mu) masua@w2l-pc:~/0l2p0/Python/ide/mu$  ^C
(mu) masua@w2l-pc:~/0l2p0/Python/ide/mu$  deactivate
masua@w2l-pc:~/0l2p0/Python/ide/mu$