mu-editor/mu #627: Feature Request: Detect multiple micro:bits when flashing

JoshuaLowe1002, 2018-08-24:
Great work on Mu so far guys, I find myself using it nearly every day and it's an awesome tool!

Earlier I was doing some radio stuff with micro:bit and had 2 micro:bits plugged in. When I wanted to flash code to them, Mu only flashed to one micro:bit.

Would it be possible to have it detect both and show a dialog asking "which one would you like to flash" and even have a button in that dialog box that allows you to flash to both micro:bits?

Josh :)

ntoll, 2018-08-24:
Greetings @JoshuaLowe1002! ;-)

So this is already catered for in uflash (see the docs here: https://uflash.readthedocs.io/en/latest/) which is ultimately what Mu uses to get MicroPython onto the devices.

Since this is, how can I put it, a more-than-beginner type feature in a beginner's editor I'm not sure putting it into Mu will help beginners. It'd certainly make the flashing of micro:bits more complicated, which is something I'd like to avoid.

Thoughts..?

JoshuaLowe1002, 2018-08-24:
Agreed that it's a more-than-beginner feature. Maybe it does make it too complicated. 
Maybe just have the dialog pop up when it detects 2 micro:bits and carry on flashing the normal way when it only detects one.

It's ultimately up to you, these are merely suggestions.

ntoll, 2018-08-24:
+1 absolutely.

Remember, we have a "rule of 3" with Mu: If three people independently ask for the same feature then it shows there's enough demand for it. ;-) We're always open to suggestions.

CleoQc, 2018-08-27:
I'm in as a second vote! I'd love to have this. It would only show up if two or more microbits are detected which would leave the UI simple for the beginner.

dfarning, 2018-08-28:
I have been thinking about this for a while.

Last week I modified the Mu code to add a module to auto detect when there was an updated firmware version for an adafruit device. The further I got, the more it seemed that Mu was not the correct place for this type of functionality.

The first five minutes are so important. The first time I sat down with Mu and CircuitPython everything just work.... I was hooked.

I was wondering about the possibility of an intermediate mu for more advanced users. This could act as a relief valve for those of us who suffer from feature creep. Alternatively, new features could be pointed towards atom plugins.

david

ZanderBrown, 2018-08-28:
The 'Mega Mu' re-enters debate