mu-editor/mu #631: Error installing on Linux Mint 18.3

abixadamj, 2018-08-29:
I want to install, but I see error:
[...]
```
Requirement already satisfied: pyzmq>=13 in /usr/lib/python3/dist-packages (from jupyter-client>=4.1->qtconsole==4.3.1->mu-editor) (15.2.0)
Collecting tornado>=4.1 (from jupyter-client>=4.1->qtconsole==4.3.1->mu-editor)
  Using cached https://files.pythonhosted.org/packages/45/ec/f2a03a0509bcfca336bef23a3dab0d07504893af34fd13064059ba4a0503/tornado-5.1.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 11, in <module>
        from setuptools.extern.six.moves import filterfalse, map
      File "/usr/lib/python3/dist-packages/setuptools/extern/__init__.py", line 1, in <module>
        from pkg_resources.extern import VendorImporter
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2927, in <module>
        @_call_aside
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2913, in _call_aside
        f(*args, **kwargs)
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2952, in _initialize_master_working_set
        add_activation_listener(lambda dist: dist.activate())
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 956, in subscribe
        callback(dist)
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2952, in <lambda>
        add_activation_listener(lambda dist: dist.activate())
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2515, in activate
        declare_namespace(pkg)
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2097, in declare_namespace
        _handle_ns(packageName, path_item)
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2047, in _handle_ns
        _rebuild_mod_path(path, packageName, module)
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2066, in _rebuild_mod_path
        orig_path.sort(key=position_in_sys_path)
    AttributeError: '_NamespacePath' object has no attribute 'sort'

```
My system is:

```
sp-43-bytom tmp # lsb_release -a
LSB Version:	core-9.20160110ubuntu0.2-amd64:core-9.20160110ubuntu0.2-noarch:printing-9.20160110ubuntu0.2-amd64:printing-9.20160110ubuntu0.2-noarch:security-9.20160110ubuntu0.2-amd64:security-9.20160110ubuntu0.2-noarch
Distributor ID:	LinuxMint
Description:	Linux Mint 18.3 Sylvia
Release:	18.3
Codename:	sylvia
sp-43-bytom tmp # uname -a
Linux sp-43-bytom 4.8.0-53-generic #56~16.04.1-Ubuntu SMP Tue May 16 01:18:56 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
sp-43-bytom tmp # python3 -V
Python 3.5.2

```

abixadamj, 2018-10-30:
Dear @carlosperate - I just switched to use Mint 19. I realized that I need to have newest python.

adasiek@adasiek-desktop:~$ python3
Python 3.6.6 (default, Sep 12 2018, 18:26:19) 
[GCC 8.0.1 20180414 (experimental) [trunk revision 259383]] on linux

On that everything wirks like a charm ;-)

closed 2018-10-30 by abixadamj

carlosperate, 2018-10-30:
I'm glad it works now and thanks for the update!