mu-editor/mu #635: Save a File crashes

standera, 2018-09-04:
1) Start Mu Editor in Micro:Bit mode,
2) Click "Save" (there might be something to save or nothing at all) -> a window dialog for saving a file appears,
3) after a few seconds (1..3) the crash window appears and Mu Editor is gone.

Windows 8, last version of Mu Editor. I have tried reinstalling couple of times (one / all users) but no success. Actually there is no way how to solve it, or is it?

tjguk, 2018-09-04:
@standera that's certainly strange. To help us debug, can you add some more info, please?

1) Do you have a micro:bit plugged in at the time?
2) When you say "the crash window" -- do you mean the generic Microsoft [This application has stopped responding] popup? Or something else?
3) Can you confirm where you're downloading the editor from, please? 
4) If you have access to any other Windows machine, have you tried there and seen the same or different results?

Thanks

tjguk, 2018-09-04:
PS feel free to add screenshots if you think it will help

ntoll, 2018-09-04:
Also, please click on the cog icon in the bottom right of the editor and paste the content of the log file. Log files are recreated each day, so make sure you have recently crashed Mu.

This is *really* odd... since we've never had any problems in any Windows version for saving a file to the filesystem. This suggests the problem may be local to your setup. Are you using a machine that is perhaps "locked down" by school/work? Is there anything about your Windows setup that makes it different to a default install? Perhaps you're using a certain sort of anti-virus software (I'm clutching at straws here, but I'm trying to think of applications which may interfere with the workings of Mu).

The more information you can provide, the easier it'll be for us to identify the cause. The log file is key to this.

Thank you! Having bug reports like yours is so helpful to making Mu a better editor.

ntoll, 2018-09-05:
@standera if you copy and paste the log file after Mu crashes (see my previous instructions above), we'll be able to confirm your hunch about non ASCII characters in the account name. ;-)

standera, 2018-09-05:
2018-09-05 08:09:27,583 - root:112(run) INFO:

-----------------

Starting Mu 1.0.0

2018-09-05 08:09:27,599 - root:113(run) INFO: uname_result(system='Windows', node='MyDELL', release='8.1', version='6.3.9600', machine='AMD64', processor='Intel64 Family 6 Model 60 Stepping 3, GenuineIntel')

2018-09-05 08:09:27,599 - root:114(run) INFO: Python path: ['C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\Python\\python36.zip', 'C:\\Program Files\\Mu\\Python', 'C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\pkgs\\IPython\\extensions']

2018-09-05 08:09:28,364 - mu.logic:540(__init__) INFO: Setting up editor.

2018-09-05 08:09:28,364 - mu.logic:558(__init__) INFO: Settings path: C:\Users\MIROSL~1\AppData\Local\python\mu\settings.json

2018-09-05 08:09:28,364 - mu.logic:559(__init__) INFO: Session path: C:\Users\MIROSL~1\AppData\Local\python\mu\session.json

2018-09-05 08:09:28,364 - mu.logic:560(__init__) INFO: Log directory: C:\Users\MIROSL~1\AppData\Local\python\mu\Logs

2018-09-05 08:09:28,364 - mu.logic:561(__init__) INFO: Data directory: C:\Users\MIROSL~1\AppData\Local\python\mu

2018-09-05 08:09:28,379 - mu.logic:574(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero

2018-09-05 08:09:30,037 - mu.logic:1110(change_mode) INFO: Workspace directory: C:\Users\Miroslav štandera\mu_code

2018-09-05 08:09:30,037 - mu.logic:623(restore_session) INFO: Restoring session from: C:\Users\MIROSL~1\AppData\Local\python\mu\session.json

2018-09-05 08:09:30,037 - mu.logic:624(restore_session) DEBUG: {'theme': 'day', 'mode': 'microbit', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': ''}

2018-09-05 08:09:30,037 - mu.logic:645(restore_session) INFO: Loaded files.

2018-09-05 08:09:30,037 - mu.logic:649(restore_session) INFO: User defined environment variables: []

2018-09-05 08:09:30,037 - mu.logic:653(restore_session) INFO: Minify scripts on micro:bit? False

2018-09-05 08:09:30,566 - mu.logic:671(restore_session) INFO: Starting with blank file.

2018-09-05 08:09:30,613 - mu.logic:1110(change_mode) INFO: Workspace directory: C:\Users\Miroslav štandera\mu_code

2018-09-05 08:09:33,895 - mu.logic:1067(select_mode) INFO: Showing available modes: ['python', 'adafruit', 'microbit', 'debugger', 'pygamezero']

2018-09-05 08:09:38,220 - mu.logic:1072(select_mode) INFO: New mode selected: python

2018-09-05 08:09:38,251 - mu.logic:1110(change_mode) INFO: Workspace directory: C:\Users\Miroslav štandera\mu_code

2018-09-05 08:12:09,634 - root:112(run) INFO:

-----------------

Starting Mu 1.0.0

2018-09-05 08:12:09,634 - root:113(run) INFO: uname_result(system='Windows', node='MyDELL', release='8.1', version='6.3.9600', machine='AMD64', processor='Intel64 Family 6 Model 60 Stepping 3, GenuineIntel')

2018-09-05 08:12:09,634 - root:114(run) INFO: Python path: ['C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\Python\\python36.zip', 'C:\\Program Files\\Mu\\Python', 'C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\pkgs\\IPython\\extensions']

2018-09-05 08:12:09,759 - mu.logic:540(__init__) INFO: Setting up editor.

2018-09-05 08:12:09,759 - mu.logic:558(__init__) INFO: Settings path: C:\Users\MIROSL~1\AppData\Local\python\mu\settings.json

2018-09-05 08:12:09,759 - mu.logic:559(__init__) INFO: Session path: C:\Users\MIROSL~1\AppData\Local\python\mu\session.json

2018-09-05 08:12:09,759 - mu.logic:560(__init__) INFO: Log directory: C:\Users\MIROSL~1\AppData\Local\python\mu\Logs

2018-09-05 08:12:09,759 - mu.logic:561(__init__) INFO: Data directory: C:\Users\MIROSL~1\AppData\Local\python\mu

2018-09-05 08:12:09,775 - mu.logic:574(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero

2018-09-05 08:12:10,165 - mu.logic:1110(change_mode) INFO: Workspace directory: C:\Users\Miroslav štandera\mu_code

2018-09-05 08:12:10,165 - mu.logic:623(restore_session) INFO: Restoring session from: C:\Users\MIROSL~1\AppData\Local\python\mu\session.json

[... 104 lines cut ...]
2018-09-05 08:27:18,941 - mu.logic:560(__init__) INFO: Log directory: C:\Users\MIROSL~1\AppData\Local\python\mu\Logs

2018-09-05 08:27:18,941 - mu.logic:561(__init__) INFO: Data directory: C:\Users\MIROSL~1\AppData\Local\python\mu

2018-09-05 08:27:18,963 - mu.logic:574(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero

2018-09-05 08:27:19,355 - mu.logic:1110(change_mode) INFO: Workspace directory: C:\Users\Miroslav štandera\mu_code

2018-09-05 08:27:19,355 - mu.logic:623(restore_session) INFO: Restoring session from: C:\Users\MIROSL~1\AppData\Local\python\mu\session.json

2018-09-05 08:27:19,355 - mu.logic:624(restore_session) DEBUG: {'theme': 'day', 'mode': 'microbit', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': ''}

2018-09-05 08:27:19,355 - mu.logic:645(restore_session) INFO: Loaded files.

2018-09-05 08:27:19,355 - mu.logic:649(restore_session) INFO: User defined environment variables: []

2018-09-05 08:27:19,355 - mu.logic:653(restore_session) INFO: Minify scripts on micro:bit? False

2018-09-05 08:27:19,724 - mu.logic:671(restore_session) INFO: Starting with blank file.

2018-09-05 08:27:19,739 - mu.logic:1110(change_mode) INFO: Workspace directory: C:\Users\Miroslav štandera\mu_code

2018-09-05 08:27:21,172 - mu.logic:1037(show_admin) INFO: Showing logs from C:\Users\MIROSL~1\AppData\Local\python\mu\Logs\mu.log

Štanderovi, Nový Bor

ntoll, 2018-09-05:
Hi @standera,

Thanks for the log. Sorry for the barrage of questions, but this will help us isolate what the problem may be.

I can see from the logs that you started the Mu several times during the session this log file covers. Can you confirm that Mu was crashing in the way you state during this session?

I think your answer will be "yes" and I find it puzzling there's no stack trace or debug information. We've setup Mu so that if it ever crashes, then it should log exactly what went wrong. However, the lack of evidence for this crash report in the log (hence my check/confirmation that you actually observed a crash), suggests that it wasn't so much Mu that crashed, as Python itself (which Mu is written in) which crashed -- if you see what I mean.

In any case, I believe your suggestion of non-ASCII in the save path might be the problem. I guess we'll start there and try to recreate the problem. Once we can observe the problem we can write a test to recreate it, then fix Mu so it no longer crashes (or change Mu so it no longer crashes Python).

Thanks for your support!

N.

tjguk, 2018-09-05:
@standera Unfortunately I can't see a screenshot in your earlier post, so it looks like Github's process has dropped it. Can you try adding directly from the GH web interface, please? I'm still unsure whether you're getting a Windows fatal error -- in which case we probably won't get anything in the logs -- or some other error.

standera, 2018-09-05:
No problem you do not bother me. I would like to help as far as I am able to follow.

I can see from the logs that you started the Mu several times during the session this log file covers. Can you confirm that Mu was crashing in the way you state during this session?

I
Indeed I can confirm YES.  

Štanderovi, Nový Bor

standera, 2018-09-05:
[image]

standera, 2018-09-05:
The image should be there. I have just uploaded it.

Štanderovi, Nový Bor

tjguk, 2018-09-05:
@standera Thanks. Honestly, this is going to be tricky to debug. I don't think it's as straightforward as the fact of your having non-ASCII characters, although that's always a suspect. We have at least one user running in a Chinese locale where *every* character is non-ASCII.

Are you in a position to run a development version of Mu? ie by creating a virtual environment and using pip to install the current version? If you don't know what I'm talking about, that's fine: I'm just trying to get a different angle on the problem and to narrow down exactly what's causing the crash.

standera, 2018-09-05:
Sadly this is an unknown territory to me.

Štanderovi, Nový Bor

grofte, 2018-09-08:
One of my students is having a problem that sounds like this.  
If Mu crashes the entire contents of the file she is working in is gone. She does not have any micro:bit attached.

carlosperate, 2020-09-03:
Thanks again for the update @standera. There has been a few updates to better support Unicode since the last release, would you be able to try Mu from the master branch and see if that still causes the crash?

https://mu.readthedocs.io/en/latest/#quickstart

standera, 2020-09-05:
Hello Carlos,

I would like to but I am not familiar with Github at all. So if there is a standard installation program (for Windows) just send me a link and I will try.

Sorry for my ignorance.

carlosperate, 2020-09-25:
Hi @standera,

You can download the (testing) latest Windows installer from the master branch here: http://mu-builds.s3-website.eu-west-2.amazonaws.com/?prefix=windows/

standera, 2020-09-29:
Hello,

sorry for being delayed. I tried the latest version (64bit) today. Still the some result. No way how to save the file. I am sending you two screen shots (see the attached files) to better illustrate the problem (unfortunately only Czech language). 

The text bellow is a report taken from the standard windows-crashing program. It may help as well. By the way hitting the help button in Mu leads to nowhere (a small bug).

Podpis problému:
Název události problému: APPCRASH
Název aplikace: pythonw.exe
Verze aplikace: 3.6.3150.1013
Časové razítko aplikace: 59d3d3a5
Název chybného modulu: StackHash_fe12
Verze chybného modulu: 6.3.9600.19678
Časové razítko chybného modulu: 5e82c88a
Kód výjimky: c0000374
Posun výjimky: PCH_A2_FROM_ntdll+0x0000000000090CBA
Verze operačního systému: 6.3.9600.2.0.0.256.48
ID národního prostředí: 1029
Další informace 1: fe12
Další informace 2: fe1200a364f573139d2ff1ee6fb7ada3
Další informace 3: 5123
Další informace 4: 512308f6725192308933b70c65d0b4f5

Miroslav Štandera, Nový Bor

dybber, 2021-02-02:
Hi @standera, it's my first look at this bug. 

I can see that you intended to include attached screenshots in your last email-reply to this thread. However, it seems that the attachments are not included when send as email, could you login to Github and upload the screenshots? (if you still have them... perhaps they are saved in your mailbox for "Sent mail")

MichelleStella, 2021-04-12:
I have the same problem. I have try to install the latest version (1.1.0 beta 3) on windows 10 (64 bits) several times

2021-04-12 10:33:12,940 - root:113(run) INFO: 

-----------------

Starting Mu 1.1.0.alpha.2
2021-04-12 10:33:12,942 - root:114(run) INFO: uname_result(system='Windows', node='Michelle-PC', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 61 Stepping 4, GenuineIntel')
2021-04-12 10:33:12,943 - root:115(run) INFO: Python path: ['C:\\Users\\Michelle\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\Michelle\\AppData\\Local\\Mu', 'C:\\Users\\Michelle\\AppData\\Local\\Mu\\Python\\python36.zip', 'C:\\Users\\Michelle\\AppData\\Local\\Mu\\Python\\DLLs', 'C:\\Users\\Michelle\\AppData\\Local\\Mu\\Python\\lib', 'C:\\Users\\Michelle\\AppData\\Local\\Mu\\Python', 'C:\\Users\\Michelle\\AppData\\Local\\python\\mu\\site-packages', 'C:\\Users\\Michelle\\AppData\\Local\\Mu\\pkgs\\IPython\\extensions']
2021-04-12 10:33:12,943 - root:116(run) INFO: Language code: fr_CH
2021-04-12 10:33:13,224 - mu.logic:587(__init__) INFO: Setting up editor.
2021-04-12 10:33:13,225 - mu.logic:609(__init__) INFO: Settings path: C:\Users\Michelle\AppData\Local\python\mu\settings.json
2021-04-12 10:33:13,225 - mu.logic:610(__init__) INFO: Session path: C:\Users\Michelle\AppData\Local\python\mu\session.json
2021-04-12 10:33:13,226 - mu.logic:611(__init__) INFO: Log directory: C:\Users\Michelle\AppData\Local\python\mu\Logs
2021-04-12 10:33:13,226 - mu.logic:612(__init__) INFO: Data directory: C:\Users\Michelle\AppData\Local\python\mu
2021-04-12 10:33:13,265 - mu.logic:625(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, debugger, pygamezero
2021-04-12 10:33:14,888 - mu.logic:1266(change_mode) INFO: Workspace directory: C:\Users\Michelle\mu_code
2021-04-12 10:33:14,892 - mu.logic:684(restore_session) INFO: Restoring session from: C:\Users\Michelle\AppData\Local\python\mu\session.json
2021-04-12 10:33:14,892 - mu.logic:685(restore_session) DEBUG: {'theme': 'day', 'mode': 'python', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2, 'window': {'x': 192, 'y': 108, 'w': 1536, 'h': 864}}
2021-04-12 10:33:14,893 - mu.logic:706(restore_session) INFO: Loaded files.
2021-04-12 10:33:14,893 - mu.logic:710(restore_session) INFO: User defined environment variables: []
2021-04-12 10:33:14,893 - mu.logic:714(restore_session) INFO: Minify scripts on micro:bit? False
2021-04-12 10:33:15,072 - mu.logic:1266(change_mode) INFO: Workspace directory: C:\Users\Michelle\mu_code
2021-04-12 10:33:15,876 - mu.logic:742(restore_session) INFO: Starting with blank file.
2021-04-12 10:43:16,945 - mu.logic:1221(select_mode) INFO: Showing available modes: ['python', 'circuitpython', 'microbit', 'esp', 'web', 'debugger', 'pygamezero']
2021-04-12 10:43:20,412 - mu.logic:1226(select_mode) INFO: New mode selected: microbit
2021-04-12 10:43:20,522 - mu.logic:1266(change_mode) INFO: Workspace directory: C:\Users\Michelle\mu_code
2021-04-12 10:51:14,649 - mu.modes.microbit:154(flash) INFO: Preparing to flash script.
2021-04-12 10:51:14,652 - mu.modes.microbit:164(flash) DEBUG: Python script:
2021-04-12 10:51:14,652 - mu.modes.microbit:165(flash) DEBUG: b'# Ecrit ton programme ici ;-)\nfrom microbit import *\r\n\r\ndisplay.scroll("BIENVENUE")'
2021-04-12 10:51:14,668 - mu.modes.microbit:212(flash) INFO: Path to micro:bit: D:\
2021-04-12 10:51:14,669 - mu.modes.base:261(find_device) INFO: Found device on port: COM4
2021-04-12 10:51:14,670 - mu.modes.base:262(find_device) INFO: Serial number: 9901000051774E45005D7013000000220000000097969901
2021-04-12 10:51:14,670 - mu.modes.microbit:217(flash) INFO: Serial port: COM4
2021-04-12 10:51:14,670 - mu.modes.microbit:218(flash) INFO: Device serial number: 9901000051774E45005D7013000000220000000097969901
2021-04-12 10:51:14,671 - mu.modes.microbit:243(flash) INFO: Checking target device.
2021-04-12 10:51:15,325 - mu.modes.microbit:247(flash) INFO: {'sysname': 'microbit', 'nodename': 'microbit', 'release': '1.0.1', 'version': 'micro:bit v1.0.1+b0bf4a9 on 2018-12-13; MicroPython v1.9.2-34-gd64154c73 on 2017-09-01', 'machine': 'micro:bit with nRF51822'}
2021-04-12 10:51:15,327 - mu.modes.microbit:266(flash) INFO: Board MicroPython: 1.0.1
2021-04-12 10:51:15,327 - mu.modes.microbit:268(flash) INFO: Mu MicroPython: 1.0.1
2021-04-12 10:51:15,330 - mu.modes.microbit:420(copy_main) INFO: Copying main.py onto device
2021-04-12 10:51:15,330 - mu.modes.microbit:430(copy_main) INFO: ["fd = open('main.py', 'wb')", 'f = fd.write', "f(b'# Ecrit ton programme ici ;-)\\nfrom microbit import *\\r\\n\\r\\ndisplay.')", 'f(b\'scroll("BIENVENUE")\')', 'fd.close()']
2021-04-12 10:51:15,801 - mu.modes.microbit:433(copy_main) INFO: (b'', b'')
2021-04-12 10:51:27,641 - mu.modes.microbit:154(flash) INFO: Preparing to flash script.
2021-04-12 10:51:27,641 - mu.modes.microbit:164(flash) DEBUG: Python script:
2021-04-12 10:51:27,642 - mu.modes.microbit:165(flash) DEBUG: b'# Ecrit ton programme ici ;-)\nfrom microbit import *\r\n\r\ndisplay.scroll("BIENVENUE")'
2021-04-12 10:51:27,642 - mu.modes.microbit:212(flash) INFO: Path to micro:bit: D:\
2021-04-12 10:51:27,645 - mu.modes.base:261(find_device) INFO: Found device on port: COM4
2021-04-12 10:51:27,645 - mu.modes.base:262(find_device) INFO: Serial number: 9901000051774E45005D7013000000220000000097969901
2021-04-12 10:51:27,646 - mu.modes.microbit:217(flash) INFO: Serial port: COM4
[... cut ...]
2021-04-12 12:00:01,076 - mu.logic:737(check_usb) INFO: microbit device connected on port: COM4(VID: 0x0D28, PID: 0x0204, manufacturer: 'Microsoft')
2021-04-12 12:02:12,516 - mu.logic:1396(show_admin) INFO: Showing admin with logs from C:\Users\Michelle\AppData\Local\python\mu\Logs\mu.log
2021-04-12 12:02:12,517 - mu.virtual_environment:702(installed_packages) INFO: Discovering installed third party modules in venv.
2021-04-12 12:02:16,357 - mu.virtual_environment:723(installed_packages) INFO: []

carlosperate, 2021-04-12:
Hi @MichelleStella,
It'd be really useful to get the Mu logs to debug this issue.
Could you trigger the problem in Mu (basically, make it crash so it's the last thing in the log) and then attach your log file here? It should be located in `%LOCALAPPDATA%\python\mu\Logs`.

MichelleStella, 2021-04-12:
> Hi @MichelleStella,
> It'd be really useful to get the Mu logs to debug this issue.
> Could you trigger the problem in Mu (basically, make it crash so it's the last thing in the log) and then attach your log file here? It should be located in `%LOCALAPPDATA%\python\mu\Logs`.

of course :
 -----------------

Starting Mu 1.1.0.beta.3
2021-04-12 19:30:50,472 - root:267(run) INFO: uname_result(system='Windows', node='Michelle-PC', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 61 Stepping 4, GenuineIntel')
2021-04-12 19:30:50,472 - root:268(run) INFO: Python path: ['C:\\Users\\Michelle\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\Michelle\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\Michelle\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\Michelle\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\Michelle\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\Michelle\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\Michelle\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\Michelle\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\Michelle\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\Michelle\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\Michelle\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\Michelle\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-04-12 19:30:50,472 - root:269(run) INFO: Language code: fr_CH
2021-04-12 19:30:50,673 - mu.virtual_environment:435(ensure_and_create) INFO: Added log handler.
2021-04-12 19:30:51,573 - mu.virtual_environment:439(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-04-12 19:30:51,597 - mu.virtual_environment:489(ensure_path) INFO: Virtual Environment found at: C:\Users\Michelle\AppData\Local\python\mu\mu_venv-38-20210412-112738
2021-04-12 19:30:51,612 - mu.virtual_environment:502(ensure_interpreter) INFO: Interpreter found at: C:\Users\Michelle\AppData\Local\python\mu\mu_venv-38-20210412-112738\scripts\python.exe
2021-04-12 19:30:52,979 - mu.virtual_environment:535(ensure_interpreter_version) INFO: Both interpreters at version 38
2021-04-12 19:30:52,991 - mu.virtual_environment:567(ensure_pip) INFO: Pip found at: C:\Users\Michelle\AppData\Local\python\mu\mu_venv-38-20210412-112738\scripts\pip.exe
2021-04-12 19:30:53,004 - mu.virtual_environment:549(ensure_key_modules) DEBUG: Verifying import of: pgzero
2021-04-12 19:30:53,892 - mu.virtual_environment:549(ensure_key_modules) DEBUG: Verifying import of: flask
2021-04-12 19:30:55,870 - mu.virtual_environment:549(ensure_key_modules) DEBUG: Verifying import of: serial
2021-04-12 19:30:56,846 - mu.virtual_environment:549(ensure_key_modules) DEBUG: Verifying import of: qtconsole
2021-04-12 19:30:57,815 - mu.virtual_environment:549(ensure_key_modules) DEBUG: Verifying import of: nudatus
2021-04-12 19:30:58,705 - mu.virtual_environment:549(ensure_key_modules) DEBUG: Verifying import of: esptool
2021-04-12 19:30:59,671 - mu.virtual_environment:451(ensure_and_create) INFO: Virtual environment already exists.
2021-04-12 19:31:00,022 - mu.logic:758(__init__) INFO: Setting up editor.
2021-04-12 19:31:00,023 - mu.logic:778(__init__) INFO: Log directory: C:\Users\Michelle\AppData\Local\python\mu\Logs
2021-04-12 19:31:00,024 - mu.logic:779(__init__) INFO: Data directory: C:\Users\Michelle\AppData\Local\python\mu
2021-04-12 19:31:00,024 - mu.logic:793(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, pyboard, debugger, pygamezero, lego, pico
2021-04-12 19:31:00,024 - mu.modes.base:61(get_default_workspace) INFO: Using workspace C:\Users\Michelle\mu_code from settings file
2021-04-12 19:31:00,209 - mu.modes.base:61(get_default_workspace) INFO: Using workspace C:\Users\Michelle\mu_code from settings file
2021-04-12 19:31:00,209 - mu.logic:1523(change_mode) INFO: Workspace directory: C:\Users\Michelle\mu_code
2021-04-12 19:31:00,209 - mu.logic:867(restore_session) DEBUG: <SessionSettings from C:\Users\Michelle\AppData\Local\python\mu\session.json>
2021-04-12 19:31:00,253 - mu.logic:889(restore_session) INFO: Loaded files.
2021-04-12 19:31:00,254 - mu.logic:892(restore_session) INFO: User defined environment variables: []
2021-04-12 19:31:00,255 - mu.logic:897(restore_session) INFO: Minify scripts on micro:bit? False
2021-04-12 19:31:00,425 - mu.modes.base:61(get_default_workspace) INFO: Using workspace C:\Users\Michelle\mu_code from settings file
2021-04-12 19:31:00,425 - mu.logic:1523(change_mode) INFO: Workspace directory: C:\Users\Michelle\mu_code
2021-04-12 19:31:01,144 - mu.logic:943(restore_session) INFO: Starting with blank file.
2021-04-12 19:31:32,069 - mu.logic:1396(show_admin) INFO: Showing admin with logs from C:\Users\Michelle\AppData\Local\python\mu\Logs\mu.log
2021-04-12 19:31:32,069 - mu.virtual_environment:702(installed_packages) INFO: Discovering installed third party modules in venv.
2021-04-12 19:31:34,308 - mu.virtual_environment:723(installed_packages) INFO: []
2021-04-12 19:33:06,353 - mu.logic:1442(show_admin) INFO: No admin settings changed.
[... cut ...]
2021-04-12 19:33:44,419 - mu.logic:1523(change_mode) INFO: Workspace directory: C:\Users\Michelle\mu_code
2021-04-12 19:33:44,891 - mu.logic:943(restore_session) INFO: Starting with blank file.
2021-04-12 19:34:08,726 - mu.logic:1396(show_admin) INFO: Showing admin with logs from C:\Users\Michelle\AppData\Local\python\mu\Logs\mu.log
2021-04-12 19:34:08,727 - mu.virtual_environment:702(installed_packages) INFO: Discovering installed third party modules in venv.
2021-04-12 19:34:09,965 - mu.virtual_environment:723(installed_packages) INFO: []

carlosperate, 2021-04-12:
Thanks for the logs @MichelleStella!
I don't see any errors or crashes though, did you open Mu and trigger a crash (or the error you've experienced) before coping the log contents?

If Mu does close down or crash, what do you see, when does it crash?

carlosperate, 2021-04-12:
@standera we've done a new release since you last tested this issue, would you be able to install the latest release and try again? Thank you again for all your testing!
https://github.com/mu-editor/mu/releases

MichelleStella, 2021-04-12:
> Thanks for the logs @MichelleStella!
> I don't see any errors or crashes though, did you open Mu and trigger a crash (or the error you've experienced) before coping the log contents?
> 
> If Mu does close down or crash, what do you see, when does it crash?

Weird..
Exactly: I triggered the crash, then forced the program to close, then reopened it and copied the log files

Here is the video of what happened (sorry for the background noise)

[https://www.youtube.com/watch?v=3ARHA6teusI&ab_channel=MichelleStella](url)

-----------------

Starting Mu 1.1.0.beta.3
2021-04-12 19:33:36,411 - root:267(run) INFO: uname_result(system='Windows', node='Michelle-PC', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 61 Stepping 4, GenuineIntel')
2021-04-12 19:33:36,411 - root:268(run) INFO: Python path: ['C:\\Users\\Michelle\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\Michelle\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\Michelle\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\Michelle\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\Michelle\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\Michelle\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\Michelle\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\Michelle\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\Michelle\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\Michelle\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\Michelle\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\Michelle\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-04-12 19:33:36,411 - root:269(run) INFO: Language code: fr_CH
2021-04-12 19:33:36,522 - mu.virtual_environment:435(ensure_and_create) INFO: Added log handler.
2021-04-12 19:33:36,981 - mu.virtual_environment:439(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-04-12 19:33:36,990 - mu.virtual_environment:489(ensure_path) INFO: Virtual Environment found at: C:\Users\Michelle\AppData\Local\python\mu\mu_venv-38-20210412-112738
2021-04-12 19:33:37,004 - mu.virtual_environment:502(ensure_interpreter) INFO: Interpreter found at: C:\Users\Michelle\AppData\Local\python\mu\mu_venv-38-20210412-112738\scripts\python.exe
2021-04-12 19:33:37,701 - mu.virtual_environment:535(ensure_interpreter_version) INFO: Both interpreters at version 38
2021-04-12 19:33:37,712 - mu.virtual_environment:567(ensure_pip) INFO: Pip found at: C:\Users\Michelle\AppData\Local\python\mu\mu_venv-38-20210412-112738\scripts\pip.exe
2021-04-12 19:33:37,727 - mu.virtual_environment:549(ensure_key_modules) DEBUG: Verifying import of: pgzero
2021-04-12 19:33:38,566 - mu.virtual_environment:549(ensure_key_modules) DEBUG: Verifying import of: flask
2021-04-12 19:33:40,167 - mu.virtual_environment:549(ensure_key_modules) DEBUG: Verifying import of: serial
2021-04-12 19:33:41,214 - mu.virtual_environment:549(ensure_key_modules) DEBUG: Verifying import of: qtconsole
2021-04-12 19:33:42,084 - mu.virtual_environment:549(ensure_key_modules) DEBUG: Verifying import of: nudatus
2021-04-12 19:33:42,973 - mu.virtual_environment:549(ensure_key_modules) DEBUG: Verifying import of: esptool
2021-04-12 19:33:43,896 - mu.virtual_environment:451(ensure_and_create) INFO: Virtual environment already exists.
2021-04-12 19:33:44,107 - mu.logic:758(__init__) INFO: Setting up editor.
2021-04-12 19:33:44,107 - mu.logic:778(__init__) INFO: Log directory: C:\Users\Michelle\AppData\Local\python\mu\Logs
2021-04-12 19:33:44,107 - mu.logic:779(__init__) INFO: Data directory: C:\Users\Michelle\AppData\Local\python\mu
2021-04-12 19:33:44,107 - mu.logic:793(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, pyboard, debugger, pygamezero, lego, pico
2021-04-12 19:33:44,108 - mu.modes.base:61(get_default_workspace) INFO: Using workspace C:\Users\Michelle\mu_code from settings file
2021-04-12 19:33:44,228 - mu.modes.base:61(get_default_workspace) INFO: Using workspace C:\Users\Michelle\mu_code from settings file
2021-04-12 19:33:44,229 - mu.logic:1523(change_mode) INFO: Workspace directory: C:\Users\Michelle\mu_code
2021-04-12 19:33:44,230 - mu.logic:867(restore_session) DEBUG: <SessionSettings from C:\Users\Michelle\AppData\Local\python\mu\session.json>
2021-04-12 19:33:44,262 - mu.logic:889(restore_session) INFO: Loaded files.
2021-04-12 19:33:44,263 - mu.logic:892(restore_session) INFO: User defined environment variables: []
2021-04-12 19:33:44,263 - mu.logic:897(restore_session) INFO: Minify scripts on micro:bit? False
2021-04-12 19:33:44,419 - mu.modes.base:61(get_default_workspace) INFO: Using workspace C:\Users\Michelle\mu_code from settings file
2021-04-12 19:33:44,419 - mu.logic:1523(change_mode) INFO: Workspace directory: C:\Users\Michelle\mu_code
2021-04-12 19:33:44,891 - mu.logic:943(restore_session) INFO: Starting with blank file.
2021-04-12 19:34:08,726 - mu.logic:1396(show_admin) INFO: Showing admin with logs from C:\Users\Michelle\AppData\Local\python\mu\Logs\mu.log
2021-04-12 19:34:08,727 - mu.virtual_environment:702(installed_packages) INFO: Discovering installed third party modules in venv.
[... cut ...]
2021-04-12 19:59:05,020 - mu.virtual_environment:702(installed_packages) INFO: Discovering installed third party modules in venv.
2021-04-12 19:59:07,284 - mu.virtual_environment:723(installed_packages) INFO: []
2021-04-12 20:39:02,057 - mu.logic:1396(show_admin) INFO: Showing admin with logs from C:\Users\Michelle\AppData\Local\python\mu\Logs\mu.log
2021-04-12 20:39:02,058 - mu.virtual_environment:702(installed_packages) INFO: Discovering installed third party modules in venv.
2021-04-12 20:39:05,067 - mu.virtual_environment:723(installed_packages) INFO: []

ntoll, 2021-07-10:
This feels like a bug in Python or PyQt (the error reported is that *Python* has stopped responding - and this is backed up by evidence because Python hasn't written anything to the log file to indicate the problem it encountered).

I wonder if there's a core dump somewhere (clutching at straws here) that we can examine to figure out the underlying problem? There's certainly nothing visible in the Mu layer of the application (hence my thought about the problem being in the Python or Qt layer). I also notice that the error happens when the OS's file dialog is displayed / triggered by Python... again, not a Mu thing per-se... Mu is just using the standard PyQt method to trigger the dialog.

ntoll, 2021-07-10:
A further thought.... perhaps it's something in PyQt and it won't be fixed until we update to, say, latest PyQt5 or the new PyQt6.