mu-editor/mu #640: Feature request: terminal remember zoom, easier resize terminal

grofte, 2018-09-08:
Dear Mu team, 

I've tested this on Linux and Mac. 

When presenting things to students in the Mu Editor we have an issue where the terminal resets zoom and its area each time we stop and then run the code again.  
- It would be great if **zoom** was remembered from ctrl + '+' or from the "Zoom +" / "Zoom -" buttons.  
- It would be great if it were easier to **catch the pixel line** to resize the terminal.  
- It would be great if the terminal remembered how I **resized** it between runs of my code.  

I am not asking for Mu Editor to remember anything after I shut it down. It should use the default values when I start it. But not load the default values every time we press the "Run"/"Stop" button or use F5.

grofte, 2018-09-08:
Actually, if the terminal just opens with the same zoom as the code writing area then that would be even nicer.

ZanderBrown, 2018-09-11:
@ntoll is that 3 now?

asweigart, 2018-10-26:
I'm on Windows 10. Ideally, it'd be nice if the interactive shell pane's zoom was the same as the currently focused file editor tab. I have tab A focused and at 100% zoom, and tab B at 200% zoom, and the interactive shell pane at 100% zoom. I switch to tab B, and then click Zoom Out to shrink it to 100%, but the interactive shell pane is now less than 100% too.

I'm not sure if this should be filed as a separate issue.

jesperp, 2018-12-16:
I cannot stress enough how annoying this is! I use a projector in my teaching. Everytime I click run I immediately have to Zoom-in for the students to see the output in the console.

Not sure what the best UX would be but remembering same zoom size on all panels would be nice!

ntoll, 2018-12-18:
Fixed in 5b82f22

commit 5b82f226 (2018-12-18): Improved zoom. Fixes #640 #614 #553.
closed 2018-12-18 by ntoll