mu-editor/mu #641: Mu exiting immediately after launching on MacOS

grofte, 2018-09-08:
Dear Mu team, 

Two of my students were unable to launch Mu after installing it on Macs. Specifically, the install goes fine, they accept the "run programs from the internet", and then the program immediately exits again (the icon shows up in the launch bar and then disappears again). It worked for everybody else.   
I would very much like to attach a log, but I don't know where the log file is.  

This is the closest I have to a log or an error message: 

```Last login: Sat Sep  8 10:36:18 on console

/Applications/mu-editor.app/Contents/MacOS/mu-editor ; exit;

Sara-Maries-MacBook-Pro:~ saramariebostefansen$ /Applications/mu-editor.app/Contents/MacOS/mu-editor ; exit;

Traceback (most recent call last):

  File "/Applications/mu-editor.app/Contents/MacOS/mu-editor", line 20, in <module>

    load_entry_point('mu-editor==1.0.0', 'console_scripts', 'mu-editor')()

  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/site-packages/pkg_resources/__init__.py", line 480, in load_entry_point

  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2693, in load_entry_point

  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2324, in load

  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2330, in resolve

  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 29, in <module>

    from PyQt5.QtCore import QTimer, Qt

ImportError: dlopen(/Applications/mu-editor.app/Contents/Resources/app_packages/PyQt5/QtCore.so, 2): Symbol not found: __os_log_default

  Referenced from: /Applications/mu-editor.app/Contents/Resources/app_packages/PyQt5/Qt/lib/QtCore.framework/Versions/5/QtCore (which was built for Mac OS X 10.11)

  Expected in: /usr/lib/libSystem.B.dylib

 in /Applications/mu-editor.app/Contents/Resources/app_packages/PyQt5/Qt/lib/QtCore.framework/Versions/5/QtCore

logout

[Proces udført]```

ntoll, 2018-09-18:
Hi,

What version of OSX are these students using? It looks like the answer may be in the error message:

```
  Referenced from: /Applications/mu-editor.app/Contents/Resources/app_packages/PyQt5/Qt/lib/QtCore.framework/Versions/5/QtCore (which was built for Mac OS X 10.11)
```

I believe this suggests the version they're on is not 10.11 or above.

carlosperate, 2020-09-07:
I'll close this issue, as it was caused by using an unsupported version of macOS, but please feel free to reopen if you have any further questions.

closed 2020-09-07 by carlosperate