mu-editor/mu #642: Mu fails to flash if you plug in a microbit after mu has started

rhubarbdog, 2018-09-10:
I don't know if this is local to my setup, I initially installed mu by cloning the main repo. following these steps  https://ubuntuforums.org/showthread.php?t=2377314 . I have attempted a new installation but am encountering problems creating a virtual environment and i'm reluctant to install it by just typing `sudo -H pip3 install mu-editor` as your requirement for a virtual environment makes me concerned that installing mu may break my python installation.

i'm using version 1.0.0 of Mu. on a Linux mint 18.2 machine 
if I start Mu with no microbit attached
load or type in a program.
attach a microbit and wait until my system has settled 
Mu successfully reports it's detected a new microbit
I click Flash, it hangs for a while then a dialog pops up reporting there has been a problem flashing the code to the microbit.

You have to load a program for this to happen, if the programs were automatically loaded at boot then they flash ok.

Here's the logs, I had to restart Mu as my first attempt to copy the logs failed. So the end of the log reports the successful auto-load of 2 programs.
 
2018-09-10 09:02:53,155 - root:112(run) INFO: 

-----------------

Starting Mu 1.0.0
2018-09-10 09:02:53,177 - root:113(run) INFO: uname_result(system='Linux', node='samsung01', release='4.13.0-32-generic', version='#35~16.04.1-Ubuntu SMP Thu Jan 25 10:13:43 UTC 2018', machine='x86_64', processor='x86_64')
2018-09-10 09:02:53,177 - root:114(run) INFO: Python path: ['/home/usrlocal/local/bin/mu-master', '/usr/lib/python35.zip', '/usr/lib/python3.5', '/usr/lib/python3.5/plat-x86_64-linux-gnu', '/usr/lib/python3.5/lib-dynload', '/usr/local/lib/python3.5/dist-packages', '/usr/lib/python3/dist-packages', '/usr/local/lib/python3.5/dist-packages/IPython/extensions']
2018-09-10 09:02:53,898 - mu.logic:540(__init__) INFO: Setting up editor.
2018-09-10 09:02:53,908 - mu.logic:558(__init__) INFO: Settings path: /home/phil/.local/share/mu/settings.json
2018-09-10 09:02:53,909 - mu.logic:559(__init__) INFO: Session path: /home/phil/.local/share/mu/session.json
2018-09-10 09:02:53,909 - mu.logic:560(__init__) INFO: Log directory: /home/phil/.cache/mu/log
2018-09-10 09:02:53,909 - mu.logic:561(__init__) INFO: Data directory: /home/phil/.local/share/mu
2018-09-10 09:02:53,946 - mu.logic:574(setup) INFO: Available modes: adafruit, debugger, pygamezero, microbit, python
2018-09-10 09:02:54,649 - mu.logic:1110(change_mode) INFO: Workspace directory: /home/phil/mu_code
2018-09-10 09:02:54,666 - mu.logic:623(restore_session) INFO: Restoring session from: /home/phil/.local/share/mu/session.json
2018-09-10 09:02:54,667 - mu.logic:624(restore_session) DEBUG: {'mode': 'microbit', 'envars': [], 'microbit_runtime': '', 'minify': True, 'theme': 'day', 'paths': []}
2018-09-10 09:02:54,667 - mu.logic:645(restore_session) INFO: Loaded files.
2018-09-10 09:02:54,667 - mu.logic:649(restore_session) INFO: User defined environment variables: []
2018-09-10 09:02:54,667 - mu.logic:653(restore_session) INFO: Minify scripts on micro:bit? True
2018-09-10 09:02:55,003 - mu.logic:671(restore_session) INFO: Starting with blank file.
2018-09-10 09:02:55,051 - mu.logic:1110(change_mode) INFO: Workspace directory: /home/phil/mu_code
2018-09-10 09:07:09,461 - mu.interface.main:244(get_save_path) DEBUG: Getting save path: /home/phil/mu_code/buttons.py
2018-09-10 09:07:09,462 - mu.logic:874(check_for_shadow_module) INFO: Checking path "/home/phil/mu_code/buttons.py" for shadow module.
2018-09-10 09:07:09,462 - mu.logic:840(save_tab_to_file) INFO: Saving script to: /home/phil/mu_code/buttons.py
2018-09-10 09:07:09,462 - mu.logic:841(save_tab_to_file) DEBUG: from microbit import *

while True:
    if button_a.is_pressed():
        display.show("A")
    elif button_b.is_pressed():
        display.show("B")
    else:
        dispaly.show("-")
        
    sleep(100)
2018-09-10 09:07:17,340 - mu.modes.microbit:239(flash) INFO: Preparing to flash script.
2018-09-10 09:07:17,340 - mu.modes.microbit:249(flash) DEBUG: Python script:
2018-09-10 09:07:17,341 - mu.modes.microbit:250(flash) DEBUG: b'from microbit import *\n\nwhile True:\n    if button_a.is_pressed():\n        display.show("A")\n    elif button_b.is_pressed():\n        display.show("B")\n    else:\n        dispaly.show("-")\n        \n    sleep(100)'
2018-09-10 09:07:17,357 - mu.modes.microbit:297(flash) INFO: Path to micro:bit: /media/phil/MICROBIT
2018-09-10 09:07:17,380 - mu.modes.base:217(find_device) INFO: Found device on port: ttyACM0
2018-09-10 09:07:17,381 - mu.modes.base:218(find_device) INFO: Serial number: 9900023431634e45003f40010000003800000000cca828bd
2018-09-10 09:07:17,381 - mu.modes.microbit:302(flash) INFO: Serial port: /dev/ttyACM0
2018-09-10 09:07:17,381 - mu.modes.microbit:303(flash) INFO: Device serial number: 9900023431634e45003f40010000003800000000cca828bd
2018-09-10 09:07:17,381 - mu.modes.microbit:328(flash) INFO: Checking target device.
2018-09-10 09:07:18,542 - mu.modes.microbit:364(flash) WARNING: Could not detect version of MicroPython.
2018-09-10 09:07:18,561 - mu.modes.microbit:381(flash) INFO: Flashing new MicroPython runtime onto device
2018-09-10 09:07:29,055 - mu.modes.microbit:505(copy_main) INFO: Copying main.py onto device
2018-09-10 09:07:29,055 - mu.modes.microbit:515(copy_main) INFO: ["fd = open('main.py', 'wb')", 'f = fd.write', "f(b'from microbit import *\\n\\nwhile True:\\n    if button_a.is_pressed()')", 'f(b\':\\n        display.show("A")\\n    elif button_b.is_pressed():\\n    \')', 'f(b\'    display.show("B")\\n    else:\\n        dispaly.show("-")\\n      \')', "f(b'  \\n    sleep(100)')", 'fd.close()']
2018-09-10 09:07:29,563 - mu.modes.microbit:518(copy_main) INFO: (b'', b'')
2018-09-10 09:07:35,051 - mu.logic:840(save_tab_to_file) INFO: Saving script to: /home/phil/mu_code/buttons.py
2018-09-10 09:07:35,051 - mu.logic:841(save_tab_to_file) DEBUG: from microbit import *

while True:
    if button_a.is_pressed():
        display.show("A")
    elif button_b.is_pressed():
        display.show("B")
[... 136 lines cut ...]
2018-09-10 09:45:38,473 - mu.logic:274(read_and_decode) INFO: Decoded with utf-8
2018-09-10 09:45:38,473 - mu.logic:287(read_and_decode) DEBUG: Detected newline '\n'
2018-09-10 09:45:38,473 - mu.logic:780(_load) DEBUG: # Write your code here :-)
from microbit import *
display.scroll("Phil")
2018-09-10 09:45:38,669 - mu.logic:706(_load) INFO: Loading script from: /home/phil/mu_code/buttons.py
2018-09-10 09:45:38,670 - mu.logic:271(read_and_decode) DEBUG: Trying to decode with utf-8
2018-09-10 09:45:38,670 - mu.logic:274(read_and_decode) INFO: Decoded with utf-8
2018-09-10 09:45:38,670 - mu.logic:287(read_and_decode) DEBUG: Detected newline '\n'
2018-09-10 09:45:38,670 - mu.logic:780(_load) DEBUG: from microbit import *

while True:
    if button_a.is_pressed():
        display.show("A")
    elif button_b.is_pressed():
        display.show("B")
    else:
        display.show("-")
        
    sleep(100)
2018-09-10 09:45:38,686 - mu.logic:645(restore_session) INFO: Loaded files.
2018-09-10 09:45:38,686 - mu.logic:649(restore_session) INFO: User defined environment variables: []
2018-09-10 09:45:38,686 - mu.logic:653(restore_session) INFO: Minify scripts on micro:bit? True
2018-09-10 09:45:38,716 - mu.logic:1110(change_mode) INFO: Workspace directory: /home/phil/mu_code
2018-09-10 09:45:43,345 - mu.logic:1037(show_admin) INFO: Showing logs from /home/phil/.cache/mu/log/mu.log

carlosperate, 2018-10-30:
@rhubarbdog are you still experiencing this issue?
If you are on Linux you might have to add yourself to the dialout group. If you are already, make sure nothing else in the system is opening a serial connection to the micro:bit.

rhubarbdog, 2018-11-22:
I think you can probably close this as it doesn't reproduce readily. I think it's an issue with `udev` on Linux. Sometimes it leaves the auto mounter running holding a lock on `/dev/ttyACM0`. Usually  I use `uflash` and `microfs` and sometimes i experience issues when running `ufs put ...` too soon after plugging a microbit in.
It may be worth adding a note to Linux and raspberry pi documentation that should a microbit not attatch to mu to wait for a few minutes (2-5) and plug it back in again. The solution uses `sudo lsof` and `sudo kill -term ...`

carlosperate, 2020-08-03:
I'll close this issue as it's been a year an half and it looks like it might be an issue with udev in linux.
Thank your for the report @rhubarbdog, if you think if any other action is required please do reopen the ticket.

closed 2020-08-03 by carlosperate