mu-editor/mu #643: Problem installing Mu Editor on Raspberry Pi: The 'pgzero' distribution was not found and is required by mu-editor

murilopolese, 2018-09-10:
I'm trying to install Mu Editor on Raspberry Pi KanoOS (derivative of Raspian). I'm running its latest version which is based on Raspian Stretch.

The output of `uname -a` is

```
Linux Kano 4.14.34-v7+ #1110 SMP Mon Apr 16 15:18 BST arm71 GNU/Linux
```

I ran both `sudo apt-get update` and `sudo apt-get dist-upgrade` and the first problem is that Mu is not listed or named as `mu-editor` as the on [instructions say](https://codewith.mu/en/howto/install_raspberry_pi) but just `mu`. 

So I ran `sudo apt-get install mu` and I got:

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  fonts-lyx ipython3 libjs-jquery-ui libqt5clucene5 libqt5designer5
  libqt5help5 libqt5serialport5 libqt5sql5 libqt5test5 libqt5xml5
  python-matplotlib-data python-pexpect python-ptyprocess
  python3-cycler python3-dateutil python3-decorator python3-ipykernel
  python3-ipython python3-ipython-genutils python3-jupyter-client
  python3-jupyter-core python3-matplotlib python3-numpy
  python3-pexpect python3-pickleshare python3-prompt-toolkit
  python3-ptyprocess python3-pycodestyle python3-pyflakes
  python3-pygments python3-pyparsing python3-pyqt5 python3-pyqt5.qsci
  python3-pyqt5.qtserialport python3-pyqt5.qtsvg python3-qtconsole
  python3-serial python3-simplegeneric python3-sip python3-six
  python3-tornado python3-traitlets python3-tz python3-wcwidth
  python3-zmq ttf-bitstream-vera
Suggested packages:
  libjs-jquery-ui-docs python-pexpect-doc python-cycler-doc dvipng
  ghostscript inkscape python-matplotlib-doc python3-cairocffi
  python3-gi python3-gi-cairo python3-gobject python3-nose
  python3-pyqt4 python3-scipy texlive-extra-utils texlive-latex-extra
  ttf-staypuft gcc gfortran python-numpy-doc python3-dev
  python3-numpy-dbg python-pyparsing-doc python3-pyqt5-dbg
  python3-wxgtk3.0 | python3-wxgtk python3-pycurl python-tornado-doc
  python3-twisted
Recommended packages:
  javascript-common libqt5sql5-sqlite | libqt5sql5-mysql
  | libqt5sql5-odbc | libqt5sql5-psql | libqt5sql5-tds
  | libqt5sql5-ibase python3-pgzero python3-pil pyflakes3
The following NEW packages will be installed:
  fonts-lyx ipython3 libjs-jquery-ui libqt5clucene5 libqt5designer5
  libqt5help5 libqt5serialport5 libqt5sql5 libqt5test5 libqt5xml5 mu
  python-matplotlib-data python-pexpect python-ptyprocess
  python3-cycler python3-dateutil python3-decorator python3-ipykernel
  python3-ipython python3-ipython-genutils python3-jupyter-client
  python3-jupyter-core python3-matplotlib python3-numpy
  python3-pexpect python3-pickleshare python3-prompt-toolkit
  python3-ptyprocess python3-pycodestyle python3-pyflakes
  python3-pygments python3-pyparsing python3-pyqt5 python3-pyqt5.qsci
  python3-pyqt5.qtserialport python3-pyqt5.qtsvg python3-qtconsole
  python3-serial python3-simplegeneric python3-sip python3-six
  python3-tornado python3-traitlets python3-tz python3-wcwidth
  python3-zmq ttf-bitstream-vera
```

After it finished installing all the dependencies listed above I went to run Mu on the terminal by calling `mu` but I got:

```
Traceback (most recent call last):
  File "/usr/bin/mu", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3019, in <module>
    @_call_aside
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3003, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3032, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 655, in _build_master
    ws.require(__requires__)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 963, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 849, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'pgzero' distribution was not found and is required by mu-editor
```
I noticed that `python3-pgzero` is on the "recommended" packages and after running `sudo apt-get install python3-pgzero` I could run `mu`.

Also, after running it I checked and I seem to be running `Mu 1.0.0.beta.15` which doesn't include a "Modes" button nor graph plotting.

ntoll, 2018-09-10:
This has, I believe, been fixed in upstream Raspbian. The RaspberryPi Foundation have certainly made sure that the version of Mu installed by apt is 1.0.0 (not the beta-15). Is your repository synced and up-to-date with upstream?

murilopolese, 2018-09-11:
Yeap, totally on our side. My bad not double checking it before. We mirror the Raspian and apparently we have frozen it for our next release in October. Sorry :x

closed 2018-09-11 by murilopolese