mu-editor/mu #652: Import issue

avbaker, 2018-09-13:
*trying to import code library "helper.py" from a file in the same directory as the code I am editing.
*using the code: import helper as h

*Mu will not compile the program with the error ModuleNotFoundError: No module named 'helper'

*the code runs fine using Thonny and IDLE, allowing me to use subroutines such as h.banner() and h.slowText() 

*the two files are in the same directory, but this is not the ~/mu_code folder. 

*I am reluctant to include the log, as it will include the code for the AQA coding project "celebrity dogs".

Using Windows 10, files stored in a directory that is syncing with Dropbox.

tjguk, 2018-09-14:
IIUC the request here is that, although the cwd is ~/mu_code, Mu should add os.path.dirname(os.path.abspath(__file__)) to sys.path?

I had this spiel written about how this wouldn't work normally in Python... and then I checked, and it does!

So +1 for adding os.path.dirname(os.path.abspath(__file__)) to sys.path by default!

tjguk, 2018-09-14:
(markup failure in previous post; the "file" inside brackets should be surrounded by double-underscores which I don't know how to escape)

ZanderBrown, 2018-09-14:
@tjguk `\_\_file\_\_` = \_\_file\_\_

carlosperate, 2018-09-17:
Just for reference, this was also suggested in https://github.com/mu-editor/mu/issues/480 (this issue covers two things, the "load" function to remember the last opened directory, and the fact that `open('file.txt')` default location is '~/mu_code/' instead of the Python script location).

bddicken, 2018-11-06:
I support this change as well.

I am a Lecturer at the University of Arizona and I suggested Mu to my students to use. I . did not know about this issue before the semester began. For several of the projects, students are required to import a graphics class that I wrote (my graphics class uses TKinter). To get around this, some students are adding the below two lines to their code:

```
import os
os.chdir(os.path.dirname(__file__))
```

It would be nice if students did not need to add this.

Are there plans to make this change in the near future? If someone could point me in right direction, I could work on this change.

wayoda, 2018-11-26:
I'm on Linux and don't have that problem but when I read about the topic on Gitter I tried a win10 install just out of curiosity. The problem is replicated here but could this message from the install-log provide a hint?

_BTW: installed Mu for the current user_
[image]

tjguk, 2018-11-26:
Thanks for picking that up @wayoda. That's actually a *different* issue which we know about (#682) and has been fixed upstream.

Whit this issue, we're fairly clear what the cause is and the fix -- someone just has to put a PR in for it and test across all platforms. (It's somewhere on my to-do list but I certainly won't object if someone else gets there first...)

lawsie, 2018-11-26:
Just ran into this same issue - have a file called `shapes.py` with some classes in and cannot import it from another file also inside `mu_code` with the line `from shapes import Triangle, Rectangle, Oval`. This is on one of our RPF online courses, although the bug has only just been noticed because the course was written with IDLE as the example editor and a participant had an issue with Mu!

bddicken, 2019-01-04:
Any update on this?

Like I mentioned, I would be willing to write a fix, if someone could point me int he correct direction. I looked through some of the code, but could not determine where the fix needed to be made.

ntoll, 2019-01-05:
Yes... there is a fix awaiting testing by folks who are still on holiday. Impending 1.0.2 release will fix this problem.

bddicken, 2019-01-06:
Oh, #726. Thanks @ntoll.

tmontes, 2019-06-01:
Dear all, and @avbaker in particular,

This issue is quite old by now and I suppose may have been fixed with PR #726, included in Mu 1.0.2 released back in January. Can you please confirm that is the case and, if so, that this issue may be closed?

Thanks.

dybber, 2020-06-13:
I can just confirm that this now works in 1.1.0.alpha.2. There hasn't been any comments on this since @tmontes asked about closing a year ago, so I presume it's okay that I'm closing it now. If you still think there's a problem @avbaker feel free to reopen.

closed 2020-06-13 by dybber