mu-editor/mu #654: The "Check" tool should check for trailing spaces

aoloe, 2018-09-14:
Currently, there seems to be no way to see if trailing spaces are left at the end of a line.

I think that the "Check" tool could highlight them...

tjguk, 2018-09-14:
I'm -0.5 on this: the check tool generates a lot of visual noise and it can be discouraging to beginners / youngsters if it warns about something which won't really affect your program.

I think we should definitely warn about things which will or could cause failure; I'm less sure about things which are cosmetic, no matter how useful in the bigger picture.

carlosperate, 2018-09-17:
I agree with @tjguk here, if the check tool provides a large list of "issues" it can be very discouraging for new programmers.
We should try to focus it on things that will cause problems running the code, as the shorter the list the more likely they are to "trust it" (specially compared to having a long list and the program running "just fine") and the more likely they are to take action.

carlosperate, 2018-09-17:
All that being said, I hope you don't feel we don't value your suggestion, because we really do!
It is immensely important for us to hear people's experience using Mu, and we aim to always have a constructive discussion about any changes introduced to the editor, as there is a very specific user target.

ntoll, 2018-09-17:
Hi @aoloe, *lots* feedback suggests that highlighting whitespace related problems is detrimental to learners. I'm afraid you're in the minority in this case, since many teachers reported that fixing whitespace problems was the least of their learners' worries. This was one of the most commented on "features" collated by the Raspberry Pi Foundation from feedback by teachers. Whitespace checking is definitely an "advanced user" feature, and Mu isn't primarily aimed at advanced users. This is why we turned off such whitespace reporting (and the feedback as a result was overwhelmingly positive).

I'll close this ticket now since the overwhelming feedback from users is NOT to switch on whitespace checking.

aoloe, 2018-09-17:
you're for sure right from that point of view.

i have been noticing that new coders are fighting with / against white spaces when joining or splitting lines.

learning not to leave them around, helps them having an environment with fewer surprises and have more confidence in the "quality" of their code.

but it's indeed something that is much more useful to me (who has a way of working that is different than the one of a beginner... and who is used to an editor that automatically removes spaces when joining lines...).
and i won't be using mu, once i've done my first project and will be able to teach it to our kids!

if you want, keep this issue in the back of your mind, and if you see people having issue with this, give it a second chance : - )

ntoll, 2018-09-17:
@aoloe absolutely... I hope I didn't sound dismissive, but you should have seen the teachers I was dealing with... they were quite adamant. ;-)

tjguk, 2018-09-17:
One thing I'd personally be open to is silently *stripping* trailing whitespace from lines. What do people feel about that?

aoloe, 2018-09-17:
i wonder what would be the right time to strip trailing spaces (moving to another line + saving?), but i agree that i don't see why somebody would want trailing spaces in a python script...

and, @ntoll: no worries: i've seen lot of my tickets being dismissed...  but it's makes me really happy, when they lead to propositions that are even better

dhalbert, 2018-09-17:
>One thing I'd personally be open to is silently stripping trailing whitespace from lines. What do people feel about that?

I have emacs set to do this on file save, and it has been perfect: never had to override, etc. No reason to mark it as an issue, just do it.

tjguk, 2018-09-18:
@ntoll do you want to pitch in on the "strip trailing spaces" idea? I'm happy to code up a PR if you're up for the notion.

ntoll, 2018-09-18:
@tjguk @dhalbert +1 on stripping trailing white spaces on click of "Check" button or "Save". (Anywhere else? Load?)

tjguk, 2019-02-17:
(Housekeeping) This was addressed by #677

closed 2019-02-17 by tjguk