mu-editor/mu #656: Two students' files went blank

greggraham, 2018-09-14:
I am teaching Python to a class of 11-12 year old boys using Mu 1.0.0 beta 16, Python 3 mode on Windows 10. Two of them today complained of their work disappearing, resulting in an empty file. I didn't see it happen, so I don't know what they did. By the time I got to their computers, they had already done some things to attempt to find their code. After class, after getting advice from gitter, I looked in the log files, and they appeared to contain the code, so I believe I was able to restore their work. I am opening this issue to document what happened.
----------------------------------------------------------------------
LOG FILE 1

2018-09-14 09:24:46,327 - root:111(run) INFO: 

-----------------

Starting Mu 1.0.0.beta.16
2018-09-14 09:24:46,327 - root:112(run) INFO: uname_result(system='Windows', node='LLab09', release='10', version='10.0.17134', machine='AMD64', processor='Intel64 Family 6 Model 60 Stepping 3, GenuineIntel')
2018-09-14 09:24:46,327 - root:113(run) INFO: Python path: ['C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\Python\\python36.zip', 'C:\\Program Files\\Mu\\Python', 'C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\pkgs\\IPython\\extensions']
2018-09-14 09:24:46,436 - mu.logic:542(__init__) INFO: Setting up editor.
2018-09-14 09:24:46,436 - mu.logic:556(__init__) INFO: Settings path: C:\Users\formii\AppData\Local\python\mu\settings.json
2018-09-14 09:24:46,436 - mu.logic:557(__init__) INFO: Session path: C:\Users\formii\AppData\Local\python\mu\session.json
2018-09-14 09:24:46,436 - mu.logic:558(__init__) INFO: Log directory: C:\Users\formii\AppData\Local\python\mu\Logs
2018-09-14 09:24:46,436 - mu.logic:559(__init__) INFO: Data directory: C:\Users\formii\AppData\Local\python\mu
2018-09-14 09:24:46,467 - mu.logic:572(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2018-09-14 09:24:47,076 - mu.logic:1077(change_mode) INFO: Workspace directory: C:\Users\formii\mu_code
2018-09-14 09:24:47,076 - mu.logic:621(restore_session) INFO: Restoring session from: C:\Users\formii\AppData\Local\python\mu\session.json
2018-09-14 09:24:47,076 - mu.logic:622(restore_session) DEBUG: {'theme': 'night', 'mode': 'python', 'paths': ['Z:\\Menereym\\mad lib 2025.py'], 'envars': [], 'minify': False, 'microbit_runtime': ''}
2018-09-14 09:24:47,076 - mu.logic:704(_load) INFO: Loading script from: Z:\Menereym\mad lib 2025.py
2018-09-14 09:24:47,092 - mu.logic:271(read_and_decode) DEBUG: Trying to decode with utf-8
2018-09-14 09:24:47,092 - mu.logic:274(read_and_decode) INFO: Decoded with utf-8
2018-09-14 09:24:47,092 - mu.logic:287(read_and_decode) DEBUG: Detected newline '\n'
2018-09-14 09:24:47,092 - mu.logic:778(_load) DEBUG: noun = "Stacy!"
verb = "oof"
blue = "dessert"
adjective = "disgusting"

print("Dominic was eating glue sticks in the cafeteria when "+noun+" "+verb+"ed")
print("into the room. 'Oh! Its  "+adjective+"' she said.") 
print("She promtly left the room.")
print("Guess nobody likes free "+blue+")
print("said Dominic.")
2018-09-14 09:24:47,639 - mu.logic:643(restore_session) INFO: Loaded files.
2018-09-14 09:24:47,639 - mu.logic:647(restore_session) INFO: User defined environment variables: []
2018-09-14 09:24:47,639 - mu.logic:651(restore_session) INFO: Minify scripts on micro:bit? False
2018-09-14 09:24:47,701 - mu.logic:1077(change_mode) INFO: Workspace directory: C:\Users\formii\mu_code
2018-09-14 09:25:52,325 - mu.interface.main:234(get_load_path) DEBUG: Getting load path: Z:/RomanS/soriano_project1.py
2018-09-14 09:25:52,325 - mu.logic:704(_load) INFO: Loading script from: Z:/RomanS/soriano_project1.py
2018-09-14 09:25:52,341 - mu.logic:271(read_and_decode) DEBUG: Trying to decode with utf-8
2018-09-14 09:25:52,341 - mu.logic:274(read_and_decode) INFO: Decoded with utf-8
2018-09-14 09:25:52,341 - mu.logic:287(read_and_decode) DEBUG: Detected newline '\n'
2018-09-14 09:25:52,341 - mu.logic:778(_load) DEBUG: name  = "Roman"
name_2 = "Roman's"
friend = "tHE vOICE IN THE hEAD"
adjective = "hot"
adjective_2 = "pen pineaple apple pen"
city = "Helsidalloustin City"
car = "Suziki instrument"

print(name + " went to " + city  +  " where " +  friend )
2018-09-14 09:26:32,574 - mu.modes.python3:191(run_script) DEBUG: name  = "Roman"
name_2 = "Roman's"
friend = "tHE vOICE IN THE hEAD"
adjective = "hot"
adjective_2 = "pen pineaple apple pen"
city = "Helsidalloustin City"
car = "Suziki instrument"

print(name + " went to " + city  +  " where " +  friend )
2018-09-14 09:26:32,574 - mu.interface.panes:624(start_process) INFO: Running script: z:\romans\soriano_project1.py
2018-09-14 09:26:32,574 - mu.interface.panes:626(start_process) INFO: Running with interactive mode.
2018-09-14 09:26:32,574 - mu.interface.panes:629(start_process) INFO: Command args: []
2018-09-14 09:26:32,574 - mu.interface.panes:647(start_process) INFO: Working directory: C:\Users\formii\mu_code
2018-09-14 09:26:32,574 - mu.interface.panes:652(start_process) INFO: Python path: ['C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\Python\\python36.zip', 'C:\\Program Files\\Mu\\Python', 'C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\pkgs\\IPython\\extensions']
2018-09-14 09:26:37,043 - mu.modes.python3:207(stop_script) DEBUG: Stopping script.
2018-09-14 09:26:42,808 - mu.logic:1099(autosave) INFO: Autosave has detected changes.
2018-09-14 09:26:42,808 - mu.logic:838(save_tab_to_file) INFO: Saving script to: Z:/RomanS/soriano_project1.py
2018-09-14 09:26:42,808 - mu.logic:839(save_tab_to_file) DEBUG: name  = "Roman"
[... 2349 lines cut ...]
2018-09-14 10:14:59,415 - root:111(run) INFO: 

-----------------

Starting Mu 1.0.0.beta.16
2018-09-14 10:14:59,415 - root:112(run) INFO: uname_result(system='Windows', node='LLab09', release='10', version='10.0.17134', machine='AMD64', processor='Intel64 Family 6 Model 60 Stepping 3, GenuineIntel')
2018-09-14 10:14:59,415 - root:113(run) INFO: Python path: ['C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\Python\\python36.zip', 'C:\\Program Files\\Mu\\Python', 'C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\pkgs\\IPython\\extensions']
2018-09-14 10:14:59,524 - mu.logic:542(__init__) INFO: Setting up editor.
2018-09-14 10:14:59,524 - mu.logic:556(__init__) INFO: Settings path: C:\Users\formii\AppData\Local\python\mu\settings.json
2018-09-14 10:14:59,524 - mu.logic:557(__init__) INFO: Session path: C:\Users\formii\AppData\Local\python\mu\session.json
2018-09-14 10:14:59,524 - mu.logic:558(__init__) INFO: Log directory: C:\Users\formii\AppData\Local\python\mu\Logs
2018-09-14 10:14:59,524 - mu.logic:559(__init__) INFO: Data directory: C:\Users\formii\AppData\Local\python\mu
2018-09-14 10:14:59,540 - mu.logic:572(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2018-09-14 10:15:00,274 - mu.logic:1077(change_mode) INFO: Workspace directory: C:\Users\formii\mu_code
2018-09-14 10:15:00,274 - mu.logic:621(restore_session) INFO: Restoring session from: C:\Users\formii\AppData\Local\python\mu\session.json
2018-09-14 10:15:00,274 - mu.logic:622(restore_session) DEBUG: {'theme': 'night', 'mode': 'python', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': ''}
2018-09-14 10:15:00,274 - mu.logic:643(restore_session) INFO: Loaded files.
2018-09-14 10:15:00,274 - mu.logic:647(restore_session) INFO: User defined environment variables: []
2018-09-14 10:15:00,274 - mu.logic:651(restore_session) INFO: Minify scripts on micro:bit? False
2018-09-14 10:15:00,852 - mu.logic:669(restore_session) INFO: Starting with blank file.
2018-09-14 10:15:00,899 - mu.logic:1077(change_mode) INFO: Workspace directory: C:\Users\formii\mu_code
2018-09-14 10:15:03,024 - mu.logic:956(show_help) INFO: Showing help.
2018-09-14 10:15:11,696 - mu.logic:999(quit) DEBUG: Session: {'theme': 'night', 'mode': 'python', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': ''}
2018-09-14 10:15:11,696 - mu.logic:1000(quit) DEBUG: Saving session to: C:\Users\formii\AppData\Local\python\mu\session.json
2018-09-14 10:15:11,696 - mu.logic:1002(quit) INFO: Quitting.

tjguk, 2018-09-16:
Well we're definitely seeing a UnicodeEncodeError on LOGFILE 1 (at 2018-09-14 09:57:37,230) where the students include some Korean? characters in the text. That's definitely a bug on our side, and I assume it's what's causing the issue since the file save will fail. I'm just checking but I'm also fairly sure we put a fairly late fix in for a unicode-related issue in certain circumstances, so it's quite possible that beta16 doesn't have that fix.

tjguk, 2018-09-16:
Hmm. Checking, I can't see a Unicode fix post-b16 which directly affects this branch, but we certainly have quite a few fixes (across one beta a a rc prior to v1.0) and at least one affected Unicode handling, so please do upgrade anyway. If I can, I'll try to use your students' code to reproduce the situation.

greggraham, 2018-09-16:
It is quite possible that they were trying to put Korean characters in their code. (One of them may be of Korean descent.) These are very clever boys; sometimes too clever for their own good. If that is the problem, I will just tell them not to do that until we get the fix.

tjguk, 2018-09-16:
I've just run the current (ie pre 1.0.1) version of Mu in circumstances which I *think* mimic your students' code and there doesn't seem to be a problem. When I get a little time I'll try to roll the code back to 1.0b16 and see if I can reproduce the crash. If so, we know where we are.

tjguk, 2018-11-25:
This is definitely a duplicate of #696 (in that we're regarding that as the master although it came later). And I can see that I'd already tested 1.0.1 here which I've just re-tested over there. I'm going to close this as a duplicate and carry on over there.

closed 2018-11-25 by tjguk