mu-editor/mu #660: Brazilian Portuguese Translation

oliveira-julio, 2018-09-22:
Hi.
In the next year i will teach a course of introduction to programming using Python. I will use the "systematic program design" of  the book "how to design programs". mu reminds me the IDE used in the book, the Dr.Racket. So i decided use him :D.

I want do the Brazilian Portuguese(pt-br) translation.
How exactly i make this?
Which tools i use to help me to translate?

malbarbo, 2018-09-23:
Hi @oliveira-julio , I made a [PR](https://github.com/mu-editor/mu/pull/613) adding Brazilian Portuguese translation some time ago, but it wasn't reviewed yet.

I'm teaching a course this semester using the "systematic program design" approach and the mu editor. I also wrote a simple image and universe module so the students can write some interactive programs. If you are interested, you can send me an email (my login at gmail).

ntoll, 2018-09-29:
All merged and will be in the upcoming 1.0.1 release. :-)

closed 2018-09-29 by ntoll