mu-editor/mu #661: Script is to long error despite stripping comments

kamar535, 2018-09-24:
Mu version: 1.0.0 running on MacOS.

I'm using the option Minify Python code before flashing.  

When trying to flash a 48 line script which repeats the following two lines:

```
# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.

````

, I'm getting the Script is to long error. After manually removing a number of comments the error goes away. I suspect the error check is done prior to stripping the comments. 

Mu log: 

```
2018-09-24 01:57:20,986 - mu.logic:706(_load) INFO: Loading script from: /Users/karl/mu_code/to-long.py
2018-09-24 01:57:20,987 - mu.logic:271(read_and_decode) DEBUG: Trying to decode with utf-8
2018-09-24 01:57:20,987 - mu.logic:274(read_and_decode) INFO: Decoded with utf-8
2018-09-24 01:57:20,988 - mu.logic:287(read_and_decode) DEBUG: Detected newline '\n'
2018-09-24 01:57:20,988 - mu.logic:780(_load) DEBUG: # Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.

# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.

# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.

# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.

# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.

# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.

# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.

# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.

# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.

# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.

# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.

# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.

# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.

# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.

# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.

# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pell [... cut ...]   """\\n    Creates \')', "f(b'a set with n random points (x, y).\\n\\n    Argument(s):\\n      n (in')", "f(b'teger) - Number of point to create.\\n\\n    Return value: A set wit')", 'f(b\'h n random points (x, y).\\n    """\\n\\n    return set(take(shuffle(a\')', "f(b'll_positions()), n))')", 'fd.close()']
2018-09-24 01:57:52,315 - mu.modes.microbit:518(copy_main) INFO: (b'', b'')
2018-09-24 01:57:56,801 - mu.logic:1037(show_admin) INFO: Showing logs from /Users/karl/Library/Logs/mu/mu.log
```

kamar535, 2018-09-24:
Related, issue #258 Remove comments before flashing program.

ZanderBrown, 2018-09-24:
Hello @kamar535!

I'm the author of Mu's comment-stripper 'nudatus' and in danger of sounding cliché it works on my machine :-)

Using your snippet:
```py
# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.
# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.
# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.
# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.
# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.
# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.
# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.
# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.
# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.
# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.
# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend lacus. Ut tristique leo elit, vel pellentesque diam hendrerit eget. Maecenas a venenatis massa. Integer feugiat nisl at orci egestas tempus. Aliquam faucibus nisl tellus, vitae tempus dui cursus in.
# Lorem ipsum dolor sit amet, consectetur adipiscing elit. Mauris ornare aliquam lectus in pretium. Morbi nec convallis turpis. Etiam nulla erat, tristique et dolor et, sollicitudin tempus dolor. Ut ex risus, cursus non auctor nec, laoreet at mi. In nec eleifend  [... cut ...]
2018-09-24 21:47:57,839 - mu.modes.microbit:274(flash) DEBUG: Script minified, 11064 bytes (99.78%) saved:
2018-09-24 21:47:57,840 - mu.modes.microbit:275(flash) DEBUG: b'\n\n\n\n\n\n\n\n\n\n\n\n\n\n\n\n\n\n\n\n\n\n\n\n'
```

So I'm wondering: How did you install Mu? Is nudatus available?

If nudatus is failing I would expect you to see a message like "Our minifier tried but your script is too long!" but if it's missing you should see "Your script is too long and the minifier isn't available"

Only other thing I can think of is that @ntoll 's new flashing system is doing something strange

kamar535, 2018-09-24:
To install Mu 1.0.0 I used the Mac OSX installer from https://codewith.mu/en/download

carlosperate, 2018-09-25:
Unfortunately Nudatus was accidentally not included in the mac version, there is a PR submitted to fix this: https://github.com/mu-editor/mu/pull/630

If you are familiar with installing python packages you can follow these instructions to install Mu in a virtual environment: https://mu.readthedocs.io/en/latest/setup.html

carlosperate, 2020-09-29:
This should have been fixed in #630 and shipped in subsequent releases, so I'll close this as resolved.

Thank you @kamar535 for the report! Please feel free to reopen if you are still experiencing this problem.

closed 2020-09-29 by carlosperate