mu-editor/mu #662: mu_code not included in sys.path on Windows 10

malbarbo, 2018-09-24:
Running

```
import sys

for p in sys.path:
    print(p)
```

on Linux prints

```
/home/malbarbo/mu_code
/usr/lib/python35.zip
/usr/lib/python3.5
/usr/lib/python3.5/plat-x86_64-linux-gnu
/usr/lib/python3.5/lib-dynload
/home/malbarbo/.local/lib/python3.5/site-packages
/usr/local/lib/python3.5/dist-packages
/usr/lib/python3/dist-packages
```

but on Windows 10 it prints

```
C:\Users\Marco\AppData\Local\Mu\Python\python36.zip
C:\Users\Marco\AppData\Local\Mu\Python\
C:\Users\Marco\AppData\Local\Mu\pkgs
```

The `mu_code` path is not included on windows, which means that files in `mu_code` dir cannot be imported.

ZanderBrown, 2018-09-24:
Duplicate of #652

carlosperate, 2018-09-26:
Also, as mentioned in https://github.com/mu-editor/mu/issues/612 there is a difference between debug and run.

Run:
```
C:\Program Files\Mu\Python\python36.zip
C:\Program Files\Mu\Python
C:\Program Files\Mu\pkgs
```

Debug:
```
c:\users\microbit-carlos\mu_code
C:\Program Files\Mu\Python
C:\Program Files\Mu\pkgs
C:\Program Files\Mu\pkgs\IPython\extensions
```

ntoll, 2018-12-20:
@carlosperate I can't see anything obvious in Mu code that adds the `mu_code` directory to sys.path when in the debugger. I guess this must be something Python does when it's being driven by the `bdb` (base debugger) class.

In any case, I've checked the above and can reproduce, so I'll apply a quick fix to do two things:

* Add the `mu_code` directory to the PYTHONPATH
* If the directory containing the Python script isn't `mu_code` add it to PYTHONPATH too.

This should ensure code in both `mu_code` and the current location of the script can be imported.

I have a feeling this is down to the way `pynsist` packages/configures Mu on Windows. I believe the Python instance is run in a locked down mode, thus excluding the more obvious directories from appearing on the path.

carlosperate, 2018-12-20:
Yes, when I had a look I didn't see anything obvious either that would explain the difference on Windows.
The fix you describe sounds good, and you also bring a good point about testing in both the pip installed Mu and the pynsist installer version as well, in case there are more differences.

We should also keep in mind the order that these paths are included in PYTHONPATH (and the order python searches for modules in this list), as the file location should take precedence over the 'mu_code' folder when looking for modules/packages.

Since you are looking into this, there was another issue about the default Current Working Directory, which is similar enough where maybe it's worth doing both around the same time? https://github.com/mu-editor/mu/issues/480

tim-mccurrach, 2018-12-20:
@ntoll :+1:

ntoll, 2018-12-20:
@carlosperate ack.

Looks like I'm taking Xmas debug requests... ;-)

tmontes, 2019-06-01:
Dear all, and @malbarbo in particular,

This issue is quite old by now and I suppose may have been fixed with PR #726, included in Mu 1.0.2 released back in January. Can you please confirm that is the case and, if so, that this issue may be closed?

Thanks.

carlosperate, 2020-09-29:
I can confirm this has been fixed in Windows.

Run mode:

```
c:\users\microbit-carlos\mu_code
C:\Users\carlos\AppData\Local\Mu\pkgs
C:\Users\carlos\AppData\Local\Mu
C:\Users\carlos\AppData\Local\Mu\Python\python36.zip
C:\Users\carlos\AppData\Local\Mu\Python\DLLs
C:\Users\carlos\AppData\Local\Mu\Python\lib
C:\Users\carlos\AppData\Local\Mu\Python
C:\Users\carlos\AppData\Local\python\mu\site-packages
C:\Users\carlos\AppData\Local\Mu\pkgs\IPython\extensions
```

Debug mode:

```
Running in debug mode. Use the Stop, Continue, and Step toolbar buttons to debug the script
c:\users\carlos\mu_code
C:\Users\carlos\AppData\Local\Mu\pkgs
C:\Users\carlos\AppData\Local\Mu
C:\Users\carlos\AppData\Local\Mu\Python\python36.zip
C:\Users\carlos\AppData\Local\Mu\Python\DLLs
C:\Users\carlos\AppData\Local\Mu\Python\lib
C:\Users\carlos\AppData\Local\Mu\Python
C:\Users\carlos\AppData\Local\python\mu\site-packages
C:\Users\carlos\AppData\Local\Mu\pkgs\IPython\extensions
C:\Users\carlos\AppData\Local\python\mu\site-packages
C:\Users\carlos\AppData\Local\Mu\pkgs\IPython\extensions
c:\users\carlos\appdata\local\mu\python\python36.zip
```

Thank you @malbarbo for the report and everybody that helped!

closed 2020-09-29 by carlosperate