mu-editor/mu #665: Windows 10 Intermittent Hangs in 1.0.1

TheKitty, 2018-10-03:
If you are reporting a bug, we would like to know:

* What you were trying to do, 
Use Mu 1.0.1 on Windows 10 latest build

* What steps you took to make this happen, 
normal get file, edit file, save file

* What you expected to happen, 
normal file handling

* What actually happened, 
Window becomes unresponsive, task pythonw reported by Win10 as taking 21% of CPU and task is "nonresponsive"

* Why this difference is problematic (it may not be a bug!),
Window hangs - can't do anything
[image]

* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 
Given above

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

2018-10-03 11:17:46,853 - root:112(run) INFO: 

-----------------

Starting Mu 1.0.1
2018-10-03 11:17:46,856 - root:113(run) INFO: uname_result(system='Windows', node='Dad-2012', release='10', version='10.0.17134', machine='AMD64', processor='Intel64 Family 6 Model 58 Stepping 9, GenuineIntel')
2018-10-03 11:17:46,856 - root:114(run) INFO: Python path: ['C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\Python\\python36.zip', 'C:\\Program Files\\Mu\\Python', 'C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\pkgs\\IPython\\extensions']
2018-10-03 11:17:46,856 - root:115(run) INFO: Language code: en_US
2018-10-03 11:17:47,199 - mu.logic:540(__init__) INFO: Setting up editor.
2018-10-03 11:17:47,200 - mu.logic:558(__init__) INFO: Settings path: C:\Users\Mike\AppData\Local\python\mu\settings.json
2018-10-03 11:17:47,200 - mu.logic:559(__init__) INFO: Session path: C:\Users\Mike\AppData\Local\python\mu\session.json
2018-10-03 11:17:47,200 - mu.logic:560(__init__) INFO: Log directory: C:\Users\Mike\AppData\Local\python\mu\Logs
2018-10-03 11:17:47,200 - mu.logic:561(__init__) INFO: Data directory: C:\Users\Mike\AppData\Local\python\mu
2018-10-03 11:17:47,230 - mu.logic:574(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2018-10-03 11:17:47,831 - mu.logic:1112(change_mode) INFO: Workspace directory: C:\Users\Mike\mu_code
2018-10-03 11:17:47,835 - mu.logic:623(restore_session) INFO: Restoring session from: C:\Users\Mike\AppData\Local\python\mu\session.json
2018-10-03 11:17:47,835 - mu.logic:624(restore_session) DEBUG: {'theme': 'day', 'mode': 'adafruit', 'paths': ['Z:\\Adafruit\\- Work Products\\Guide edits\\CRICKIT\\Intro Crickit\\microbit\\microbit-neo.py', 'Z:\\Adafruit\\- Work Products\\Guide edits\\Make it ___\\Pulse\\pulse.py', 'Z:\\Adafruit\\- Work Products\\Guide edits\\Make it ___\\Pulse\\df2t.py', 'Z:\\Adafruit\\- Work Products\\Guide edits\\Make it ___\\Pulse\\dft.py', 'Z:\\Adafruit\\- Work Products\\Guide edits\\Make it ___\\hot-cold\\code-cpx-final.py', 'I:\\code.py'], 'envars': [], 'minify': False, 'microbit_runtime': ''}
2018-10-03 11:17:47,835 - mu.logic:708(_load) INFO: Loading script from: Z:\Adafruit\- Work Products\Guide edits\Make it ___\Pulse\pulse.py
2018-10-03 11:17:47,843 - mu.logic:271(read_and_decode) DEBUG: Trying to decode with utf-8
2018-10-03 11:17:47,844 - mu.logic:274(read_and_decode) INFO: Decoded with utf-8
2018-10-03 11:17:47,845 - mu.logic:287(read_and_decode) DEBUG: Detected newline '\n'
2018-10-03 11:17:47,845 - mu.logic:782(_load) DEBUG: # Adafruit Circuit Playground Express Pulse Monitor
#
# This code uses the light sensor built in to Circuit Playground Express
# to detect the pulse of a finger.
# Save this code as code.py on a Circuit Playground Express board running
# CircuitPython (see https://github.com/adafruit/circuitpython).  You will
# also need to load neopixel.mpy onto the board's /lib folder (from
# https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/releases).
#
# When the first three NeoPixels light up white you're ready to read -
# the light sensor (look for the eye graphic on the board, it's right
# below the three lit NeoPixels) and look at the serial terminal from the board
# at 115200 baud to see pulse rate printed.  This works best
# holding your finger perpendicular to the sensor, like:
#       ||
#       || <- Finger
#       ||
#    ________  <- Circuit Playground
#
# Author: Mike Barela for Adafruit Industries, 9/2018
# License: MIT License (https://en.wikipedia.org/wiki/MIT_License)
import time
[... 624 lines cut ...]
        if p1 == midpoint or p0 < midpoint < p1 or p0 > midpoint > p1:
            crossings += 1
#        print((readings[i]-midpoint,))
        time.sleep(0.1)
    # Finally use the number of crosssings and the amount of time in the sample
    # window to calculate how many times the pulse was detected with
    # sensor.  Use that period to calculate frequency (beats per second)
    # and BPM (beats per minute).
    period = elapsed / (crossings / 2.0)
    frequency = 1.0/period
    bpm = frequency * 60.0
    print('Frequency: {0:0.5} (hz)\t\tBPM: {1:0.5}\t\tPeriod: {2:0.5} (seconds)'.format(frequency, bpm, period))
2018-10-03 11:17:48,523 - mu.logic:645(restore_session) INFO: Loaded files.
2018-10-03 11:17:48,523 - mu.logic:649(restore_session) INFO: User defined environment variables: []
2018-10-03 11:17:48,523 - mu.logic:653(restore_session) INFO: Minify scripts on micro:bit? False
2018-10-03 11:17:48,583 - mu.interface.main:711(show_message) DEBUG: Could not find an attached Adafruit CircuitPython device.
2018-10-03 11:17:48,583 - mu.interface.main:712(show_message) DEBUG: Python files for Adafruit CircuitPython devices are stored on the device. Therefore, to edit these files you need to have the device plugged in. Until you plug in a device, Mu will use the directory found here:

 C:\Users\Mike\mu_code

...to store your code.
2018-10-03 11:17:50,363 - mu.logic:1112(change_mode) INFO: Workspace directory: C:\Users\Mike\mu_code
2018-10-03 11:17:51,656 - mu.logic:1039(show_admin) INFO: Showing logs from C:\Users\Mike\AppData\Local\python\mu\Logs\mu.log

Thank you for contributing to Mu! :-)

TheKitty, 2018-10-03:
On a hunch I unplugged my Circuit Playground Express, Mu hang corrected itself and it opened the load file dialog. There must be something in Windows USB that is causing pythonw to hang and not open file dialog boxes

tjguk, 2018-10-03:
Well this is going to be fun to debug! It would be interesting to know whether anyone else has experienced this with CircuitPlayground. Have you asked on CP forums? (I assume there are some).

Can you tell me whether it starts to respond if you leave it for long enough? I don't know how long you waited, but will it come back if you wait 1 minute? 5  minutes? 15?

The fact that it's consuming CPU is interesting because the natural hypothesis is that it's trying to talk to the filesystem on the CP device and waiting for some kind of ack. But by definition that's IO-bound and I wouldn't expect to see if chewing CPU

tjguk, 2018-10-03:
@ntoll Do you have a Circuit Playground? If you bring it tomorrow I can try against my Win10 box to see if we can reproduce

carlosperate, 2018-10-03:
Long shot, but does problem still happen if you turn off malwarebytes? Not that you should, but just trying to eliminate that as a possible cause.

dhalbert, 2018-10-03:
We also have issues with Kaspersky, AIDA64, and Hard Disk Sentinel.

carlosperate, 2018-10-03:
> We also have issues with Kaspersky, AIDA64, and Hard Disk Sentinel.

Do you mean you've seen this issue in Mu when run alongside those applications? Or CircuitPython in general?

dhalbert, 2018-10-03:
> Do you mean you've seen this issue in Mu when run alongside those applications? Or CircuitPython in general?

There are problems with CircuitPython and/or the UF2 bootloader mock disk. Either the UF2 `...BOOT` drive does not appear, or `CIRCUITPY` does not appear, or the whole machine hangs. AIDA64 had trouble with UF2, but they were responsive, bought hardware, and believe they've fixed the problem.

carlosperate, 2018-10-03:
Oh, I see what you mean now, thanks for the clarification.

carlosperate, 2021-12-29:
It looks like this issue wasn't related to Mu and was fixed in CircuitPython a while ago, so I think we can close this one. Thanks everyone!

closed 2021-12-29 by carlosperate