mu-editor/mu #666: The Mu editor is not responding.

sanchit002sanchit002, 2018-10-08:
If you are reporting a bug, we would like to know:

* What you were trying to do,
* What steps you took to make this happen,
* What you expected to happen,
* What actually happened,
* Why this difference is problematic (it may not be a bug!),
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

ntoll, 2018-10-08:
Hi @sanchit002sanchit002,

Could you please provide some more details -- for example, as described in the description area of the bug report. Once we have more information, we'd love to help.

Thank you!

sanchit002sanchit002, 2018-10-08:
The mu editor is not responding.
Hi sir / mam,
I have installed mu for the first time.
after installing I've chosen python 3 mode, then mu launched and the mu editor is not responding.
I've been waiting for long time, But nothing happened
[image]
.
so When I forced to close the mu, a window pops up the image is attached.
I have python 3.7 already installed and I'm using it with LiClips as my IDE.
I'm running on my machine dell XPS 12 9q23.
Your help will be appreciated.
Thank you.

ntoll, 2018-10-08:
Hi @sanchit002sanchit002,

Thank you so much for your information and screen shot.

May I ask how you installed Mu? Was it from the official installer downloaded from the Mu website?

I also notice that you're using JAWS screen reader. Do you have any other screen readers installed which you could try. We've noticed that some readers work better than others when it comes to working with Mu and Python.

Finally, the error message you're seeing is because Python (which is also the language Mu is written in) has encountered a problem, rather than Mu having a problem itself. This more fundamental error suggests to me it's something like an issue with JAWS interacting with the Python language.

Please let me know how you get on with another screen reader.

Perhaps @bmustill-rose might be able to suggest another screen reader..?

sanchit002sanchit002, 2018-10-08:
since mu is not responding how should I attach the logs ?
I am running windows 10 1803 64 bit

sanchit002sanchit002, 2018-10-08:
Yes, I have downloaded from official website.
I also tried with nvda.
But some times nvda is also stops responding when loaded with mu.

carlosperate, 2018-10-08:
Does that mean Mu does sometimes work with NVDA?

Would you be able to check if Mu is responsive without any screen reader at all?
Knowing if Mu opens correctly without one helps us immensely to reduce the debugging scope. 

> Finally, the error message you're seeing is because Python (which is also the language Mu is written in) has encountered a problem, rather than Mu having a problem itself. This more fundamental error suggests to me it's something like an issue with JAWS interacting with the Python language.

While I agree, we also need to keep in mind that if the Mu code locks up for whatever reason (I had it happen during development when I made some silly mistakes), the same type of error appears where it is  reported that Python is not responding.

sanchit002sanchit002, 2018-10-08:
Sir,
I’ve tried with NVDA as well.
But when I am running Mu, NVDA is also not responding and sometimes laptop is getting slow down.

From: Nicholas Tollervey
Sent: 08 October 2018 19:10
To: mu-editor/mu
Cc: sanchit; Mention
Subject: Re: [mu-editor/mu] The Mu editor is not responding. (#666)

Hi @sanchit002sanchit002,
Thank you so much for your information and screen shot.
May I ask how you installed Mu? Was it from the official installer downloaded from the Mu website?
I also notice that you're using JAWS screen reader. Do you have any other screen readers installed which you could try. We've noticed that some readers work better than others when it comes to working with Mu and Python.
Finally, the error message you're seeing is because Python (which is also the language Mu is written in) has encountered a problem, rather than Mu having a problem itself. This more fundamental error suggests to me it's something like an issue with JAWS interacting with the Python language.
Please let me know how you get on with another screen reader.
Perhaps @bmustill-rose might be able to suggest another screen reader..?

dangmanhcuong87, 2018-10-10:
Hi all. I've tried using MU twice with NVDA, and it was crashed once. Beside, it was crash all times using Jaws.