mu-editor/mu #674: Fix byte decoding of stdout from subprocess

ntoll, 2018-10-14:
Try this in Python3 mode:

```
print(b'\xe2\x94\x80\xe2'.decode('utf-8'))
```

The code which led to this problem is:

```
def menu():
    print('┌──────────────────────────┐')
    print('│ The Treasure Hunt │')
    print('│ 1. Play │')
    print('│ 2. Help │')
    print('│ 3. Quit │')
    print('└──────────────────────────┘')

menu()
```

The problem is byte stream from the subprocess only reads 256 bytes at a time and the byte is thus corrupted.

tim-mccurrach, 2018-10-14:
I’ve written a fix for this problem, but in the process discovered a new related bug. Potential fixes for the new bug, might circumnavigate the original bug so I’ve held off on a pull request for now. (I also wasn’t sure about opening a separate issue, as they both relate to the same fundamental issue, and this would risk fixing 2 separate symptoms rather than the original cause.)

**The new bug** 
The following code produces differing results when run multiple times (on v1.01, on windows 64bit both production and development versions):
```
a="a"*300
print(a)
```
Some (most) of the time it will print the full string, along with the new line plus ">>>" prompt. However, other times, it will only print a certain section of the string and stop short (presumably the first 256 characters). In such cases pressing enter again will cause the remainder (or for really long strings, the next section of the string) to be printed, and there will be multiple ">>>" prompts appended to the end of the string. Behaviour becomes increasingly unpredictable when we replace 300 with larger numbers.

It seems like `read_from_stdout` does not always get called when there is still data ready to be read. For what it’s worth, the code below fixes the original problem, but might not be necessary if a fix to the above didn’t involve reading in chunks:

```
def detect_missing_utf8_bytes(self, args):
    """
    Return the number of bytes missing from the end of a utf-8 encoded
    byte string. Return -1 if the byte string is ill-formed.
    """
    if args[-1] >> 7 == 0:  # Single byte character
        return 0
    for continuation_bytes in range(min(3, len(args))):
        if not args[-1 * continuation_bytes - 1] >> 6 == 2:
            break
    else:
        return -1
    initial_byte = args[-1 * continuation_bytes - 1]
    if initial_byte >> 5 == 6:  # 2-byte character
        character_length = 2
    elif initial_byte >> 4 == 14:  # 3-byte character
        character_length = 3
    elif initial_byte >> 3 == 30:  # 4-byte character
        character_length = 4
    else:
        return -1
    return character_length - continuation_bytes - 1

def read_from_stdout(self):
    """
    Process incoming data from the process's stdout.
    """
    data = self.process.read(256)
    if data:
        # Check the final character has not been 'cut in half' by the
        # 256 byte chunk. Read additional bytes from std_out if so.
        missing_bytes = self.detect_missing_utf8_bytes(data)
        if missing_bytes == -1:
            logger.info('Data from std_out ends with ill-formed utf-8 byte'
                        ' sequence: {}').format(data[-4:])
            return
        elif missing_bytes:
            data += self.process.read(missing_bytes)
        # append data
        self.append(data)
        self.on_append_text.emit(data)
        cursor = self.textCursor()
        self.start_of_current_line = cursor.position()
```

tim-mccurrach, 2018-10-18:
I’ve done some experimenting with the above, and it can be fixed relatively easily. However...

Whilst the Unicode bytes are dealt with, this still leaves some other situations where splitting the data causes problems:

- `/r/n` which should be one new line, can be split into `… /r` and `/n …` giving two new lines.
- Split ANSI codes are also going to cause a problem.

If the above alone are the only things to deal with, that’s not too much of a problem. I don’t know how many other situations there are that I’ve not thought of. I guess dealing with just the above is better than not dealing with them at all though??

ntoll, 2018-12-13:
Fixed by #723

closed 2018-12-13 by ntoll