mu-editor/mu #681: Code completion doesn't work with non ascii-characters

jonatanskogsfors, 2018-10-20:
I was trying out Mu and wrote som variables in my native language (Swedish). 

```
första_talet = 1
andra_talet = 2
svar = första_talet + andra_talet
```
 
When writing the last row I get code completion for `andra_talet` but not for `första_talet`. I expect code completion for all legal variables.

Unicode support for variables is a great thing in Python, especially for kids.

Environment:
mu-editor 1.0.1 
macOS Mojave (10.14)

Log:
```
2018-10-20 22:50:24,176 - root:112(run) INFO: 

-----------------

Starting Mu 1.0.1
2018-10-20 22:50:24,176 - root:113(run) INFO: uname_result(system='Darwin', node='wotan.local', release='18.0.0', version='Darwin Kernel Version 18.0.0: Wed Aug 22 20:13:40 PDT 2018; root:xnu-4903.201.2~1/RELEASE_X86_64', machine='x86_64', processor='i386')
2018-10-20 22:50:24,176 - root:114(run) INFO: Python path: ['/Applications/mu-editor.app/Contents/MacOS', '/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/mu-editor.app/Contents/Resources/app', '/Applications/mu-editor.app/Contents/Resources/app_packages', '/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']
2018-10-20 22:50:24,176 - root:115(run) INFO: Language code: en
2018-10-20 22:50:24,553 - mu.logic:540(__init__) INFO: Setting up editor.
2018-10-20 22:50:24,553 - mu.logic:558(__init__) INFO: Settings path: /Users/jonatan/Library/Application Support/mu/settings.json
2018-10-20 22:50:24,553 - mu.logic:559(__init__) INFO: Session path: /Users/jonatan/Library/Application Support/mu/session.json
2018-10-20 22:50:24,553 - mu.logic:560(__init__) INFO: Log directory: /Users/jonatan/Library/Logs/mu
2018-10-20 22:50:24,553 - mu.logic:561(__init__) INFO: Data directory: /Users/jonatan/Library/Application Support/mu
2018-10-20 22:50:24,571 - mu.logic:574(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2018-10-20 22:50:24,751 - mu.logic:1112(change_mode) INFO: Workspace directory: /Users/jonatan/mu_code
2018-10-20 22:50:24,755 - mu.logic:623(restore_session) INFO: Restoring session from: /Users/jonatan/Library/Application Support/mu/session.json
2018-10-20 22:50:24,755 - mu.logic:624(restore_session) DEBUG: {'theme': 'day', 'mode': 'python', 'paths': ['/Users/jonatan/mu_code/test.py'], 'envars': [], 'minify': False, 'microbit_runtime': ''}
2018-10-20 22:50:24,755 - mu.logic:708(_load) INFO: Loading script from: /Users/jonatan/mu_code/test.py
2018-10-20 22:50:24,756 - mu.logic:271(read_and_decode) DEBUG: Trying to decode with utf-8
2018-10-20 22:50:24,756 - mu.logic:274(read_and_decode) INFO: Decoded with utf-8
2018-10-20 22:50:24,757 - mu.logic:287(read_and_decode) DEBUG: Detected newline '\n'
2018-10-20 22:50:24,757 - mu.logic:782(_load) DEBUG: 
2018-10-20 22:50:24,875 - mu.logic:645(restore_session) INFO: Loaded files.
2018-10-20 22:50:24,876 - mu.logic:649(restore_session) INFO: User defined environment variables: []
2018-10-20 22:50:24,876 - mu.logic:653(restore_session) INFO: Minify scripts on micro:bit? False
2018-10-20 22:50:24,892 - mu.logic:1112(change_mode) INFO: Workspace directory: /Users/jonatan/mu_code
2018-10-20 22:50:29,644 - mu.logic:842(save_tab_to_file) INFO: Saving script to: /Users/jonatan/mu_code/test.py
2018-10-20 22:50:29,645 - mu.logic:843(save_tab_to_file) DEBUG: f
2018-10-20 22:50:29,655 - mu.logic:1138(autosave) INFO: Autosave detected and saved changes in /Users/jonatan/mu_code/test.py.
2018-10-20 22:50:34,485 - mu.logic:842(save_tab_to_file) INFO: Saving script to: /Users/jonatan/mu_code/test.py
2018-10-20 22:50:34,485 - mu.logic:843(save_tab_to_file) DEBUG: första_talet = 1
andra
2018-10-20 22:50:34,489 - mu.logic:1138(autosave) INFO: Autosave detected and saved changes in /Users/jonatan/mu_code/test.py.
2018-10-20 22:50:39,486 - mu.logic:842(save_tab_to_file) INFO: Saving script to: /Users/jonatan/mu_code/test.py
2018-10-20 22:50:39,486 - mu.logic:843(save_tab_to_file) DEBUG: första_talet = 1
andra_talet = 2

2018-10-20 22:50:39,490 - mu.logic:1138(autosave) INFO: Autosave detected and saved changes in /Users/jonatan/mu_code/test.py.
2018-10-20 22:50:44,485 - mu.logic:842(save_tab_to_file) INFO: Saving script to: /Users/jonatan/mu_code/test.py
2018-10-20 22:50:44,485 - mu.logic:843(save_tab_to_file) DEBUG: första_talet = 1
andra_talet = 2
svar = första_talet 0
2018-10-20 22:50:44,488 - mu.logic:1138(autosave) INFO: Autosave detected and saved changes in /Users/jonatan/mu_code/test.py.
2018-10-20 22:50:49,487 - mu.logic:842(save_tab_to_file) INFO: Saving script to: /Users/jonatan/mu_code/test.py
2018-10-20 22:50:49,490 - mu.logic:843(save_tab_to_file) DEBUG: första_talet = 1
andra_talet = 2
svar = första_talet + andra_talet
2018-10-20 22:50:49,494 - mu.logic:1138(autosave) INFO: Autosave detected and saved changes in /Users/jonatan/mu_code/test.py.
2018-10-20 22:50:54,486 - mu.logic:842(save_tab_to_file) INFO: Saving script to: /Users/jonatan/mu_code/test.py
2018-10-20 22:50:54,486 - mu.logic:843(save_tab_to_file) DEBUG: första_talet = 1
andra_talet = 2
svar = första_talet + andra_talet

2018-10-20 22:50:54,490 - mu.logic:1138(autosave) INFO: Autosave detected and saved changes in /Users/jonatan/mu_code/test.py.
2018-10-20 22:50:54,946 - mu.logic:1039(show_admin) INFO: Showing logs from /Users/jonatan/Library/Logs/mu/mu.log

```

tjguk, 2018-10-21:
Thanks for picking this up, @jonatanskogsfors. I think that all the principal Mu developers are Uk-based so we tend to miss non-obvious Unicode issues. Off the top of my head this one may be more involved because of interactions with the QScintilla control, but we certainly need to track it down.

tmontes, 2019-01-24:
Investigated this a bit. Here's what I've learned so far:
* QScintilla's code completion setup, as set in `editor.py:EditorPane.set_theme()`, uses a Lexer to help out.
* Browsing the docs I found out about the Lexer's [wordCharacters method](http://pyqt.sourceforge.net/Docs/QScintilla2/classQsciLexer.html#a5c184e036d1b4272e997cb9fc56b1285) that, by default, returns nothing, meaning:

> upper and lower case alphabetic characters and underscore

...I gather that "upper and lower case alphabetic" actually means upper and lower case ASCII letters, which is consistent with the behaviour described by @jonatanskogsfors which I also experienced myself.

More:
* Without much further information, I added a `wordCharacters(self)` method to the `PythonLexer` class used by the editor.
* I first tried returning a simple string like `"abc"`: seemed to work in the sense that words/identifiers containing letters others that those three were missing from the auto-completion drop down.
* I then I tried returning the concatenation of `strings.ascii_letters`, `strings.ascii_digits`, `'_'`, and `'áéíóú'` (a few accented Portuguese ones): in short `'abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ0123456789_áéíóú`'.
* This lead to an early failure `TypeError: function takes exactly 5 arguments (1 given)`, which I did not investigate.

Nearly there:
* Knowing that the editor is in "UTF-8 mode" (whatever that means, exactly in relation to its behaviour and APIs), I tried returning the UTF-8 encoded byte string of that.
* Result is kind of promising and half-disconcerting:
  * Auto completion window includes identifiers with both ASCII and the "áéíóú" letters I specified.
  * However, the identifiers in the drop-down window are displayed as Latin1 decoded from UTF-8 bytes.

See screenshot below:

[image]

For confirmation:

```python
>>> s1 = 'maria_á'
>>> s2 = 'maria_é'
>>> s1.encode('utf8').decode('latin1')
'maria_Ã¡'
>>> s2.encode('utf8').decode('latin1')
'maria_Ã©'
```

Outstanding questions:
* Why on earth is that data being decoded as Latin1 / ISO-8859-1?
* Is there any setting that must be put in place for it to be decoded as UTF-8?
* Or is this an unfortunate QScintilla bug?

Would love to hear from others with more experience with Qt / QScintilla on this (given that mine is virtually zero!).

tmontes, 2019-01-24:
Investigated further:
* Grabbed QScintilla source from [here](https://www.riverbankcomputing.com/software/qscintilla/download).
* Both in `Qt4Qt5/qsciscintilla.cpp` and in [these docs](http://pyqt.sourceforge.net/Docs/QScintilla2/classQsciScintilla.html) found two interesting methods:
  * `bool isWordCharacter (char ch) const`
  * `const char *QsciScintilla::wordCharacters() const`
* The first one is called in the `void QsciScintilla::handleCharAdded(int ch)` method which, at some point, goes and `// See if we might want to start auto-completion`.
* It is also called in `void QsciScintilla::startAutoCompletion(...)` in the process of inspecting the document's content and producing the possible matches list.
* Tried calling them (their Python equivalents) from `editor.py:EditorPane.__init__`.
* About `self.isWordCharacter`:
  * Thought it would be called by other parts of the code like it is in the underlying C++ source code. But overriding it at Python level had no effect.
  * Tried calling it explicitly with `print(self.isWordCharacter('a'))` which worked fine, printing `True`.
  * Then tried `print(self.isWordCharacter('á'))`which raised an exception: `TypeError: a bytes-like object is required, not 'str'` -- yikes, it liked `str` before, now it wants `bytes`...
  * Then tried `print(self.isWordCharacter('á'.encode('utf-8')))`which raised a different exception: `TypeError: bytes or ASCII string of length 1 expected` -- ups, feels like no more than one byte is expected; this makes sense based on the C++ method signature.
* About `self.wordCharacters()`:
  * Tried `print(repr(self.wordCharacters()))` without any Lexer customisation (see my previous comment wrt to `PyhonLexer.wordCharacters`) which worked fine printing `'_abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ0123456789'`, consistent with line 58 of `Qt4Qt5/qsciscintilla.cpp` where `static const char *defaultWordChars` is declared and defined.
  * Repeating the same `print` with a customized Lexer kept failing unless the `Lexer.wordCharacters()` method returned a `str` with ASCII only letters/symbols.

Parting thoughts:
* Either I'm failing to see something pretty obvious...
* ...or, per the C++ level implementation, it feels like handling non-ASCII UTF-8 encoded characters as valid "word characters" for the purposes of auto-completion may be out of Scintilla's current abilities.

Last crazy attempt:
* Disabled UTF-8 mode by passing `False` to both `self.setUtf8(...)` calls in `editor.py` (why do we have two?).
* Adapted the `Lexer.wordCharacters` to return a Latin-1 encoded string pretty much like in my previous comment.
* It worked (screenshot below)

[image]

Preliminary conclusion:
* The underlying C++ code does not seem ready to handle single/multi-byte variable text encodings like UTF-8 (maybe it is, I'm no expert on the code base here!).
* It all seems to work under the old-fashioned premise that 1 byte ~ 1 letter, thus disabling UTF-8 mode and (guessing) using Latin 1 instead.

Next steps:
* Not really sure...
* Would still love to hear somebody else's take on this topic.
* Would also love to know where an "issue" could be logged. I found no obvious way to do that on the Riverbank Computing website.

*over and out* :)

carlosperate, 2020-10-26:
> Would also love to know where an "issue" could be logged. I found no obvious way to do that on the Riverbank Computing website.

If the issue was with QScintilla it would be in their mailing list, but if the issue is with Scintilla that will likely be here? https://sourceforge.net/p/scintilla/bugs/

It's been a while so we could try the latest version and see if the issue persist?