mu-editor/mu #682: Installing on Windows raises a PermissionError against the registry

tjguk, 2018-10-21:
This is a placeholder issue until I can reproduce it but, from memory:

Win10 Pro 17134
64-bit 1.0.1 installer from codewith.mu 
Installed with default settings
During the installation messages I noticed a PermissionError exception relating to a registry key.

At the time I was with a bunch of Year 8s and was moving fairly quickly so I didn't pick up the message. I'll look for NSIS logs and/or reproduce. There were no apparent problem actually running Mu afterwards, although we were only doing very simple things

tjguk, 2018-10-21:
Ok; so the offending code is in a file which is generated by the Pynsist packaging. If you've built, you can see the code here: mu\build\nsis\_system_path.py:add_to_system_path

It seems that the function is being called without specifying whether it's an all-users install or not. The function defaults to True so the code will attempt to open the system Environment reg key for writing -- which won't be allowed.

It look as if this is an upstream bug in pynsist so I'll look for somewhere to report it to there. Meanwhile, the only fallout from our point of view is that Mu's location isn't added to the PATH env var -- which is unlikely to affect anyone badly.

tjguk, 2018-10-21:
Looks like this is https://github.com/takluyver/pynsist/issues/167

takluyver, 2018-10-30:
I've had a go at fixing this in Pynsist; if you have time to test with it, that would be welcome:

https://github.com/takluyver/pynsist/pull/170

carlosperate, 2018-10-30:
I've created a branch (https://github.com/mu-editor/mu/tree/pynsist) that installs that version of pynsist (https://github.com/takluyver/pynsist/pull/170/commits/1d9af0c759811fc66aed3305741eaf0aba316b74), so AppVeyor can build an installer for testing, hopefully that helps @tjguk.

https://s3-eu-west-2.amazonaws.com/mu-builds/windows/mu_2018-10-30_14_38_pynsist_65c805c_64bit.exe

https://s3-eu-west-2.amazonaws.com/mu-builds/windows/mu_2018-10-30_14_49_pynsist_65c805c_32bit.exe

tjguk, 2018-11-03:
I can confirm that the x64 installer works ok. If I get a chance I'll try the 32-bit but I don't imagine there's a problem. Thanks for doing those, @carlosperate. And thanks for the fix, @takluyver .

AFAICT we're not vendoring the pynsist code with mu, so it looks as though the actions are that maintainers install the updated pynsist and that @carlosperate  branch be merged so Appveyor etc. pick up the right version of pynsist.

takluyver, 2018-11-03:
Thanks @tjguk. Just to check, did you try both installing and uninstalling with both a user install and an admin install?

I think @carlosperate's branch is only needed for this testing. Once we're happy with the fix, I'll make a new Pynsist release, and you should benefit from it automatically the next time you build Mu. It doesn't look like you're pinning Pynsist, so you shouldn't need to do anything.

carlosperate, 2018-11-03:
Yep, the CI builds will install the latest version from the requirements.txt file, so when a new version is released it will start using that one.

takluyver, 2018-11-06:
OK, released Pynsist 2.3 - you can hopefully close this.

tjguk, 2018-11-11:
Fixed upstream. Thanks @takluyver

closed 2018-11-11 by tjguk