mu-editor/mu #683: Blank line appears in program output

tjguk, 2018-10-21:
(This was spotted by a Year 8 group I was teaching who took the trouble to make sure it could be reproduced and in the same place).

Take the code in gist https://gist.github.com/tjguk/61bd3e38c56fb5e1467381e5a9aa117e and run it in Mu. Between number 57 & 58 a blank line appears. It always appears and it always appears there. If I change the program to run without fizzing or buzzing, no blank line appears. Which leads me to think it's a buffer-length thing.

I've only tried on Win10 / Mu 1.0.1 from the installer

[image]

tjguk, 2018-10-21:
Relevant logs attached though I'm not seeing anything especially useful

[mu-logs.txt](https://github.com/mu-editor/mu/files/2499045/mu-logs.txt)

tim-mccurrach, 2018-10-22:
I'm not 100% sure, I can't run the code cos not with laptop at the moment, but this looks like it might be related to issue #674.

I think that the '\r\n' that's appended to whatever is printing, gets split across the 256 byte chunk in `read_std_out` and you end up with `... \r` and `\n...` giving two new lines instead of one. I'm nearly done with a pull request to fix this, that I'll probably submit this evening :)

tim-mccurrach, 2018-10-22:
I think this is related to issue #674 

The `\r\n` on the end of a print gets split (by the 256 byte chunks) into a `...\r` and a `\n...` giving two new lines instead of one. I'm working on a PR for this which I'll submit this evening, I've just got to write the tests :)

sorry, if this is a duplicate reply. I just replied but it didn't seem to go through??

tjguk, 2018-10-22:
Thanks @tim-mccurrach that certainly sounds plausible and along the lines of what I expected. I'll look out for your PR whenever and test it against our code.

carlosperate, 2020-10-26:
Looks like #723 has fixed the issue as I am not able to replicate any more :)

closed 2020-10-26 by carlosperate