mu-editor/mu #686: Undefined Symbol

rganswindt, 2018-10-23:
I get the following error(s):
```
ralf@Nancy:~$ mu-editor
Traceback (most recent call last):
  File "/home/ralf/.local/bin/mu-editor", line 7, in <module>
    from mu.app import run
  File "/home/ralf/.local/lib/python3.5/site-packages/mu/app.py", line 29, in <module>
    from PyQt5.QtCore import QTimer, Qt
ImportError: /home/ralf/.local/lib/python3.5/site-packages/PyQt5/QtCore.so: undefined symbol: PySlice_AdjustIndices
```
PyQT5 package version is 5.11.3 and mu-editor installation did not complain:
```
ralf@Nancy:~$ pip3 install --upgrade PyQT5
Collecting PyQT5
  Downloading https://files.pythonhosted.org/packages/d4/bf/d884da8e2f7096d201c891d515eb6813a8e85df5eb6f5e12e867bf1d831c/PyQt5-5.11.3-5.11.2-cp35.cp36.cp37.cp38-abi3-manylinux1_x86_64.whl (117.8MB)
    100% |████████████████████████████████| 117.8MB 177kB/s 
Requirement already satisfied, skipping upgrade: PyQt5_sip<4.20,>=4.19.11 in ./.local/lib/python3.5/site-packages (from PyQT5) (4.19.13)
mu-editor 1.0.1 has requirement pyqt5==5.11.2, but you'll have pyqt5 5.11.3 which is incompatible.
Installing collected packages: PyQT5
  Found existing installation: PyQt5 5.11.2
    Uninstalling PyQt5-5.11.2:
      Successfully uninstalled PyQt5-5.11.2
  Rolling back uninstall of PyQt5
Could not install packages due to an EnvironmentError: [Errno 13] Permission denied: '/usr/local/lib/python3.5/dist-packages/PyQt5-5.11.3.dist-info'
Consider using the `--user` option or check the permissions.

pip3 install --upgrade mu-editor
...
Requirement already satisfied, skipping upgrade: pyqt5==5.11.2 in ./.local/lib/python3.5/site-packages (from mu-editor) (5.11.2)
...

ralf@Nancy:~$ pip3 show PyQT5
Name: PyQt5
Version: 5.11.2
Summary: Python bindings for the Qt cross platform UI and application toolkit
Home-page: https://www.riverbankcomputing.com/software/pyqt/
Author: Riverbank Computing Limited
Author-email: info@riverbankcomputing.com
License: GPL v3
Location: /home/ralf/.local/lib/python3.5/site-packages
Requires: PyQt5-sip
Required-by: QScintilla, PyQtChart
```
This may be a PyQT5 bug, but may still be of interest to you.

RG

ZanderBrown, 2018-10-24:
Looks a lot like #575

darrell24015, 2018-10-25:
I seem to be having the same issues as posted in #686 #575 and #538 - will try to either update PyQt5 or uninstall/reinstall PyQt5 with 
`pip3 install --upgrade pyqt5`
or
`pip3 uninstall pyqt5 `
`pip3 install pyqt5`

Is this what is intended to fix the issues?  I'm running Ubuntu 16.04.3, fully patched and has Python 3.5.2 installed. Thanks!

carlosperate, 2018-10-25:
How did you install Mu?
If you installed it from the master branch from this repository then you can pull the latest version and `pip install -e .` or `pip install .` (depending if you want to be able to edit the Mu code from that directory, or simply install it in the python/virtual environment)
If Mu is installed using `pip install mu-editor` then version 1.0.1 should not have been able to run with Python 3.5.

tmontes, 2019-05-31:
Hi all, is there any particular action than can/should be taken in this issue's context? There have been recent changes in the way Mu is installed from source/wheel and how dependencies are managed.

With no feedback, maybe this issue can be closed?

tjguk, 2019-05-31:
Agree with @tmontes. This hasn't moved for a while.

@rganswindt please feel free to try with the latest (dev) release and re-open if you think there's an issue which we (the Mu team) can address

carlosperate, 2020-09-10:
It's been a couple of years, and as updated dependencies have likely resolved this issue I'll close it. Please feel free to reopen this issue If anybody is still experiencing this.

closed 2020-09-10 by carlosperate