mu-editor/mu #689: Plotter capability not available in 1.0.1 on Raspberry Pi (latest Stretch raspbian)

topshed, 2018-10-26:
After complicated install (see issue #688), I got Mu 1.0.1 installed on a PI, but Plotter button is missing.

2018-10-26 11:48:09,456 - root:112(run) INFO: 

-----------------

Starting Mu 1.0.1
2018-10-26 11:48:09,457 - root:113(run) INFO: uname_result(system='Linux', node='raspberrypi', release='4.14.71-v7+', version='#1145 SMP Fri Sep 21 15:38:35 BST 2018', machine='armv7l', processor='')
2018-10-26 11:48:09,457 - root:114(run) INFO: Python path: ['/usr/local/bin', '/usr/lib/python35.zip', '/usr/lib/python3.5', '/usr/lib/python3.5/plat-arm-linux-gnueabihf', '/usr/lib/python3.5/lib-dynload', '/usr/local/lib/python3.5/dist-packages', '/usr/local/lib/python3.5/dist-packages/mu_editor-1.0.1-py3.5.egg', '/usr/local/lib/python3.5/dist-packages/semver-2.8.1-py3.5.egg', '/usr/local/lib/python3.5/dist-packages/requests-2.20.0-py3.5.egg', '/usr/local/lib/python3.5/dist-packages/pyserial-3.4-py3.5.egg', '/usr/local/lib/python3.5/dist-packages/pyflakes-2.0.0-py3.5.egg', '/usr/local/lib/python3.5/dist-packages/pycodestyle-2.4.0-py3.5.egg', '/usr/local/lib/python3.5/dist-packages/pigpio-1.41-py3.5.egg', '/usr/local/lib/python3.5/dist-packages/nudatus-0.0.3-py3.5.egg', '/usr/local/lib/python3.5/dist-packages/matplotlib-2.2.2-py3.5-linux-armv7l.egg', '/usr/local/lib/python3.5/dist-packages/guizero-0.5.4-py3.5.egg', '/usr/local/lib/python3.5/dist-packages/appdirs-1.4.3-py3.5.egg', '/usr/local/lib/python3.5/dist-packages/urllib3-1.24-py3.5.egg', '/usr/local/lib/python3.5/dist-packages/idna-2.7-py3.5.egg', '/usr/local/lib/python3.5/dist-packages/chardet-3.0.4-py3.5.egg', '/usr/local/lib/python3.5/dist-packages/certifi-2018.10.15-py3.5.egg', '/usr/local/lib/python3.5/dist-packages/pytz-2018.6-py3.5.egg', '/usr/local/lib/python3.5/dist-packages/python_dateutil-2.7.4-py3.5.egg', '/usr/local/lib/python3.5/dist-packages/pyparsing-2.2.2-py3.5.egg', '/usr/local/lib/python3.5/dist-packages/kiwisolver-1.0.1-py3.5-linux-armv7l.egg', '/usr/local/lib/python3.5/dist-packages/cycler-0.10.0-py3.5.egg', '/usr/lib/python3/dist-packages', '/usr/lib/python3/dist-packages/IPython/extensions']
2018-10-26 11:48:09,457 - root:115(run) INFO: Language code: en_GB
2018-10-26 11:48:09,937 - mu.logic:540(__init__) INFO: Setting up editor.
2018-10-26 11:48:09,938 - mu.logic:558(__init__) INFO: Settings path: /home/pi/.local/share/mu/settings.json
2018-10-26 11:48:09,939 - mu.logic:559(__init__) INFO: Session path: /home/pi/.local/share/mu/session.json
2018-10-26 11:48:09,939 - mu.logic:560(__init__) INFO: Log directory: /home/pi/.cache/mu/log
2018-10-26 11:48:09,940 - mu.logic:561(__init__) INFO: Data directory: /home/pi/.local/share/mu
2018-10-26 11:48:10,046 - mu.logic:574(setup) INFO: Available modes: debugger, microbit, pygamezero, python, adafruit
2018-10-26 11:48:10,221 - mu.logic:1112(change_mode) INFO: Workspace directory: /home/pi/mu_code
2018-10-26 11:48:10,223 - mu.logic:623(restore_session) INFO: Restoring session from: /home/pi/.local/share/mu/session.json
2018-10-26 11:48:10,227 - mu.logic:624(restore_session) DEBUG: {}
2018-10-26 11:48:10,228 - mu.logic:1069(select_mode) INFO: Showing available modes: ['debugger', 'microbit', 'pygamezero', 'python', 'adafruit']
2018-10-26 11:48:14,278 - mu.logic:673(restore_session) INFO: Starting with blank file.
2018-10-26 11:48:14,351 - mu.logic:1112(change_mode) INFO: Workspace directory: /home/pi/mu_code
2018-10-26 11:48:15,876 - mu.logic:1039(show_admin) INFO: Showing logs from /home/pi/.cache/mu/log/mu.log

ntoll, 2018-10-26:
Hi... as mentioned elsewhere, the plotter is not available in Mu on RPi because the right sort of dependencies are not in Raspbian (yet). This is something over which we don't have any control. When Raspbian is fixed we'll modify Mu to run with the plotter on RPi.

closed 2018-10-26 by ntoll

pintman, 2019-09-20:
Is there a feature request to follow for this? Otherwise it would be great to have this issue open until there is a plotter support on the raspberry pi.