mu-editor/mu #691: pass keyword to cause next line to reduce indent

makeyourownalgorithmicart, 2018-10-28:
I think it would be helpful if the editor understood the "pass" keyword as the end of an indent, and upon pressing enter, the next line had its indent reduced by one.

Other editors do this, eg jupyer notebooks, pycharm.

For example, consider the following code

```
for i in range(3):
    print(i)
    if ( i == 2):
        print("**")
        pass
    pass
```

When return is hit after `range(3):` the editor helpfully indents the next line ready for `print(i)`.

When the first `pass` is entered and return hit, the editor **doesn't** undo the indent level. It would be lovely if it did.

If this feature was implemented, the editing caret with return to indent level 0 after the second `pass`.

PS some coders, myself included, use pass to deliberately mark the end of an indent section. It helps avoid errors, and helps with clearer thinking about what the code is intended to do, akin to "intentional programming".

I am using mu editor 1.01 on macos Mojave.

carlosperate, 2018-10-30:
Thank you for the report @makeyourownalgorithmicart. This is something done within [Qscintilla](https://www.riverbankcomputing.com/software/qscintilla/intro), the code editor used in Mu, and unfortunately Mu has no control on how the automatic indentation is calculated.

A change in this area would have to be implemented in that project and features can be requested in their mailing list: https://www.riverbankcomputing.com/software/qscintilla/intro

Coincidentally, a few days ago somebody reported something similar, where with having comments at the end of the line would remove the automatic indentation of the next line:

```python

def foo():
    # This works
    pass

def bar():    # this doesn't
pass

```

makeyourownalgorithmicart, 2018-10-31:
thank you for getting back - i appreciate it .. i will follow up with Qscintilla

closed 2018-10-31 by makeyourownalgorithmicart