mu-editor/mu #692: Allow users to override locale setting

ntoll, 2018-10-29:
Some people want to force English as the default UI language. Add a setting in the admin pane.

tjguk, 2018-11-11:
See also #693

tulphoon, 2019-02-04:
I'd like to pick up this issue, if nobody is currently working on it.

I'm not a very good programmer, but I hope I can improve by working on this :)

tmontes, 2019-02-05:
Dear All,

I've been thinking a lot about Mu's UI/UX, as the recent lot of issues I've been creating confirms. See, for example,  #762, #761, #757, and #758.

In particular, I'll be facing a use case that justifies this soon (even though, probably, we won't have an updated Mu release by then... that's a whole different issue!): I'll be doing a intro to programming workshop with a few Portuguese kids and I know for a fact that at least one of them will be using his father's laptop, running English localised Windows. This will lead to an English localised Mu interface as well, per the current implementation. Wouldn't it be nice to remove that extra language barrier for this kid if Mu could speak to him in Portuguese?

With this in mind, I came here to suggest a possible UI/UX approach to a user-based interface language selection which I think should not be "buried" inside the Mu Administration modal dialog.

Why I think it should not be "buried" within the Mu Administration dialog:
* Beginner launches Mu and faces an unfamiliar UI language. Even if said beginner is knowledgable enough to click the Cog Wheel icon to get into the Mu Administration dialog, how easy/difficult will it be for him/her to locate the "Change Language here" widget? How many steps and how much guidance will be needed? Is this a good 1st time experience or can we do better? I tend to think that the Mu Administration dialog is for advanced beginners (whatever that means!) and that beginner beginners shouldn't need to care about it. :)

Proposal:
* Let's go for a balanced approach between "somehow-in-your-face-and-a-single-click-away" and "please-stay-out-of-my-sight-and-dont-grab-my-attention-too-much".
* We could have a mini-icon button (a Flag? a Globe? a Speech Bubble?) next to the Cog Wheel icon button in status bar.
* Clicking it would bring up a modal dialog, pretty much like the current Mode Selection / Mu Administration ones, for consistency, reading, top to bottom:
  * A "prompting phrase", displayed in all languages Mu supports, saying "Choose your language". It might need some kind of horizontal scrolling/animation to accommodate all localisations of said prompt.
  * Much like the Mode Selection dialog, a list of languages with flag icons to the left, language names to the right, written in the language itself (ie: "Deutsche", "English", "Español", "Français", ...).
  * A short instruction phrase, automatically localised per the clicked language in the list saying something along the lines of "English will be used, please click Ok and restart Mu".
  * "Cancel" + "Ok" buttons.

Here's a mockup I created to illustrate the idea:

[image]

Icon credits:
* Toolbar Flag Icon from [here](https://www.iconfinder.com/icons/2008361/flag_international_language_icon).
* Round and simplified flag icons, for consistency with Mu's round buttons, from [here](https://www.flaticon.com/packs/countrys-flags).

Thoughts:
* Maybe using flags to indicate language is wrong (but they're nice and attractive, at least for kids!) There are many English and Spanish speaking countries around the world, each with their own flags: even though the underlying localisation engine supports such variations (i.e.: Australian English, Canadian English, UK English, etc) and the current Mu version includes two variations of Portuguese (PT and BR), I'm not 100% sure about the flags thing.
* Ideally, selecting a language and clicking Ok would update the whole UI at runtime, not requiring a restart. From a few tests I explored yesterday, either using the current `gettext` module or going with `Qt`'s localisation tools, seems to require non-trivial overall changes to support that.
* Thus the only runtime localised message at the bottom of the dialog telling users, in their own language, please click Ok and restart Mu, like I've included in the mockup where Spanish is selected.
* If, however, we could implement runtime UI language updates, I would then suggest we avoid a modal dialog and consider something non-modal, like a drop-down (drop-up?) list from the status bar button icon, for an even more direct and streamlined experience.
* Maybe the "detected" system language could be somehow indicated in the language list? Not sure if that would be useful, though.

Related:
* The flag (globe? speech bubbles?) icon in the status bar, next to the Cog Wheel one works better along with my other UI proposals in #761 and #762 that "clean up" the status bar, as side effects of its ideas.

Thanks for reading through this. I would love to hear feedback.

ntoll, 2019-02-19:
OK... some quick feedback.

The manual language configuration has only been asked for by "advanced" users (i.e. developers who have their OS set to one locale, but program in "English").

We've not had any feedback from teachers or beginners about manually setting this sort of thing, except for comments like, "We really like that there is a native Chinese translation, thank you!".

The feedback I got from non-native English speaking teachers (for example, a discussion at PyParis with a group of French computing teachers was invaluable) was that beginners have no problem with Python being an "English" programming language, but they struggle if the tools are not in their native language. A comment (again from someone in France) I got via email was that it'd be good to be able to start in French and then switch to English since most of the other development tools are "in English".

It's for this reason (along with the fact that teachers have asked to manually set the size of button icons, select if tooltips are displayed and modify the colours of a "custom theme" to help dyslexic coders) that I'm going to implement a "UI configuration" tab in the admin dialog.

The behaviour is currently that Mu uses the OS's current locale for the translation (and falls back to English if no translation exists). I'm not convinced that beginner users will want to be able to select a custom locale with an extra button. In fact, feedback from UK teachers and the Raspberry Pi Foundation education team suggests that less buttons is better since it's one less thing for a kid (especially) to accidentally click. This is the thinking behind putting all the "extra-non-coding related stuff" into the admin dialog that can only be accessed via the cog.

Thoughts..?

tmontes, 2019-02-19:
> We've not had any feedback from teachers or beginners about manually setting this sort of thing, except for comments like, "We really like that there is a native Chinese translation, thank you!".

I'd certainly be using that in two weeks: I'll be working with four kids, one of them confirmed to be using an English localised Windows laptop... A Portuguese Mu for him would be easier. :)

(what I'll do: I'll hack his Mu installation and hard code it to `pt_PT`) :)

> The feedback I got from non-native English speaking teachers (for example, a discussion at PyParis with a group of French computing teachers was invaluable) was that beginners have no problem with [...]

Generally agreed, yes. However lots of computers in PT are English localised, interestingly. That can be an obstacle for kids, mostly. :)

> It's for this reason (along with the fact that teachers have asked to manually set the size of button icons, select if tooltips are displayed and modify the colours of a "custom theme" to help dyslexic coders) that I'm going to implement a "UI configuration" tab in the admin dialog.

Makes all the sense.

Related to your comment but off-topic in this issue: would love [to contribute](https://github.com/tmontes/mu/tree/no-call-tips) with a checkbox that would enable/disable the "helper editor popups" that display, for example, function signatures, etc. Those are very useful for intermediate beginners, but very distractive for complete beginners. What do you say? :)

 
> The behaviour is currently that Mu uses the OS's current locale for the translation (and falls back to English if no translation exists). I'm not convinced that beginner users will want to be able [...]

You've won me with the "one less thing for beginners to accidentally click": into the Admin / UI dialog tab it goes! :)

aroberge, 2019-04-30:
I **strongly** supports the idea of having a visible way for users to change the locale. 

Suppose I speak Italian, but not English, and have an Italian based OS. Currently, Mu does not have an Italian version - so it will default to English. I bet you that many Italian speaker would be appreciate being able to switch between French/Spanish/Portuguese and likely be able to figure out whatever they need from these languages, instead of being stuck with English.  (The same argument will hold for other language combination if and when an Italian translation is made.) 

It is wrong to presume that English is automatically the best choice if the translation in one's preferred language is not available.

Choosing between a editor mode or REPL may seem like a basic option but, for someone that is just beginning, it is definitely not basic as it requires some knowledge about programming. Still, Mu, and rightly so, chose to make this kind of choice easily available in its UI.  

Choosing the language for the UI [*or for friendly tracebacks* ;-)] is something even more basic than choosing a programming mode. It does not require any programming knowledge, and yet can be crucial in reducing the intimidation factor for beginners. In my opinion, such a choice should be available in Mu's UI.

I do not subscribe to the argument that this is not a problem because no feedback has been received from teachers or beginners about manually setting this sort of thing.  Most often, one does not hear from people who feel they cannot use a tool because of a language barrier.  **Anecdote**: I started my programming journey 15 years ago and wrote http://rur-ple.sourceforge.net/ because I wanted my kids to learn programming in French. I had looked at http://gvr.sourceforge.net/ and did not realise that it used the same strategy as Mu, picking up the locale from Windows - installed in English on my computer.  At the time, GvR **did** have a French version, something I only found out much later; had I known, I would not have written rur-ple.  And yes, rur-ple has a language selector as one of the core UI element.  And, no, I did not give any feedback to the GvR developers on this issue - at least, not until rur-ple was already well under way and already included a language selector.  

**Note** I am definitely not suggesting something as clumsy looking as what I had in rur-ple:

[image]

aroberge, 2019-12-18:
This tread refers to an Admin/Ui dialog tab. Does this refer to the "gear" icon at the bottom? If so, I do not see any such language choice in the current version (1.1.0.alpha.2)

ZanderBrown, 2021-03-03:
http://www.flagsarenotlanguages.com/blog/

rex2112, 2021-03-05:
I also would like to have this functionality available. In my case, although I live in Mexico and Spanish is my mother tongue, I started using computers and programming several decades ago when everything was in English. I feel much more comfortable with computing concepts in English than in Spanish.

Although not a deal breaker, most software packages allow changing the user language to any other available. It would be nice to have this available. I am guessing a setting under settings.json ("language_code": "en_US", for example).

Thanks.

> 
> I **strongly** supports the idea of having a visible way for users to change the locale.
>

deeplook, 2021-03-12:
Same here. Switching between multiple languages would make Mu more user-friendly because people don't have to switch their OS locales or install Mu for multiple users on the same machine. This is especially true for parents and children. It's hard to believe that this is not immediately considered to be a valuable feature.