mu-editor/mu #693: Start-up Switches

tjguk, 2018-10-29:
We've had several suggestions which amount to: can I start up Mu in *this* way, Amnesia mode,  
override locale, forget previous open file, use specific theme / layout.

The thinking behind these is that teachers can set up a shortcut which specifies certain characteristics so you don't have to run through a litany of "Go Dark; Close open files; Switch to micro:bit mode" etc.

I'm probably something like 25/75 on the notion. Most of the things are *that* hard to do manually and there's benefit in the learners being aware of how to do them. However, someone mentioned MLD or SLD kids who struggle to do many things and would benefit from the leg-up that a preconfigured environment could give them. And there are things which are more involved -- eg the full "amnesia" mode requested in #690 

If we wanted to go this way there seem to be two obvious approaches (which could be combined):

* Add a command-line switch for each option (--theme=daytime --mode=microbit etc.)
* Add a command-line switch for a .settings file (--settings=mu.microbit-daytime.settings)

carlosperate, 2018-10-29:
Back when Mu was a single file executable you could have a `settings.json` file in the same directory than the executable, and that file would take precedence.
The intention was to allow a somewhat scalable (copy along exe + json file) and simple way to customise Mu, and it also helped with portability (you could keep both Mu and the settings file in your thumb drive).

I'd be more inclined to have a command line argument to point to a different settings file, than to add a flag per available setting, just for the simplicity of the implementation.
Now that we have a separate settings and session json files, technically a shortcut could point to a remote read-only settings file, and still have individual session data for each user. However, I'm not sure how the keys are separated, right now all my settings are in `session.json` and my `settings.json` file is empty.

tjguk, 2022-07-16:
To make this concrete, I propose to add a section to the `settings` module which does the `argparse` dance to read into a `Settings`-style object called, eg, `Switches`. Those could be then be used within the codebase as `settings.switches['whatever']`