mu-editor/mu #694: Mu on debian pixel Raspberry Pi OS

DavidWhaleMEF, 2018-10-30:
I tried to install Mu on the Raspberry Pi OS Debian Pixel
[image]
 (running in a virtual box on Mac OS 10.10.5) and it seems to (incorrectly) install the older version of Mu...

[image]

[image]

carlosperate, 2018-10-30:
Hi David,

Is it possible you are running Raspbian Jessie? Try running `cat /etc/os-release`.
If you are, it's possible Mu v1.0 is only available in Raspbian Stretch, can you upgrade?

I am not really sure if the Raspbian release is the problem, although it's quite possible since Jessie is running Python 3.4 and old PyQt5 releases that might not be compatible with current Mu.

carlosperate, 2018-10-30:
Running on the latest Stretch image should work.
It would also be interesting to see if the plotter is available on x86 Raspbian, as it's not currently available in the ARM release, but upstream debian packages for PyQtChart should be available.

carlosperate, 2018-10-31:
Maybe it can be installed via apt? There is this package in stretch: https://packages.debian.org/stretch/python3-pyqtgraph

As you have raised in a different issue the current mu wheel might not work, but installing mu-editor via apt should install everything necessary for Mu to work without charts, and manually installing that package should be able to unlock it.