mu-editor/mu #695: Mu won't run on Mac OSX 10.10.5

DavidWhaleMEF, 2018-10-30:
I followed the instructions on the Mu install page for Mac OSX here...

https://codewith.mu/en/howto/install_macos

The package installed and I have an icon in Applications. I have system preferences set to allow the app to run.

When I double click the icon it appears in the task bar dock, but then disappears almost immediately. There are no messages anywhere, it just fails to run.

I ran it from the command line and I see this error:

```
Davids-Air-2:~ davidw$ /Applications/mu-editor.app/Contents/MacOS/mu-editor ; exit;
Traceback (most recent call last):
  File "/Applications/mu-editor.app/Contents/MacOS/mu-editor", line 20, in <module>
    load_entry_point('mu-editor==1.0.1', 'console_scripts', 'mu-editor')()
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/site-packages/pkg_resources/__init__.py", line 480, in load_entry_point
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2693, in load_entry_point
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2324, in load
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2330, in resolve
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 29, in <module>
    from PyQt5.QtCore import QTimer, Qt
ImportError: dlopen(/Applications/mu-editor.app/Contents/Resources/app_packages/PyQt5/QtCore.so, 2): Symbol not found: __os_log_default
  Referenced from: /Applications/mu-editor.app/Contents/Resources/app_packages/PyQt5/Qt/lib/QtCore.framework/Versions/5/QtCore (which was built for Mac OS X 10.11)
  Expected in: /usr/lib/libSystem.B.dylib
 in /Applications/mu-editor.app/Contents/Resources/app_packages/PyQt5/Qt/lib/QtCore.framework/Versions/5/QtCore
logout

[Process completed]
```

This leads me to think that as my Mac is 10.10.5, and QtCore was built for 10.11, that the answer is 'no hope'.

I tried also to install the new Mu on debian pixel in a virtual box, and I got the old Mu, see.. 
https://github.com/mu-editor/mu/issues/694

So, I wonder if someone could help with the following

1) Is it impossible to get the latest Mu running on Mac OSX 10.10.5 (I don't want to upgrade my OS yet as I have lots of events to go to and I don't want to risk trashing my OS so close to them (had bad upgrade experiences before)

2) Could you document some 'minimum OS requirements' on the install pages please, so that it is possible to know *before* downloading and installing what the minimum system requirements are? It would be helpful to set users expectations, especially where (if confirmed here) I don't meet the minimum system requirements.

3) Could you add a friendly pop-up GUI box to whatever launches the application, indicating that Mu has failed to run (and hopefully the reason why, and where to go to fix it)

4) On a related note, is there an *easy* way for me to get the latest Mu running in Debian Pixel in a virtual box (the other ticket) as that would be a good temporary get out of jail free card for me.

Thanks for your help.

DavidWhaleMEF, 2018-10-30:
Ok, just for reference @carlosperate @ntoll I have managed to download the latest web based PythonEditor using the excellent instructions and that has the latest MicroPython baked in it. That will give me a way to do my demos while disconnected from the internet for now. I need the latest code as I am using the set_pull for my 8 home-made touch inputs, and the version of MicroPython bundled with Mu 0.9.13 (the newest version that I can install on this version of Mac OSX) doesn't support that.

So, my immediate demo panic is over, but the points 1..4 above are still relevant (and the UX of the application starting and stopping with no message is likely to confuse real users, so it would be good to catch any exception at startup of Mu that causes it to shut down, and at least display a 'sorry' message!

carlosperate, 2018-10-30:
Hi David,

Due to PyQt5 limitation Mu for macOS only run on 10.11 or later.

To answer some of your questions:

1. Unfortunately not the current version. If you are really keen you could try older version of PyQt5 until one of them launches, but there could be other incompatibilities in the way.

2. It's currently documented in the installation instructions: https://codewith.mu/en/howto/install_macos

3. That would be quite hard, if there are issues importing Mu's GUI framework, then we cannot really show a window without it. I guess we could have a friendlier error message on the console, but other than saying sorry and "please report it on GH" there is no real advantage. When double clicking the Mu icon you would still have the same behaviour with the dock icon appearing and disappearing.

4. I think this is already covered in #694.

DavidWhaleMEF, 2018-10-30:
1. sounds like too much pain, when all I want to do is use set_pull on my mac - life is too short!

2. Oh yes, sorry, I totally missed that! (It's very tiny, but it is there, I missed it!! My bad!)

3. Hmm, shame, it's a really bad UX for people. Is there no way to pop up a simple native GUI box (isn't tkinter bundled in the latest mu anyway, so it could be used to display a simple pop-up).

4. Yes thanks, I'll try that when I have fatter internet (currently tethered on mobile)

So, I think one issue remains on this ticket for possible discussion - tkinter is bundled, so could that be used to pop up an error message if the application fails to start?

Thanks for your help!

brentru, 2018-11-07:
I'm encountering the same issue described with a web installation of the MU DMG on macOS 10.13.6

carlosperate, 2018-11-07:
What do you mean web installation? that you downloaded the DMG from the codewith.mu website?

Could copy here get the error that it shows you? That will help us a lot.

DavidWhaleMEF, 2018-11-08:
My experience was the same - there is no error message. The icon in the taskbar jumps up as it loads and then fails to load the application. The first comment on this issue explains how I got an application log with a sort of error message - perhaps @brentru you can try that method from a terminal window and post the contents of your terminal window here, so we can diagnose? Thanks!

P.S. I opened terminal and typed this...

/Applications/mu-editor.app/Contents/MacOS/mu-editor

pacav69, 2018-11-18:
@DavidWhaleMEF I too have the same issue in that upon startup of mu the icon appears then disappears with no error message.
@carlosperate I would recommend upon startup of mu of having a check of the version of MAC OSX and informing user with a popup window "This software requires a minimum of MAC OSX 10.11"

mimiflynn, 2020-03-22:
I have Catalina installed and this is happening to me with a new download of Mu.

carlosperate, 2020-03-22:
Hi @mimiflynn, as you are running Catalina this would be a different issue.
This is issue might be more relevant for you: https://github.com/mu-editor/mu/issues/997
If you could provide more info there it could also help us debug the problem.

keyra-coder, 2020-04-07:
A number of my students describe the same behaviour with Codewith.mu being clicked from the Apps  - it pops down onto a taskbar and disappears in a flash
I need this to work  in PyGame Zero mode for kids! 
Please let me know the solution

keyra-coder, 2020-04-07:
OsX  Yosemite 10.10.5

whaleygeek, 2020-04-07:
Hi @keyra-coder sadly the only fix seems to be to upgrade the Mac OS in order to use the latest Mu.

I have 10.10.5 and I have the same issue. I tracked it down to the fact that @ntoll used a slightly more up to date version of QT in the latest Mu compared to the orignal 0.9.3 of Mu that I still use.  OSX 10.10.5 does not support this new QTlib version.

You can see this is the issue if you look at the error message (or try to start mu from a terminal prompt) it complains about a missing qtlib/incompatible version).

I don't know of a fix apart from upgrading to a later OSX, but @ntoll or @carlosperate might have some magic for the Mac to allow this shared library to be upgraded on the mac independently, or there may be a way to down-grade the QT requirement in the latest Mu (but both sound hard to me).

I suffer the same fate with the latest version of Oracle Virtual Box, which also uses an updated qtlib that is not supported on the 10.10.5 OSX Yosimite, so I have to use an older version of that too.

Given that my Macbook air that uses Yosimite 10.10.5 is dated (early 2014) - i.e. is 6 years old, I do remember a conversation at the time I discovered this that '6 years old for an OS is a bit old' - however like many schools, I choose not to upgrade as I know that some of the other apps that I rely on heavily are broken in the later OS releases and I don't want to make that move yet.

whaleygeek, 2020-04-07:
Also pinging @jaustin and @microbit-giles as I know that Giles had the same issue (it is possible that he eventually upgraded his OS, but I know he had issues with it breaking other things which he was trying to avoid at the time)

whaleygeek, 2020-04-07:
If you look back at the original issue I logged, here is the evidence that it is an old qtlib

```ImportError: dlopen(/Applications/mu-editor.app/Contents/Resources/app_packages/PyQt5/QtCore.so, 2): Symbol not found: __os_log_default
  Referenced from: /Applications/mu-editor.app/Contents/Resources/app_packages/PyQt5/Qt/lib/QtCore.framework/Versions/5/QtCore (which was built for Mac OS X 10.11)
  Expected in: /usr/lib/libSystem.B.dylib
```

MartyOz, 2020-04-30:
Hi, 

I am having the same issue as described by DavidWhaleMEF at the start of this thread. 

This is my first time posting, so please excuse any lack of pertinent information. I am starting to teach myself to learn Python and was trying to install MU Editor.  

I am running macOS Sierra Version 10.12.6. It would seem, from reading this thread, that MU Editor would run on my laptop. However, I am having the same problem as DavidWhelanMEF. Specifically, to quote him: 

"I followed the instructions on the Mu install page for Mac OSX here...
https://codewith.mu/en/howto/install_macos

The package installed and I have an icon in Applications. I have system preferences set to allow the app to run.

When I double click the icon it appears in the task bar dock, but then disappears almost immediately. There are no messages anywhere, it just fails to run."

I ran terminal and typed the following command:

/Applications/mu-editor.app/Contents/MacOS/mu-editor

This is what i received:

[image]

I copied what I saw others were doing in this thread. I don't fully understand what is being shown there.

I am wondering if anyone can help me figure out what my issue is. I have looked throughout the web and the issue that comes up most seems to be OS that are older but I am running the most current. 

Any advice would be greatly appreciated for a true beginner. 

Thanks, 
M

tmontes, 2020-04-30:
@MartyOz,

Thanks for sharing your experience. I'm guessing you are trying to run Mu 1.0.3 which is a recent update that implicitly (our fault!) now requires a more recent version of macOS -- if you dig deep into the message you shared (thanks!) you should find this snippet:

> (which was built for Mac OS X 10.13)

That's the hint. The docs at https://codewith.mu/en/howto/1.0/install_macos haven't been updated to reflect that new reality, and have been misleading since 1.0.3 was released (sorry for that!). I have just created a PR such that that web page is updated to reflect the fact that the current Mu 1.0.3 requires macOS 10.13 or later.

Not all is lost, however. I believe (no time to confirm now) that the current Mu 1.1.0.alpha2 will still work with computers running macOS 10.11 or later. If you don't mind, I'd suggest you give it a try by downloading it from the top of https://codewith.mu/en/howto/1.0/install_macos, right there in the box titled **TRY THE ALPHA OF THE NEXT VERSION OF MU!**.

Thanks again and, please, let us know of your progress.

MartyOz, 2020-05-07:
Hi @tmontes ,

Thank you very much for your help. I followed your suggestion and I am now able to use MU. Thanks to you! 

I am starting my studies using Automate the Boring Stuff with Python by Al Sweigart. If you have any additional suggestions for me to check out after that, I would welcome them. 

Again, I really appreciate your help in getting me up and running. 

Cheers, 
M

tmontes, 2020-05-10:
@MartyOz,

Glad I could help, but please keep in mind that in the future Mu will invariably require a more modern version of macOS -- much like 1.0.3, it is very likely that the next release, following the 1.1.0.alpha2, will require macOS 10.13 or even something more recent.

Enjoy Mu!

carlosperate, 2020-10-26:
As we cannot really do much about the PyQt5 minimum OS version and we need to update to the latests PyQt to be able to support new operating systems, then all we can really do is document this.

The website has been updated to indicate the High Sierra min version.

While the latest alpha is compatible with older macOS versions, as it is using an older version of PyQt, it also means it has bugs in the latest macOS as well. But while that is okay for an alpha, we will need to update for more stable releases.

So the next v.1.1 alpha version will likely update the PyQt dependency to, at least, the same version used in Mu v1.0.3, and High Sierra will be the min OS supported.

closed 2020-10-26 by carlosperate

PythonCoderUnicorn, 2021-03-05:
so far I am not happy with CircuitPython and its poor website UX design, lack of information regarding the need for a micro USB cable (it assumes you have one and fails to tell you that it is necessary). Following their instruction their stupid Mu editor fails to work. I granted access permissions to let it open and run but fails to work.  
On a mac, Big Sur version 11.2.1 . 
Just not sure what to do

ntoll, 2021-03-05:
Hi @PythonCoderUnicorn - I'm sorry to hear you've been having problems. I believe the next version of Mu will fix the issues you will have encountered. A release is imminent.

I created Mu. I'm a volunteer, as are all the other wonderful folks who work on Mu. While we all understand and empathise with the frustration you must feel, please [be respectful](https://github.com/ntoll/being_together) in the way you address people and the software on which they work.

Thank you. Peace.